heishamon/HeishaMon #265: Questions about automating all this cool stuff

gust4vsson, 2022-08-30:
Heyo, I'm planning on buying the 12kW J heat pump.

Since the electricity prices are ridiculous, I'm paying by the hour now with different prices. I use Home Assistant.

The plan is to let the heat pump heat water in buffert tanks during cheap hours, which will be pumped out on a zone according to set roomp temp, with a separate zone water pump.

During expensive hours I want to completely block the heat pump from running. But I still need the zone water pump to be running, and also the main water pump should be running after anti-freeze  conditions.

Is this possible? What command do I send to the heat pump to block it from running?

str3000, 2022-08-30:
Hello gust4vsson,
since a few weeks I have a 12kw T-CAP Super Silent heat pump.
Nearly same requirements like you, but I want to use my own solar energy.
I have a 3-way valve connected and 2 * 850l for hot water and heating. Optional pcb is installed.
Only one zone and using the embedded pump

I start with the cloud adapter and change it last week with the HeishaMon adapter and I'm absolute happy with it.

For stopping the heat pump, it exists more than one possibility.
Switching off the heat pump, hot water works fine, heater water will not work in case of water flow pump is off.
You're right.
I'm using the sg-ready and it works well. In the documentation is a max. time limit defined, but it works also longer :-)

And I can request heat pump remotely (most together with sg-ready) to user power and go over the regular set temperature.
With the heat pump I will reach 57°C in the buffer tank.
For heating water, I have an additional heating rod installed, but not used up to now.

I'm using ioBroker, but that certainly doesn't matter

Gettings

[image]

IgorYbema, 2022-08-30:
There is a slack link in the readme of heishamon github page. You can put your questions about this there. This issue list is for posting bugs/issues about heishamon. Can you close thise issue please?

gust4vsson, 2022-08-30:
Ah sorry for the inconvenience. Thanks.

closed 2022-08-30 by gust4vsson