heishamon/HeishaMon #266: Parallel Mode Not working

loemic87, 2022-08-31:
Hello,

i ordered a heishamon PCB and 5m cable from tindie. I wabt to usw heishamon parallel to cz-taw1. 

I soldered the connector for cn-nmode in the cable. 

Heishamon works in single Mode, wether connected via cn-nmode oder cn-cnt. Parallel to cz-taw1 it doesnt work. Cz-taw1 doesnt Work then too. It gets an IP Adresse but i can't control the HP via Cloud. 
Heishamon is in listen Mode when connected parallel to cz-taw1. 

Any ideas for me?

IgorYbema, 2022-08-31:
Did you enable 'listen only mode' in heishamon when in parallel? Heishamon can only be used to listen (monitor) and not send commands in parallel mode. But if you don't set this mode, cz-taw1 will not work also anymore

loemic87, 2022-08-31:
Hi @IgorYbema 

As i wrote i Put IT in listen Mode when connected parallel to cz-taw1. In Order to be Safe i also did a reboot and Checked the setting before connnecting to HP.

If Its useful, i am running a WH-MDC09J3E5.

loemic87, 2022-08-31:
i just attached my settings to the previous post. Are there any issues?

IgorYbema, 2022-08-31:
Can  you send the console log for about 30 seconds while this is connected in parallel?

loemic87, 2022-08-31:
Of course... Here is the log:

`Wed Aug 31 10:34:10 2022 (2499222): Heishamon stats: Uptime: 0 days 0 hours 41 minutes 39 seconds ## Free memory: 72% ## Heap fragmentation: 23% ## Max free block: 21832 bytes ## Free heap: 28648 bytes ## Wifi: 100% (RSSI: -48) ## Mqtt reconnects: 1 ## Correct data: 0.00%
Wed Aug 31 10:34:15 2022 (2504223): Heishamon stats: Uptime: 0 days 0 hours 41 minutes 44 seconds ## Free memory: 73% ## Heap fragmentation: 12% ## Max free block: 25888 bytes ## Free heap: 29208 bytes ## Wifi: 100% (RSSI: -50) ## Mqtt reconnects: 1 ## Correct data: 0.00%
Wed Aug 31 10:34:20 2022 (2509224): Heishamon stats: Uptime: 0 days 0 hours 41 minutes 49 seconds ## Free memory: 72% ## Heap fragmentation: 11% ## Max free block: 25888 bytes ## Free heap: 28832 bytes ## Wifi: 100% (RSSI: -48) ## Mqtt reconnects: 1 ## Correct data: 0.00%
Wed Aug 31 10:34:25 2022 (2514225): Heishamon stats: Uptime: 0 days 0 hours 41 minutes 54 seconds ## Free memory: 72% ## Heap fragmentation: 10% ## Max free block: 25888 bytes ## Free heap: 28808 bytes ## Wifi: 100% (RSSI: -48) ## Mqtt reconnects: 1 ## Correct data: 0.00%
Wed Aug 31 10:34:30 2022 (2519226): Heishamon stats: Uptime: 0 days 0 hours 41 minutes 59 seconds ## Free memory: 72% ## Heap fragmentation: 10% ## Max free block: 25888 bytes ## Free heap: 28808 bytes ## Wifi: 100% (RSSI: -48) ## Mqtt reconnects: 1 ## Correct data: 0.00%
Wed Aug 31 10:34:35 2022 (2524227): Heishamon stats: Uptime: 0 days 0 hours 42 minutes 4 seconds ## Free memory: 72% ## Heap fragmentation: 11% ## Max free block: 25888 bytes ## Free heap: 28888 bytes ## Wifi: 100% (RSSI: -48) ## Mqtt reconnects: 1 ## Correct data: 0.00%
Wed Aug 31 10:34:40 2022 (2529228): Heishamon stats: Uptime: 0 days 0 hours 42 minutes 9 seconds ## Free memory: 71% ## Heap fragmentation: 10% ## Max free block: 25888 bytes ## Free heap: 28600 bytes ## Wifi: 100% (RSSI: -48) ## Mqtt reconnects: 1 ## Correct data: 0.00%`

IgorYbema, 2022-08-31:
Does not seem to see any data indeed. Is the CZ-taw1 connected during this logging? The CZ-TAW1 should provide the communcation and heishamon only listens to the same answers.
If the CZ-TAW1 is connected, I do not understand currently why it is not working.

Can you try to connect the CZ-TAW1 first, check if it works without heishamon. And then plugin the heishamon parallel on the CN-NMODE and check what happens then.

loemic87, 2022-08-31:
Cz-taw1 ist connected but doesnt Work in this constellation. 

I will try Out again

loemic87, 2022-08-31:
CZ-TAW1 alone ist working correctly.
Heishamon (listen Mode Off) connected to CN-CNT ist working (100% correct)
Heishamon (listen Mode Off) connected to CN-NMODE ist working (100% correct)
Heishamon (listen Mode on) connected to CN-NMODE and CZ-TAW1 connected to CN-CNT both are Not working

IgorYbema, 2022-08-31:
Ok. So to be clear. When you plugin the heishamon when cz-taw1 is still working it stops working also?

loemic87, 2022-08-31:
I shut Off the Power between each step. 
Should i Plug in heishamon to CN-NMODE when HP is under Power and CZ-TAW1 is working?

IgorYbema, 2022-08-31:
Yes please try that. Should be ok.

IgorYbema, 2022-08-31:
Also I have a plan to check what is going wrong:
- First remove from the CN-NMODE the tx and rx pins , so only 5v and gnd are connected. And reconnect heishamon then. Check if CZ-TAW1 still works (if not, there is something wrong with powering of the heishamon causing the CZ-taw1 not to work)
- Then if ok, then add the rx pin (the one next to the gnd) but still remove tx pin. So only 5v, rx and gnd are connected. To check if adding rx isn't the problem (rx is for sending data to the heatpump, but in listen only  mode this should keep quiet).
- Then if ok, then add the tx pin, but remove the rx pin again.  So only 5v, tx and gnd are connected. To check if CZ-TAW1 fails after connecting tx.

loemic87, 2022-08-31:
> Yes please try that. Should be ok.

Tried... Cz-taw1 seems to stay online. But i cannot do anything in the Cloud. Nee settings are ignored by the HP.
Heishamon doesnt Work either.

loemic87, 2022-08-31:
> Also I have a plan to check what is going wrong:
>
> * First remove from the CN-NMODE the tx and rx pins , so only 5v and gnd are connected. And reconnect heishamon then. Check if CZ-TAW1 still works [...]

rx and tx removed... CZ-TAW1 is not working

IgorYbema, 2022-08-31:
Yes, heishamon should get power from HP. Please remove the external power source. That could cause your situation as there are now two power sources.

IgorYbema, 2022-08-31:
Also, you could keep using the external power source, but then remove the 5v pin from the CN-NMODE connector (only use tx, rx, and gnd)

loemic87, 2022-08-31:
Okay, that was the Problem. Thank you! It works perfect now

closed 2022-08-31 by loemic87