heishamon/HeishaMon #267: Enhancement request - MQTT Discovery

hendrer, 2022-09-04:
Would it be possible to implement MQTT Discovery for Home Assistant instead of relying on the YAML here? I've seen this done with Solar Assistant and it's very clean. 

Let me know if you need more details and if you're interested to implement.

Cheers

hendrer, 2022-09-04:
Example of MQTT Discovery implementation for Home Assistant. https://solar-assistant.io/help/home-assistant/setup

hendrer, 2022-09-04:
HA documentation https://www.home-assistant.io/docs/mqtt/discovery/

IgorYbema, 2022-09-04:
Maybe for v4.0 release but be my guest to try to implement it. It has been asked before but no one seems to really need it.

It also breaks current mqtt logic (need to build a new mqtt tree) and that will then break other implementations.

PGnl, 2022-09-04:
Need it, is indeed true.
I however would like it. Would save me from having to set up heisha on HA for other people as it would automatically be recognised.

not sure how the mqtt auto discovery works, as I have some devices where I never added a mqtt server and they where seamlessly found by HA via the JSON.
Also no way to put in a mqtt server. 
seems that the info is gathered from the JSON directly.
If that is true, just a conversion would be sufficient to the HA mqtt format with a port to a json page.
But maybe I am mistaken on how it works.

geduxas, 2022-09-04:
Truly nothing special it will not bring.. almost everything is in yaml file, and automation just skips yaml configuration.. 

I think it's best to build HA addon to integrate heishamon... and there could me more possibilities.

kamaradclimber, 2022-10-15:
I would be interested by the auto discovery as well. Of course we can do without it, it's mostly for convenience.

kamaradclimber, 2022-10-17:
In the meantime, I've started a prototype of an integration on https://github.com/kamaradclimber/heishamon-homeassistant. It's already usable and reproduce all sensors from https://github.com/Egyras/HeishaMon/blob/master/Integrations/Home%20Assistant/heishamon.yaml (except inputs and automations).

Alwik, 2022-10-20:
Great work @kamaradclimber ! I have tested already, and it works perfectly, thanks a lot!

Alwik, 2022-10-20:
@geduxas
HACS -> Integrations -> 3-dot menu in top right -> Custom repositories -> 
Repository = https://github.com/kamaradclimber/heishamon-homeassistant
Category = Integration
After a reboot it should detect a new integration

geduxas, 2022-10-20:
> @geduxas HACS -> Integrations -> 3-dot menu in top right -> Custom repositories -> Repository = https://github.com/kamaradclimber/heishamon-homeassistant Category = Integration After a reboot it should detect a new integration

Sorry i was too quick :D i found it's already HACS accessable. I thought about this kind of integration but didn't have chanse to start! Nice work. Do you think to make climate component for that? Maybe it's on the way too?

kamaradclimber, 2022-10-20:
Thanks for the feedback @geduxas :) I hope it will be useful.
I have not thought of doing a climate component, I need to have a look at those to see if it would fit since heatpumps are usually handling temperature by themselves. Let me toy a bit with it!

Regarding integration in hacs, right now it is a custom repository. I would like to make it a default repository but given number of pending PR on the hacs repository, it might take a while.

If you have suggestions/bugs, feel free to open issues on the integration repository.

geduxas, 2022-10-20:
> Thanks for the feedback @geduxas :) I hope it will be useful. I have not thought of doing a climate component, I need to have a look at those to see if it would fit since heatpumps are usually [...]

I am talkong about mqtt.climate component.. it will be desplayed in HA in climate card.. so it will be easer to adjust setpoint temperature.. i have done some testings.. but with integration it could be fulfilled. :) Becouse it should be case sensitive.. one have heatshit curved settings... Another will be on directe temperature settings.. also 1/2 zones..

geduxas, 2022-10-20:
here is mine: 
[image]

kamaradclimber, 2022-10-23:
I've added a climate entity for the water heater and opened https://github.com/kamaradclimber/heishamon-homeassistant/issues/25 about creating climate component for zone1 & zone2.

I would suggest to continue discussion there to avoid overwhelming this issue which is unrelated.

hendrer, 2022-10-24:
> kamaradclimber/heishamon-homeassistant#25

I spent some time implementing the MQTT native integrations that might help
you all. See attached. It has some personal sensors in there but the top
half is relevant

mqtt:
    climate:
        # PHP Controller
        - name: PHP Controller
          precision: 1.0
          temperature_unit: C
          min_temp: 0
          max_temp: 40
          current_temperature_topic: "panasonic_heat_pump/main/Room_Thermostat_Temp"
          #power_command_topic: "panasonic_heat_pump/commands/SetHeatpump"
          #payload_on: 1
          #payload_off: 0
          modes:
            - "off"
            - "heat"
            - "auto"
            - "cool"
          mode_state_topic: "panasonic_heat_pump/main/Operating_Mode_State"
          mode_state_template: >-
            {%- if value in ("0","4") -%}
               
                    heat
            
            {%- elif value in ("1","5") -%}
               cool
               
            {%- elif value in ("2","6","7","8") -%}
                auto
            
            {%- else -%}
                off
            {%- endif -%}
          mode_command_topic: "panasonic_heat_pump/commands/SetOperationMode"
          mode_command_template: >-
            {%- if value == "heat" -%}
                
                {%- if states('climate.php_dhw_controller')  in ("heat") -%} 
                    4
                {%- else -%}
                    0
                {%- endif -%}
                
            {%- elif value == "cool" -%}
                 {%- if states('climate.php_dhw_controller')  in ("heat") -%} 
                    5
                {%- else -%}
                    1
                {%- endif -%}
            {%- elif value == "auto" -%}
                 {%- if states('climate.php_dhw_controller')  in ("heat") -%} 
                    6
                {%- else -%}
                    2
                {%- endif -%}
            {%- else -%}
                3
            {%- endif -%}
          preset_modes:
            - "Heat"
            - "Cool"
            - "Auto(Heat)"
            - "DHW"
            - "Heat+DHW"
            - "Cool+DHW"
            - "Auto(Heat)+DHW"
            - "Auto(Cool)"
[... 907 lines cut ...]
        value_template: >-
          {%- if states('sensor.php_w_consumption') | float(0) > 0 -%}
            {{ '%0.1f' % ((states('sensor.php_w_production') | float ) / (states('sensor.php_w_consumption') | float )) }}
          {%- else -%}
            0.0
          {%- endif -%}
        availability_template: >-
          {%- if is_number(states('sensor.php_w_consumption')) and is_number(states('sensor.php_w_production')) %}
            true
          {%- else %}
            false
          {%- endif %}

  #Calculation Energy consumed (kWh) - DHW
  - platform: integration
    source: sensor.php_dhw_power_consumed
    name: PHP DHW Energy Consumed
    unit_prefix: k
    method: left
  #Calculation Energy consumed (kWh) - Heat    
  - platform: integration
    source: sensor.php_power_consumed
    name: PHP Heat Energy Consumed
    unit_prefix: k
    method: left

closed 2023-01-16 by Egyras