heishamon/HeishaMon #268: V3.0 - S0 metering stops working

Bucky2k, 2022-09-07:
Hey,

no matter what I try - enlarge the accepted pulse length range, etc - I do not get any S0 metering signals shown. With V2.0 it shows the power consumption as intended, but not with V3.0. Any hint of a possible cause? PCB is from Igor inkl. OpenTherm (but not used)

IgorYbema, 2022-09-07:
What are your settings (screenshot please)? And if you revert to v2.0 does it work again?

Bucky2k, 2022-09-07:
Settings 2.0:
[image]
Settings 3.0:
[image]

Back to V2.0 the S0 are counting mostly fine, however some peaking of the detected watts from time to time. 
DRS255BC is rated with 90ms pulse length

Bucky2k, 2022-09-07:
After a while, at 3.0, pulse quality is shown to be 0 on the active S0 (i.e. S0 2):
[image]

IgorYbema, 2022-09-07:
Thanks. I don't know why it isn't working. Maybe you can send some text log from the console. But you must be sure that during the log there must have been some pulses. 

Maybe try swapping the two wires from the s0. But that should not change anything as it already works in v2.0. But doesn't hurt to try :-)

IgorYbema, 2022-09-07:
Got another idea. Maybe there is just too much noise received. V2.0 has other logic to filter noise. V3.0 is working differently. This is too remove those high peaks you see in v2.0 but if the noise is too much every pulse is counted as bad.
Some people reported that when grounding the meter to the chassis of the heatpump (the wire which is also connected to the gnd side of the s0 port of heishamon) fixes the noise. Others fix it by using proper shielded cable.

Bucky2k, 2022-09-07:
3.0: Noting related to S0 in the console / pulses not detected at all

2.0 (swapping S1 S2 done):

`Measured Watthour on S0 port 1: 4.00
Measured total Watthour on S0 port 1: 2542.00
Calculated Watt on S0 port 1: 1998
Measured Watthour on S0 port 1: 5.00
Measured total Watthour on S0 port 1: 2547.00
Calculated Watt on S0 port 1: 3662
Requesting new 1wire temperatures
Heishamon stats: Uptime: 0 days 0 hours 2 minutes 40 seconds ## Free memory: 74% 26920 bytes ## Wifi: 100% ## Mqtt reconnects: 1 ## Correct data: 100.00%
Requesting new panasonic data
sent bytes: 111 including checksum value: 18 
Received 203 bytes data
Checksum and header received ok!
received TOP1 Pump_Flow: 15.53
received TOP8 Compressor_Freq: 41
received TOP40 DHW_Energy_Production: 5800
received TOP50 Discharge_Temp: 89
Measured Watthour on S0 port 1: 6.00
Measured total Watthour on S0 port 1: 2553.00
Calculated Watt on S0 port 1: 6091
Measured Watthour on S0 port 1: 5.00
Measured total Watthour on S0 port 1: 2558.00
Calculated Watt on S0 port 1: 17647
Requesting new 1wire temperatures
Received 1wire sensor temperature (28fae96a060000d0): 51.44
Received 1wire sensor temperature (28f3cb6a0600003d): 48.88
Heishamon stats: Uptime: 0 days 0 hours 2 minutes 50 seconds ## Free memory: 65% 23560 bytes ## Wifi: 100% ## Mqtt reconnects: 1 ## Correct data: 100.00%
Requesting new panasonic data
sent bytes: 111 including checksum value: 18 
Received 203 bytes data
Checksum and header received ok!
received TOP8 Compressor_Freq: 40
received TOP49 Main_Hex_Outlet_Temp: 58
Measured Watthour on S0 port 1: 3.00
Measured total Watthour on S0 port 1: 2561.00
Calculated Watt on S0 port 1: 2224
Measured Watthour on S0 port 1: 3.00
Measured total Watthour on S0 port 1: 2564.00
Calculated Watt on S0 port 1: 2006
Requesting new 1wire temperatures
Received 1wire sensor temperature (28fae96a060000d0): 51.50
Heishamon stats: Uptime: 0 days 0 hours 3 minutes 0 seconds ## Free memory: 70% 25528 bytes ## Wifi: 100% ## Mqtt reconnects: 1 ## Correct data: 100.00%
Requesting new panasonic data
sent bytes: 111 including checksum value: 18 
Received 203 bytes data
Checksum and header received ok!
received TOP1 Pump_Flow: 15.17
received TOP40 DHW_Energy_Production: 5600
Measured Watthour on S0 port 1: 4.00
Measured total Watthour on S0 port 1: 2568.00
Calculated Watt on S0 port 1: 2010
`

Bucky2k, 2022-09-07:
Cable Aquarea/Heishamon is 8m, extended with ethernet Cat6 shielded. Cable S0 to heishamon is 5m shielded telephone cable. Pulse length measured by precision Multimeter: 78 to 82 ms

Bucky2k, 2022-09-07:
There we have one of the mentioned peaks in 2.0. Are those normal in V2.0 or do they indicate a problem with noise?

[image]

IgorYbema, 2022-09-07:
I'll check if I can add some more debug. Weird it isn't showing anything now. I'll do that tomorrow

Bucky2k, 2022-09-07:
That would be great, thx! However, the 3.0 console shows S0 messages if I trigger the S0 manually by bridgeing the S0 connectors to GND. Due to undefined pulse length with bad pulses / low pulse quality indication. So I guess the extended cable to the aquarea could be the reason. Although it is only 8-max 10m, the cat 7 ethernet cable used for the extension does not follow your recommendation of being at least AWG24... I used a normal patch cable. So your theory of noise/bad power supply of the heishamon remains a candidate

Bucky2k, 2022-09-07:
I added a small elco / tantal to the gnd/5V pins for stabilizing the supply. While it did not solve the problem, I now get some reporting of bad pulses:

Wed Sep  7 23:00:52 2022 (376774): Pulses seen on S0 port 1: Good: 0 Bad: 72 Average good pulse width: 0
Wed Sep  7 23:00:52 2022 (376775): Measured Watthour on S0 port 1: 0.00
Wed Sep  7 23:00:52 2022 (376781): Measured total Watthour on S0 port 1: 3136.00
Wed Sep  7 23:00:52 2022 (376787): Calculated Watt on S0 port 1: 0
Wed Sep  7 23:00:52 2022 (376792): Pulses seen on S0 port 2: Good: 0 Bad: 0 Average good pulse width: 0
Wed Sep  7 23:00:52 2022 (376802): Measured Watthour on S0 port 2: 0.00
Wed Sep  7 23:00:52 2022 (376810): Measured total Watthour on S0 port 2: 399.00
Wed Sep  7 23:00:52 2022 (376815): Calculated Watt on S0 port 2: 0

So I guess before you spend more time, I will exchange the extension wire to the aquarea by an AWG24. Also, I only measure 4,7V supply voltage. Either the Aquarea is not delivering more or I have indeed a voltage drop of 0,3V due to the cable

IgorYbema, 2022-09-08:
Maybe the distance for the 5v supply is too much. You can remove the 5v
supply pin and power the heishamon externally (usb power cable), just
connect it to the gnd/5v of the dallas/temp sensor connector.

Bucky2k, 2022-09-08:
Yes, before I pull out the cable from the pipes I should try this first. However, I would use 5V/GND pins of the programming 10-pin connector. Isn't the Dallas 3.3V regulated? At least your picture on tindie suggests it:

[image]

IgorYbema, 2022-09-08:
Hi,
Old picture :) v3 board.
The newer (also the opentherm version you have) is 5v on the dallas pin.
Should be printed there also. But you can also use the header pins indeed

Bucky2k, 2022-09-08:
Ok, separated the 5V pin from Aquarea, supplied separately via the header pins (tested both power supply + power bank incl. smoothening via elco - even more bad /no good reads. So cable should not be the primary issue... 

Hmm - I guess I just order a replacement board just to be sure it's not hardware related. Shall I use tindie or can you sent me a replacement board directly (open therm) against PayPal payment?

IgorYbema, 2022-09-08:
Pretty sure it is not the board. It is working in v2.

It must be the cable which is delevering too much noise (picking up electrostatic noise). Can you try the grounding trick?

Bucky2k, 2022-09-08:
Nope, not working Aquarea is grounded by PE directly to the e-cabinet. Wired S0 GND to PE but no change. 

A pity I do not have an oscilloscope. Not sure if the meters are pulling the S0 fully down. If I fake impulses by just bridging the S0 pins on the meter, the heishamons starts detecting good impulses

Bucky2k, 2022-09-08:
Maybe I try another type of meters, although I think many are using the DRS255BC. But potentially still all on 2.0

IgorYbema, 2022-09-08:
Can you try to connect the heishamon with a short cable directly to the meter? So power it again with a powerbank. Not connected to the heatpump ofcourse. But just check if it works directly with the meter.

geduxas, 2022-09-08:
Sorry for interruption, but after upgrading to V3, I have noticed that my power meter started to work :D till then I unable to get correct data.. if you remember me @IgorYbema .. also could you clarify some details for S0 configuration? And what should pulse quality be? I have 50-70%

IgorYbema, 2022-09-08:
@Bucky2k so you have reading but all bad? What are the fault messages? (maybe I do need to add more debug for that, too long ago I created that)
@geduxas I have pulse quality 99% :) .. the higher the better. If you have lower, then it is picking up noise.

Bucky2k, 2022-09-08:
The problem is, it's very different depending on the meter attached, if direct supply or from Aquarea, etc. 

With independent 5V supply, no aquarea and only 20cm cable to the meters, I still have poor pulse quality:

Sep  8 14:03:55 2022 (4615660): Pulses seen on S0 port 1: Good: 9 Bad: 75 Average good pulse width: 80

On the other meter, I have more than 10000 bad pulses only after some minutes. 

On the first day of use I choose wrong polarization of the S0, but this should not have hurt the GPIOs I think. But something is heavily unstable, which is why I also question the hardware just to rule this out.

Bucky2k, 2022-09-08:
I ordered the normal PCB on tindie just to be exclude it as a possible cause. And some small revenue for you considering your support time on this 😉

IgorYbema, 2022-09-08:
Noticed the order. I'll ship it tomorrow.

Bucky2k, 2022-09-12:
Hi Igor, any chance you missed flashing the new heishamon board? Received it today, comes up with AT-Thinker-... AP name instead of Heishamon after startup and no webpage accessible under 192.168.4.1...

IgorYbema, 2022-09-13:
Seems like it. Do you have cable to load it yourself?

Bucky2k, 2022-09-15:
End of Thread - it was indeed a hardware issue. New heishamon runs fine

closed 2022-09-15 by Bucky2k