heishamon/HeishaMon #269: change of mqtt sensors in homeassistant 2022.12.0

HerbyR56, 2022-09-08:
I get a warning in homeassistant that manual configured mqtt sensors have a changed format. 
See https://www.home-assistant.io/integrations/sensor.mqtt/#new_format.
**old:**
 sensor:
  - platform: mqtt 

**new:**
mqtt:
  sensor:

I started to change the heishamon.yaml, but I'm not successfull with it. I got the error the the mqtt platforn cannot get merged. Can you support me in changing that yaml file, so it's running with the new versions of homeassistant?

geduxas, 2022-09-08:
Update your yaml, check in repository, couple days it have been updated

HerbyR56, 2022-09-08:
It works, I have only added the Topic 36 and Topic 93, which I found additional interesting

closed 2022-09-08 by HerbyR56