heishamon/HeishaMon #271: V3 serial interface issues

geduxas, 2022-09-11:
I have noticed strange issue with V3. 
Sometimes it's seams Panasonic is locking out from serial communication. I have 2 random issues for now, and both are after rebooting heishamon.
I have V3 PCB, , i think i am using heishamon from v1, and didn't have any issues with it. Couple days ago tried S0 settings, and rebooted heishamon from menu, after that communication with heat pump gone, from console log i got only records about reading timeout, after that tried reboot couple times, also tried factory reset and downgrade firmware (v2, v1) nothing helped. After i came home and power cycled whole heat pump unit, communication have been back. After that tried to reboot heishamon couple of times, and couldn't reproduce.. but yesterday just by accident pushed reboot button in menu, and again,  got read timeout.. 
I have 12kW T-CAP h series model, with additional extension card.

IgorYbema, 2022-09-11:
Do you have CZ-TAW1 in parallel (so heishamon should be in listen only mode)?
Or did you by accident turn on the optional pcb emulation mode? (as you have the extra board already you should not do this)

geduxas, 2022-09-11:
No I don't, didn't touch emulation, and Don't have cz-taw1. Panasonic just stopped responding after restarting heishamon.. i have also tried physically disconnect heishamon while that happened..

IgorYbema, 2022-09-11:
Heatpump still works and the panasonic controller also? Or do they stop also? This would be a new one for me then.

geduxas, 2022-09-11:
Everything seams works fine, truly didn't noticed that, latest issue i noticed after 9h, and dhw didn't cool down..

geduxas, 2022-09-11:
Oh yes! LCD kept responsive, i switch unit off and then i have disconnect circuit breaker

IgorYbema, 2022-09-11:
So everything on Panasonic works fine but heishamon not and you need to turn off and on the Panasonic to fix heishamon again. Weird.

geduxas, 2022-09-11:
It's happened again! I tried reboot 3 times, nothing happened, and in forth time i went in to settings menu, saved configuration, and after that hit reboot.

[image]

IgorYbema, 2022-09-11:
Can't understand why. If a hard reset of the heatpump is necessary to fix this it looks like an issue on the heatpump side. But proper debugging is hard for this

geduxas, 2022-09-11:
I am running heishamon >2y, didn't have any issues till this thursday. 

It's seams that Panasonic lookups serial interface on reboot.. and yes only hard reset of pump solves problem

IgorYbema, 2022-09-11:
Older heishamons can have easier comms problems than the most recent design (electrical design difference on the board). Maybe you are on an edge case and due to other external reason (extra noise?) it is now only visible.
If you can reproduce it often and it really bothers you, I can send you a new one and check if it works better with the new design. At least if you got the heishamon from me (the tindie shop), then what was your order number?

geduxas, 2022-09-11:
Yes, bought cable + board. Rebooted once again without changing anything, and it's lost comuncation again..

geduxas, 2022-09-11:
Found order no in email

Tindie Order [#207935]
Aug. 5, 2020, 1:37 p.m.

Also i think I'll try downloading to v2 and try

IgorYbema, 2022-09-11:
Ok just let me know if you want me to send you a new one to test

geduxas, 2022-09-11:
Downgraded to v2, after factory reset and connect to wifi phone finds strange AP.. and it's dissapperes then i disconnect heishamon.. hashemon seams works fine.. do i need somhow special clean procedures of esp?

[image]

IgorYbema, 2022-09-11:
Do you have special wifi characters in your ssid or password? Seeing this with others also. A upgrade to V3 solves it. But try factory reset please. Double click on reset with 0.5 sec pause until blue led goes in steady on/off. Then reset again to start

geduxas, 2022-09-11:
Nothing special..  ok I'll stick to v2 for couple days, now reboot'ed 4 times no issues with serial.

MiG-41, 2022-09-12:
> I have 12kW T-CAP h series model, with additional extension card.

I belive her is the key , that there is CZ-NS4P in beetwen.
@geduxas ,by accident did you have enabled "Emulate optional PCB" ?

geduxas, 2022-09-12:
No I didn't enable emulation. With V2 seams no issues after reboot, will continue to test.

geduxas, 2022-09-26:
@IgorYbema still no issues with V2... could it be related to V3? or it could be board issues?

mabek, 2022-09-26:
I also have heavy issues with V3 but its hard to specify. 
Overall the UI seems slow. Sometimes values do not update at all. Without factory reset no values were shown. So I think I will go back to V2 as well.

geduxas, 2022-09-26:
> I also have heavy issues with V3 but its hard to specify. Overall the UI seems slow. Sometimes values do not update at all. Without factory reset no values were shown. So I think I will go back to V2 as well.

on my case interface works ok, just after reboot panasonic lock out from serial communication.. 

maybe you have Wifi issues? do you get good reception?

mabek, 2022-09-26:
> on my case interface works ok, just after reboot panasonic lock out from serial communication..
>
> maybe you have Wifi issues? do you get good reception?

Wifi is at 100%. Can you describe what you mean with "reboot panasonic lock out from serial communication" ?

geduxas, 2022-09-26:
> Wifi is at 100%. Can you describe what you mean with "reboot panasonic lock out from serial communication" ?

After rebooting heishamon, Panasonic heatpump stops responding to request. Disconnect heishamon Don't help.. i needed to powercycle whole heatpump unit..

mabek, 2022-09-26:
> After rebooting heishamon, Panasonic heatpump stops responding to request. Disconnect heishamon Don't help.. i needed to powercycle whole heatpump unit..

Thank you! After turning the power off and on again now it is as follows:

MQTT Topics working again except Heat_Energy_Production which is now always 0
Webinterface does not show any values on the main page. 1-Wire and s0 works
Webinterface still 2-3x slower than with v2

mabek, 2022-09-28:
I went back to V2 and now all my Issues are gone. Hopefully someone finds a fix for a V3.1

drschrick12, 2022-12-03:
Dears, I’ve exactly the same issue.
My board is v3 PCB and only firmware v2 is function correctly.

In firmware v3 the webinterface is very laggy and slow and over mqtt it shows online and short time later offline and then again online.

Tindie order #192596 from May 2nd, 2020.

Thank you.

geduxas, 2022-12-03:
> Dears, I’ve exactly the same issue. My board is v3 PCB and only firmware v2 is function correctly.
> 
> In firmware v3 the webinterface is very laggy and slow and over mqtt it shows online and short time later offline and then again online.
> 
> Tindie order #192596 from May 2nd, 2020.
> 
> Thank you.

How strong is your wifi? What numbers do you see in heishamon?

drschrick12, 2022-12-04:
My wifi is 76%

geduxas, 2022-12-12:
I am here again.. everything seams to work great for couple of months. Today after couple reboots heishamon stucked again. 

Mon Dec 12 10:54:22 2022 (1428540): Heishamon stats: Uptime: 0 days 0 hours 23 minutes 48 seconds ## Free memory: 70% ## Heap fragmentation: 23% ## Max free block: 21112 bytes ## Free heap: 28016 bytes ## Wifi: 92% (RSSI: -54) ## Mqtt reconnects: 1 ## Correct data: 0.00%
Mon Dec 12 10:54:22 2022 (1428544): Requesting new panasonic data
Mon Dec 12 10:54:22 2022 (1428544): sent bytes: 111 including checksum value: 18 
Mon Dec 12 10:54:24 2022 (1430547): Previous read data attempt failed due to timeout!
Mon Dec 12 10:54:24 2022 (1430547): Received 0 bytes data

closed 2024-09-12 by Egyras