heishamon/HeishaMon #274: HeatpumpType for WH-MXC09J3E8 is missing

RBogie, 2022-09-19:
I don't know where the types are defined in the code exactly, so instead of opening an MR, I'm opening this issue instead.

I have a WH-MXC09J3E8 heatpump (monoblock, 3 phase 9kw T-CAP J series), which currently reports 255 as `Heat_Pump_Model`. I assume that 255 means that the model is unknown.

Below is a hexdump of a packet that the heishamon received from the heatpump:
```
Received 203 bytes data
data: 71 C8 01 10 56 55 52 49 00 59 00 01 00 00 00 00 00 00 00 00 5A 15 12 55 56 16 55 56 5A 1A 00 00 
data: 00 00 00 00 00 00 80 80 99 8A B4 71 71 97 99 00 00 00 00 00 00 00 00 00 00 00 80 85 5B 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D AD AD 7B 8F 99 A3 7B 8F 98 85 80 8A 8F 96 9E 8A 8A 94 9E 85 96 
data: 94 11 79 78 BE 15 7E 7C 1F 7C 7E 00 00 00 55 65 55 21 85 15 59 05 0C 13 69 01 00 00 00 00 00 00 
data: 00 32 D4 0B 87 84 73 90 0C 84 84 A5 00 B2 90 A2 A5 94 32 AD 99 32 32 32 80 AD A4 AB 93 9C 8F 96 
data: 9B 9D 9D 3D 01 01 01 00 00 22 00 01 01 01 01 79 79 01 01 70 00 00 83 00 00 01 00 00 1A 00 00 0A 
data: 02 01 01 01 01 01 01 02 00 00 E8 
```

I think this means that the unique code for this model is `00 32 D4 0B 87 84 73 90 0C 84`. Could this be added to HeishaMon?

IgorYbema, 2022-09-19:
It is in decode.h 
https://github.com/IgorYbema/HeishaMon/blob/main/HeishaMon/decode.h#L34
and
https://github.com/IgorYbema/HeishaMon/blob/main/HeishaMon/decode.h#L61

You can try to create a PR if you like :) Or just let us do it

mi-hol, 2024-09-06:
should be closed

closed 2024-09-12 by Egyras