heishamon/HeishaMon #276: Energy Meter reset / configure signal

Revo7, 2022-09-30:
Hello, i want to ask, how to reset / delete the meter value ? from S0_1 and S0_2 sensor ? Hope it is posible ?

And i don't know but i think it's not right measure, i don't know how to configure it. I have WZE-3 energy meter. There are possible to connect Signal. 

Pulse output SO + SO- open collector
Connection voltage SO + SO- max. 27 V DC
Connection current SO + SO- max. 27 mA
Pulse time SO + SO- 35 ms
Pulse: (1 Wh/imp) 1000 imp/kWh

And in the HeishaMon settings:
[image]
and i get PulseQuality very poor, and the Watt meter are abstract:
[image]

What i make wrong ? :/

Bender2k, 2022-11-18:
Hi,

i have no answer but the same question.

I installed Heishmon later than my Jeisha und therefore the energy meter already counted some hundret kWh.
I want to update the values in Heishamon so the MQTT value for the total consumption is correct.

Are there (MQTT?) commands i can send to Heishamon to change the values? Or is something like a Putty connection possible? Or an update for the Console so its possible to enter simple commands?

IgorYbema, 2022-11-18:
Heishamon picks up the last value sent to mqtt after reboot. So to reset them, turn off heishamon and send a new value to the mqtt topic (for example use mqtt explorer). Then start up heishamon, it will then pick up this new value as a start value.

Bender2k, 2022-11-18:
Oh wow. This means the Total Watthour is not saved on the Heishamon PCB?

Therefore when the MQTT connection is interruppted, even only some minutes per month, the total Watthour will be wrong again according to the energy meter?

Your project and your work ist FANTASTIC, but if this is like i understood, this feature is not very reliable. :(

MiG-41, 2022-11-18:
It is storaging in ESP memory , but if in mqtt is value ,it uses this one as actual.

Bender2k, 2022-11-18:
Yes i understood, but one example:

If the MQTT wifi connection is interrupted at TotalWatthour 3000, the Heishamon board continues counting to 3100.
After that there is a power interruption to the Heishamon PCB and everthing is connecting again, the Heishamon Value will be reset from 3100 to 3000?

IgorYbema, 2022-11-18:
Indeed, it stores in ESP flash also. But then again, it can never be always correct as it has only S0 to count pulses. It is more an indication about how much Watthour it uses. Not a high reliable solution. That is also not what it is designed for. It should use modbus then for that.
You can always sync the kWm meter value with the internal heishamon value (using the mqtt trick). But as said, it can't be correct always.

Bender2k, 2022-11-18:
Got you! Thanks.

I just dont trust Wifi and for me Wired signals (Even if it is only S0) is ten times more reliable for me.

I think i can work with your proposal and will test it!

But just take it as a improvement suggestion to make the value editable on the web interface instead of MQTT. ;)

Revo7, 2022-11-18:
Ok, so we can change or reset the value, but what with signal quality ? Could you show your config and what kind of device you use ?

closed 2024-09-12 by Egyras