heishamon/HeishaMon #277: Build in Platformio

mocholor, 2022-10-03:
Hey guys!
Struggling to build this GREAT project in Platformio.
platformio.ini:
[env:d1_mini]
platform = espressif8266

board = d1_mini
framework = arduino
lib_deps = 
	locoduino/RingBuffer@^1.0.3
	knolleary/PubSubClient@^2.8
	jenscski/DoubleResetDetect@^1.0.0
	bblanchon/ArduinoJson@6.18.3
	milesburton/DallasTemperature@3.9.0
	paulstoffregen/OneWire@2.3.5
	links2004/WebSockets@2.3.5

But get nasty 'MMU_SEC_HEAP' was not declared in this scope and other non declared MMU stuff.
What am I missing here?

gregor-samosir, 2022-10-05:
Hi,

this is the main part I am using on pio:

`[platformio]
src_dir = src
build_dir = .pioenvs
build_cache_dir = .cache
default_envs = d1_mini_h1_ota

[env:d1_mini_h1_ota]
platform = espressif8266@^2
board = d1_mini
framework = arduino
build_flags = ${esp82xx_defaults.build_flags}
build_unflags = -Wall
monitor_speed = 115200
check_tool = clangtidy
check_flags = clangtidy: --checks=-*,cert-*,clang-analyzer-* --fix
board_build.filesystem = littlefs
board_build.flash_mode = dout
lib_deps = 
*put in your libs*

[esp82xx_defaults]
build_flags = 
	-D NDEBUG
	-Wl,-Map,firmware.map
	-D SERIAL_BUFFER_SIZE=256
`

mocholor, 2022-10-05:
Still:
src\rules.cpp:78:43: error: 'MMU_SEC_HEAP' was not declared in this scope
   78 | unsigned char *mempool = (unsigned char *)MMU_SEC_HEAP;

IgorYbema, 2022-10-05:
I don't use platformio but you need the mmu=3216 build flag. I understand in pio it should be:
build_flags=-DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x4000 -DMMU_SEC_HEAP=0x40108000 -DMMU_SEC_HEAP_SIZE=0x4000

mocholor, 2022-10-05:
yeeeey. That did the trick. Got a gazillion of warnings, but finally =====SUCCESS=====
Thank you!
Had also to force espressiv8266 to 4

Final platformio.ini ist here
[env:d1_mini]
platform = espressif8266@^4

board = d1_mini
framework = arduino

build_flags=-DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x4000 -DMMU_SEC_HEAP=0x40108000 -DMMU_SEC_HEAP_SIZE=0x4000

lib_deps = 
	locoduino/RingBuffer@^1.0.3
	knolleary/PubSubClient@^2.8
	jenscski/DoubleResetDetect@^1.0.0
	bblanchon/ArduinoJson@6.18.3
	milesburton/DallasTemperature@3.9.0
	paulstoffregen/OneWire@2.3.5
	links2004/WebSockets@2.3.5

closed 2022-10-05 by mocholor

IgorYbema, 2022-10-11:
Hi,
Check the github build yaml code:
https://github.com/IgorYbema/HeishaMon/blob/main/.github/workflows/main.yml
You need to enable some extra compile options. You are missing mmu=3216 and
eesz=4M2M if I am not mistaken. Not sure how platformio calls them. But it
is necessary to point some code to extra memory space.