heishamon/HeishaMon #279: Serious problem with flickering of the values with integration with HA

michalwoz, 2022-10-07:
I experienced serious issue with integration with HA on version 3.0.
I'm running default HA configuration. Tried to change heat mode and target temp. The problem is fast switching (once per second) of the value after trying to set the value. Haven't confirmed yet, but problem may occur only when value is changed fast from HA GUI. Example:

1. wanted to change target temp from 20 to 25
2. you missclicked and dropped value at 24
3. you corrected and set 25 - here the flickering begins:

There is 0 seconds or 1 second between points on X axis:
[image]

I'm attaching console log. As we can see we receive "set z1 heat requests" HeishaMon may actually do what it is requested for. And the problem may be HA configuration.

The same situation with setting heat mode. Actually I noticed the problem with that, when I heard 1 per second relay clicking in heat pump. I consider the issue dangerous because of 2 things: exhausting HP memory write cycles limit and frequent powering on/off hp or its components.

> Fri Oct  7 08:53:34 2022 (1278532): set z1 heat request temperature to 25
Fri Oct  7 08:53:34 2022 (1278533): sent bytes: 111 including checksum value: 249 
Fri Oct  7 08:53:35 2022 (1278959): Received 203 bytes data
Fri Oct  7 08:53:35 2022 (1278960): Checksum and header received ok!
Fri Oct  7 08:53:35 2022 (1278962): received TOP7 Main_Target_Temp: 24
Fri Oct  7 08:53:35 2022 (1278967): received TOP27 Z1_Heat_Request_Temp: 24
Fri Oct  7 08:53:35 2022 (1278971): received TOP42 Z1_Water_Target_Temp: 24
Fri Oct  7 08:53:35 2022 (1279051): set z1 heat request temperature to 24
Fri Oct  7 08:53:35 2022 (1279052): sent bytes: 111 including checksum value: 250 
Fri Oct  7 08:53:35 2022 (1279558): Received 203 bytes data
Fri Oct  7 08:53:35 2022 (1279559): Checksum and header received ok!
Fri Oct  7 08:53:35 2022 (1279561): received TOP7 Main_Target_Temp: 25
Fri Oct  7 08:53:35 2022 (1279566): received TOP27 Z1_Heat_Request_Temp: 25
Fri Oct  7 08:53:35 2022 (1279570): received TOP42 Z1_Water_Target_Temp: 25
Fri Oct  7 08:53:36 2022 (1280133): set z1 heat request temperature to 25
Fri Oct  7 08:53:36 2022 (1280134): sent bytes: 111 including checksum value: 249 
Fri Oct  7 08:53:36 2022 (1280307): Requesting new 1wire temperatures
Fri Oct  7 08:53:37 2022 (1281024): Received 203 bytes data
Fri Oct  7 08:53:37 2022 (1281024): Checksum and header received ok!
Fri Oct  7 08:53:37 2022 (1281027): received TOP7 Main_Target_Temp: 24
Fri Oct  7 08:53:37 2022 (1281032): received TOP27 Z1_Heat_Request_Temp: 24
Fri Oct  7 08:53:37 2022 (1281035): received TOP42 Z1_Water_Target_Temp: 24
Fri Oct  7 08:53:37 2022 (1281040): received TOP64 High_Pressure: 10.0
Fri Oct  7 08:53:37 2022 (1281089): set z1 heat request temperature to 24
Fri Oct  7 08:53:37 2022 (1281090): sent bytes: 111 including checksum value: 250 
Fri Oct  7 08:53:37 2022 (1281555): Received 203 bytes data
Fri Oct  7 08:53:37 2022 (1281556): Checksum and header received ok!
Fri Oct  7 08:53:37 2022 (1281558): received TOP7 Main_Target_Temp: 25
Fri Oct  7 08:53:37 2022 (1281563): received TOP27 Z1_Heat_Request_Temp: 25
Fri Oct  7 08:53:37 2022 (1281567): received TOP42 Z1_Water_Target_Temp: 25
Fri Oct  7 08:53:38 2022 (1281656): set z1 heat request temperature to 25
Fri Oct  7 08:53:38 2022 (1281657): sent bytes: 111 including checksum value: 249 
Fri Oct  7 08:53:38 2022 (1282154): Received 203 bytes data
Fri Oct  7 08:53:38 2022 (1282155): Checksum and header received ok!
Fri Oct  7 08:53:38 2022 (1282157): received TOP7 Main_Target_Temp: 24
Fri Oct  7 08:53:38 2022 (1282163): received TOP27 Z1_Heat_Request_Temp: 24
Fri Oct  7 08:53:38 2022 (1282166): received TOP42 Z1_Water_Target_Temp: 24
Fri Oct  7 08:53:38 2022 (1282171): received TOP64 High_Pressure: 10.2
Fri Oct  7 08:53:38 2022 (1282226): set z1 heat request temperature to 24
Fri Oct  7 08:53:38 2022 (1282227): sent bytes: 111 including checksum value: 250 
Fri Oct  7 08:53:39 2022 (1282653): Received 203 bytes data
Fri Oct  7 08:53:39 2022 (1282654): Checksum and header received ok!
Fri Oct  7 08:53:39 2022 (1282656): received TOP7 Main_Target_Temp: 25
Fri Oct  7 08:53:39 2022 (1282661): received TOP27 Z1_Heat_Request_Temp: 25
Fri Oct  7 08:53:39 2022 (1282665): received TOP42 Z1_Water_Target_Temp: 25
Fri Oct  7 08:53:39 2022 (1282722): set z1 heat request temperature to 25
Fri Oct  7 08:53:39 2022 (1282723): sent bytes: 111 including checksum value: 249 
Fri Oct  7 08:53:39 2022 (1283153): Received 203 bytes data
Fri Oct  7 08:53:39 2022 (1283153): Checksum and header received ok!
Fri Oct  7 08:53:39 2022 (1283156): received TOP7 Main_Target_Temp: 24
Fri Oct  7 08:53:39 2022 (1283161): received TOP27 Z1_Heat_Request_Temp: 24
Fri Oct  7 08:53:39 2022 (1283164): received TOP42 Z1_Water_Target_Temp: 24
Fri Oct  7 08:53:39 2022 (1283251): set z1 heat request temperature to 24
Fri Oct  7 08:53:39 2022 (1283251): sent bytes: 111 including checksum value: 250 
Fri Oct  7 08:53:40 2022 (1283752): Received 203 bytes data
Fri Oct  7 08:53:40 2022 (1283753): Checksum and header received ok!
Fri Oct  7 08:53:40 2022 (1283755): received TOP7 Main_Target_Temp: 25
Fri Oct  7 08:53:40 2022 (1283760): received TOP27 Z1_Heat_Request_Temp: 25
Fri Oct  7 08:53:40 2022 (1283764): received TOP42 Z1_Water_Target_Temp: 25

geduxas, 2022-10-07:
Hi, it's not related to V3.. i have same issues and from first versions.. more it is related to HA itself.. or maybe with automation rules.

michalwoz, 2022-10-07:
@geduxas No relation to HeishaMon version is with that is what I expected. Is it possible that HeishaMon sends messages with topic `panasonic_heat_pump/commands/SetZ1HeatRequestTemperature`? Or this topic may be sent only from HA?

geduxas, 2022-10-07:
Automation scripts are in heishamon.yaml already.. 
they purpose to listen of heishamon heat shift, and change change heat shift selector in homeassistant side.. also another trigger  is watching homeassistant slide, and triggers topic to change heat shift.. and in rarely cases it will chats 2 values and starts to flickering.. personally I don't like this kind of implementation, and I have setup mqtt climate component for that.. 

 

> @geduxas No relation to HeishaMon version is with that is what I expected. Is it possible that HeishaMon sends messages with topic `panasonic_heat_pump/commands/SetZ1HeatRequestTemperature`? Or this topic may be sent only from HA?
> 
> I don't have any automations yet.

PGnl, 2022-10-07:
@geduxas 
I am also investigating the mqqt climate component at the moment (just started) as I do not like the slider. Could you share your climate integration?

michalwoz, 2022-10-07:
Changes made in GUI of HA with interval of ~30+ seconds seems to be OK. Maybe 15+or even 10+ seconds is also OK.

BTW my config:
[image]

geduxas, 2022-10-07:
> @geduxas I am also investigating the mqqt climate component at the moment (just started) as I do not like the slider. Could you share your climate integration?

For now it's need to fully rewrite because of mqtt changes in HA.. i am still not done that.. if i will finish it i will post it here

PGnl, 2022-10-07:
If you can share, I can see if I can find the time to update to the new mqqt format. 
example typically saves quite some time.
As I was planning to do so anyway, we might be able to help each other.

geduxas, 2022-10-07:
@PGnl here is my config:

```yaml
climate:
#HeatShift
- platform: mqtt
  name: HeatPumpSetpoint
  initial: 0
  min_temp: -5
  max_temp: 5
  action_topic: "panasonic_heat_pump/main/ThreeWay_Valve_State"
  action_template: >-
    {% set values = { "0":"heating", "1":"idle" } %}
    {{ values[value] if value in values.keys() else "idle"}}
  mode_state_topic: "panasonic_heat_pump/main/Operating_Mode_State"
  mode_state_template: >-
    {% set values = { "0":"heat", "1":"cool", "2":"auto", "4":"heat", "5":"cool", "6":"auto", "7":"auto", "8":"auto" } %}
    {{ values[value] if value in values.keys() else "off"}}
  modes:
    - auto
    - heat
    - cool
    - "off"
  temperature_command_topic: "panasonic_heat_pump/commands/SetZ1HeatRequestTemperature"
  temperature_state_topic: "panasonic_heat_pump/main/Z1_Heat_Request_Temp"
  current_temperature_topic: "panasonic_heat_pump/main/Main_Outlet_Temp"
  aux_state_topic: "panasonic_heat_pump/main/Room_Heater_State"
  precision: 1.0

#DHWShift
- platform: mqtt
  name: DHWSetPoint
  initial: 50
  min_temp: 40
  max_temp: 75
  action_topic: "panasonic_heat_pump/main/ThreeWay_Valve_State"
  action_template: >-
    {% set values = { "0":"idle", "1": "heating"} %}
     {{ values[value] if value in values.keys() else "idle"}}
  mode_state_topic: "panasonic_heat_pump/main/Operating_Mode_State"
  mode_state_template: >-
    {% set values = { "3":"heat", "4":"heat", "5":"heat", "6":"heat", "8":"heat" } %}
    {{ values[value] if value in values.keys() else "off"}}
  modes:
    - heat
    - 'off'
  aux_state_topic: "panasonic_heat_pump/main/DHW_Heater_State"
  temperature_command_topic: "panasonic_heat_pump/commands/SetDHWTemp"
  temperature_state_topic: "panasonic_heat_pump/main/DHW_Target_Temp"
  current_temperature_topic: "panasonic_heat_pump/main/DHW_Temp"
  precision: 1.0
```

PGnl, 2022-10-07:
@geduxas 

I did the heat part. Seeems to be correct and works for me. Still need to find a way to have it work for me for 2 zones. But that is for later and more for me, as I seem to be one of the few to use 2 zones. 
Believe I will eventually take out the mode selector in zone 2 and use an additional switch to adjust for zone1, zone 2 or zone 1 and 2.

What I did not get, is why you included       aux_state_topic: "panasonic_heat_pump/main/Room_Heater_State".
This topic is linked to the backup heater and also cannot be controlled via the thermostat, nor is it used.
Doen't seem to interfere with the functionality though.

Haven't done DHW (I don't have it, but should be equal setup)

`mqtt:
  climate:
    - name: HeatPumpSetpoint
      modes:
        - "auto"
        - "heat"
        - "cool"
        - "off"
      mode_state_topic: "panasonic_heat_pump/main/Operating_Mode_State"
      mode_state_template: >
        {% set values = { "0":"heat", "1":"cool", "2":"auto", "4":"heat", "5":"cool", "6":"auto", "7":"auto", "8":"auto" } %}
        {{ values[value] if value in values.keys() else "off"}}
      initial: 0
      min_temp: -5
      max_temp: 5
      temperature_command_topic: "panasonic_heat_pump/commands/SetZ1HeatRequestTemperature"
      temperature_state_topic: "panasonic_heat_pump/main/Z1_Heat_Request_Temp"
      current_temperature_topic: "panasonic_heat_pump/main/Main_Outlet_Temp"
      aux_state_topic: "panasonic_heat_pump/main/Room_Heater_State"
      precision: 1.0
`
For anyone using a thermostat (like me), it becomes:

`mqtt:
  climate:
    - name: Zone 1 Panasonic
      modes:
        - "auto"
        - "heat"
        - "cool"
        - "off"
      mode_state_topic: "panasonic_heat_pump/main/Operating_Mode_State"
      mode_state_template: >
        {% set values = { "0":"heat", "1":"cool", "2":"auto", "4":"heat", "5":"cool", "6":"auto", "7":"auto", "8":"auto" } %}
        {{ values[value] if value in values.keys() else "off"}}
      initial: 18
      min_temp: 7
      max_temp: 30
      temperature_command_topic: "panasonic_heat_pump/commands/SetZ1HeatRequestTemperature"
      temperature_state_topic: "panasonic_heat_pump/main/Z1_Heat_Request_Temp"
      current_temperature_topic: "panasonic_heat_pump/main/Z1_Temp"
      aux_state_topic: "panasonic_heat_pump/main/Room_Heater_State"
      precision: 1.0`

[image]

MiG-41, 2022-10-08:
Regarding the topic , it is becouse you change SetCommand too fast :) 
When ou change once , and then immidietly change to another set second time , HA will be not able to get confirmation from HP ( from Heishamon via mqtt and another topic). 
So i.e. you send one b one temperature set to 20 and 25 , then neverending loop will be created :)
(and this is becouse of HA , other systems not resend comands ,if confirmation is other then set).

What i made ,is ,beore action mqtt.publish i add delay 3s (in this way i have time to correct value , if i set it wrongly) , so something like this:
`  action:
  - delay: '2'
  - service: mqtt.publish`

But to do that i needed to cut this whole action section from heishamon,yaml  and move it to automation.yaml file. (It was 2 years ago ,so perhaps now it is possible ).

geduxas, 2022-10-08:
@PGnl for aux heat, i just wanted to see when backup heater kick in :) just for information purpose.. i have done with 2 zones too..

geduxas, 2022-10-09:
@PGnl your idies for idle.. seams it works fine this is for heating, for dhw need to switch heating to idle.. 

```yaml
        {% set values = { "0":"heating", "1":"idle" } %}
        {% if states('sensor.aquarea_compressor_frequency') | float > 0 and  is_state('sensor.aquarea_defrost_state', 'Inactive') %}
           {{ values[value] if value in values.keys() else "idle"}}
        {% else %}
           {{ 'idle' }}
        {% endif %}
```

PGnl, 2022-10-10:
@geduxas 
That is a clean way of doing it indeed.
My issue is/was that I have 2 zones and the pana does not tell me, at least I haven’t seen it, which of the zones it is heating or even both.
So I am using the outlet temp of each zone in combination with frequency now.
Then I can switch pump and fans based on that as well. This way this automatically gives me a delay, which enables a form of slow(er) startup. 
anyway, thanks for the thinking along.

michalwoz, 2022-11-17:
For many days of observation, I can 100% confirm, that this problem occurs NOT only after fast changes of values via HA interface. You can click only once, and it is enough to trigger constant toggling of value.

geduxas, 2022-11-17:
> For many days of observation, I can 100% confirm, that this problem occurs NOT only after fast changes of values via HA interface. You can click only once, and it is enough to trigger constant toggling of value.

Yes, it exists, ant it is annoyng.. but it's depends to "refresh" time.. 

Also i suggest you to switch in to heishamon integration, instead of manual MQTT

https://github.com/kamaradclimber/heishamon-homeassistant

@kamaradclimber made great integration, and everything works amazing!

MiG-41, 2022-11-17:
Did you tried with delay ? 
After puting this i forgot that such flickering exist...

bluberly, 2022-12-01:
I can reproduce the fast flipping of values for heat curve office, DWH temperature but even whisper mode as well in Home Assistant. It first happened when I tried to change the heat curve offset on my smartphone. As the slider is small it was a bit forth an back and the it started toggling very quickly. For testing a counter I implemented, I quickly changed the whisper mode and the same happened. Only HA restart can stop it. Just reloading the Template/MQTT entities in HA did not help.

There are a couple of ideas above. What is a recommended/safe change to the current HA integration code?

geduxas, 2022-12-01:
@bluberly i would recommend you to switch into integration mode :) please see earlier comment's.. yes flapping problems are with automation, and HA refreshing delay...

michalwoz, 2022-12-01:
@bluberly by whisper mode you mean quiet mode? You named it from icon?

I confirmed, that you don't even need to change the value quickly. It is ALSO enough to change any value once over long time period. Just my guess: it happens when you change the value in the moment very close to messages received from HP.
I will apply @geduxas advice to switch into integration mode, when I find some time.
If I have a vote I will vote to change that default HA yaml. From the one side all of us have quite a lot of technical knowledge, but on the other side it is very easy to use HeishaMon with HA, so some people with less experience can use it. I think that it is very positive and we all wish more HeishaMon users. I think that it will be better to protect their heat pumps.

kamaradclimber, 2022-12-02:
TBH, I'm not sure the integration I wrote solves the flickering issue at the moment. There is inherently a race condition when updating a setting and receiving the "latest" value from MQTT.
Since this is real code however, we have more options to tackle this.

bluberly, 2022-12-02:
@michalwoz yes, I meant the quiet mode.
@geduxas I spent quite some effort to localize the heishamon HA code in this repository, changed entity names and created quite some derived entities, so I do not really want to use the integration right now. And as @kamaradclimber says, it's not fully clear if the new code addresses the race condition.
I wondering why it should not be possible to address the root cause of this flickering by modifying the HA code for sensors/entities? Or do you see missing functionality on HA templating which only can be done with Python code of an integration?

geduxas, 2022-12-02:
> @michalwoz yes, I meant the quiet mode. @geduxas I spent quite some effort to localize the heishamon HA code in this repository, changed entity names and created quite some derived entities, so I do [...]

No its possible to work around and with manual integration.. Just i saying that i have same problems till not switched to integration component, for now it seams everything works perfect.. also you could use in dual mode.. all automation switch to integration and leave with custom configuration in manual... 

you could fix it but need to calculate timings..
how often hieshamon updates values and retrasmits then? It could be set manualy in heishamon settings. What you need to step away is to not get in loop mode.. 

then you make switch in HA, it triggers automation rule, which will send operation command to heishamon, in that time heishamon could update/retransmitt current value, so HA could set and get value in same time... if i remember correctly get/change value is another automation script.. and just happens that one automation rule tries to set value (change component state in HA) but another automation rule get old retransmitted value and set's back it.. which will trigger another change value automation.. and while that will be sent, another component will get updated value from previous set.. to exit out of loop you just need to swicht off automation for temprary..  and thats it.. 

don't know maybe its possible with automation to block another automation work while changes are not accepted.. just i lived with that and now switched to integration version, which helpfully don't get this behaviour.. 

so in my opinion it would be best to add all effort and contribution to integration, and make it perfect tool :) as @michalwoz told it would be easier tool for less experienced users.. 

P.S. could you share your modifications? maybe we could find something interesting there?

bluberly, 2022-12-02:
Well, my changes to the original code are not worth to share, I added unique_id's and localized them to German language. The work went into derived values, such as COP calculations for heating and DHW using Shelly 3EM data (and doing a "total COP" dividing all produced energy by the total energy consumed, correcting the produced energy (minus 15%, used the power curves from the service manual at different temperatures to get a rough approximation), creating counters for deice and on/off operations (and created a daily counter from it), doing integrations of consumed and produced energy.

What you write above makes sense, I'm just far too newbie to turn this into HA code :-). The HA documentation from one side is good, on the other side I find it often hard, sometimes confusing to understand.

geduxas, 2022-12-02:
> Well, my changes to the original code are not worth to share, I added unique_id's and localized them to German language. The work went into derived values, such as COP calculations for heating and [...]

Well done, i am little bit unsure if calculating power from external device makes sense.. but also don't know how exactly panasonic calculates it's power values..  i think it count's only compressor work, and counts how much heat taken from hex.. also your mentioned counters would be nice to have also.. what do you think @kamaradclimber ?

kamaradclimber, 2022-12-02:
I'm measuring the power consumption from an external point of view as well to have an completely trustable value.
But I don't use it for COP computation though.

geduxas, 2022-12-02:
> I'm measuring the power consumption from an external point of view as well to have an completely trustable value. But I don't use it for COP computation though.

Yes for total measurements it's ok :) but for cop mesurements it's different storry..

I just mentioned you that it would be nice to have counters which mentioned @bluberly .. about start/stop count.. defrost count.. also energy consumption :) i thought i could make it.. but didn't fell so good to implement it :)

Oekn5w, 2022-12-28:
Coming back to the flickering issue, while I didn't check the custom integration, I believe to have resolved this by ignoring mqtt topic updates immediately after the commands have been sent.

This was only present in the input_number stuff (DHW Target, Z1 and Z2 Delta), not the mode or boolean states.

From what I've seen in the MQTT Explorer after you send the command, the corresponding topic get immediately updated with the old value. This triggered the input_number to revert back to the previous temperature, retriggering the command to be updated (now with the old value). Then after 1-2 seconds the HP "confirms" the updated value and sends to new value to Heishamon.

From what I've seen in the source, there is no explicit readback of the affected values after the command, so this must be something the HP sends (I didn't dive that deep into the comm part).

As I had transitioned the Heishamon related automations to Appdaemon to not have them flood the Homeassistant logbook, I fixed the flickering by saving the timestamp of the command and prevent updates of its corresponding input_number until 1.5 seconds have passed.

michalwoz, 2022-12-28:
> This was only present in the input_number stuff (DHW Target, Z1 and Z2 Delta), not the mode or boolean states.

I'm 100% sure that it affects mode too.

michalwoz, 2022-12-28:
@Oekn5w 
> This was only present in the input_number stuff (DHW Target, Z1 and Z2 Delta), not the mode or boolean states.

I'm 100% sure that it affects mode too.

Oekn5w, 2022-12-28:
@michalwoz should have been more clear - quiet and powerful mode are not affected, we do not change the general mode (yet)

geduxas, 2022-12-28:
Nice work @Oekn5w :) also as i remember, once i got flickering on HP operation button.. which turns on/off heatpump. I think there will be problems in all automation rules.

ankl00, 2023-01-01:
I think I have something similar happening with my setup. I change the quiet mode once via automation and after that there is never ending loop (or until HA restart):

Set quiet level triggered by state of Set Quiet Mode
05:59:59 - 7 hours ago
Set quiet level triggered by state of Set Quiet Mode
05:59:58 - 7 hours ago
Set quiet level triggered by state of Set Quiet Mode
05:59:58 - 7 hours ago
Set quiet level triggered by state of Set Quiet Mode
05:59:57 - 7 hours ago
Set quiet level triggered by state of Set Quiet Mode
05:59:56 - 7 hours ago

I do not have issue with Z1 setting which is also updated via my automation.

What is the latest recommended solution for this issue (novise user)? Switch to Kamarad integration? Add some delay somewhere? Stop using automated quiet mode setting?

geduxas, 2023-01-01:
For me is best solution to switch for integration mode.. and continue development in there..

geduxas, 2023-01-01:
Integration is the way how in HomeAssistant should work devices.. manual implementation is like temporary solution, or add on if something is missing in integration. But for first starters, or in low of understanding, best choice is integration.

closed 2024-09-12 by Egyras