heishamon/HeishaMon #280: Possibility for viewing state of waterpump Z1/Z2

PGnl, 2022-10-16:
I have an actual optional PCB installed for dual zone.
Now, I would like to view which of the zones the pana is heating at any given time.
As my mixing valve is controlled by a relay, I needed constant power, which I have taken from the pump power from zone 1 and 2 that normally controls the optional waterpump.
While this has worked perfectly, I would like to automate some switches more directly via node red.
For the emulated PCB the following  topic exists:
OPT0 | optional/Z1_Water_Pump | Z1 water pump action request (0=off, 1=on)
same for zone 2.
Is there a possibility for a topic that shows the actual state of the pumps? Such that we can make it a read topic?

regards,

Paul

geduxas, 2022-10-16:
Panasonic heats all zones together.. just controls temperature via mixing valves.. i have  2 zones too.. don't think there is registers..

IgorYbema, 2022-10-16:
I believe we tried too look for those values when you have the real optional pcb installed and coudn't find them. Can't remember correctly anymore.
The emulated version we can see because the communication is then done by heishamon so it receives the extra values.

Do you still have the panasonic service center website access? Can you check if the pump values are visible there? (with cz-taw1 conneced)

If it is in the normal data we should look at http://heishamon.local/debug and see which values change if the pump starts or stops (but hard to see because there will change a lot I guess)

PGnl, 2022-10-16:
The service cloud is able to see it. 
no values now though, as I use heisha.
I can request pump activity (on/off) in the graph.
[image]

So it should be possible.
This would allow for people to actually switch fans etc based on this signal.

IgorYbema, 2022-10-16:
Ok that seems to be available indeed. Can you try to find it yourself with the debug url?
See the well known bytes https://github.com/IgorYbema/HeishaMon/blob/main/ProtocolByteDecrypt.md which we already know.
Should be after byte 110 and on. See byte 177 where someone thought this would be the mixing valve values so probably the pump values are close by.

MiG-41, 2022-10-16:
As i remember ,these values in Service Cloud are not corresponding to real waterpump status, but only indicates that zone is active or not (and for this we have topic and Set Command).
And in reality it is hard to see only by checking pumps ,becouse when you are not using thermostats for zones ,in reality pumps are running all the time ,only mixing valves de deciding if zone will be closed and Flow is only in zone (not thru the buffer)   ,or open thru the buffer si HP could add some heat to that zone.

MiG-41, 2022-10-16:
Only thermostats (inner or outher ,or termistors) and DHW mode  ,and set zone inactive can stop waterpump in zone...

PGnl, 2022-10-16:
@MiG-41 Not sure what you mean. I have a dual zone set-up using two sensors, bth controlled by the main control unit.
This means my thermostat ( main controlunit) controls both the zones and switches the pump on/off.
These are currently hooked up to a relais controlling my mixing valve.
This works well, however would like to maonitor that signal to easilly switch on/off other switches.

@IgorYbema 
What I have found is that byte 116 seems to switch accurately from 99 to A9 (A9 when heat is requested, 99 when not).
So lowering the temp requested will stop heat request for the zone and switch to 99. this seems to work when both having zone 2 requesting and stopping request from zone 1, and when no heat request from zone 2 at all and stopping/starting request from zone 1.
Haven't found similar for zone 2 yet, and will try to check for longer.
Could also be some kind of temp or other but for now stays constant, so will keep watching this signal for longer.

MiG-41, 2022-10-17:
@PGnl , i also have two zones , but using Heishamon with OptionalPCB Emulation , so i see exacly when HP is starting waterpumps....

But you can compare byte #6 and #117 , not sure now if booth have identical information about active zones , maybe #117 have waterpumps...

closed 2024-09-12 by Egyras