heishamon/HeishaMon #282: No Issue, just a question

Bogeyof, 2022-11-05:
Sorry for writing here but I have no clue how to send a simple question / message...

I programmed an extensive routine for setting the target temperature for the Panasonic HP, which sends a new setting for Z1_Heat_Request_Temp to the heat pump via Heishamon quite often (every few minutes).
Do you think this destroys the internal flash memory by frequent writing or are the registers kept in RAM and there is no problem?

geduxas, 2022-11-05:
No it will not be kept in ram. It will written.. becouse after power lose you have last set value..  
Why you are setting it so often? Is there any point for that?

Bogeyof, 2022-11-05:
It´s a try for 9kW T-Cap to lower power of heating. I´m setting Z1_Heat_Request_Temp to a value, so that Main_Target_Temp = Main_Outlet_Temp - 1 (if possible, Z1_Heat_Request_Temp has only the range from -5 to 5). Everytime Main_Outlet_Temp is growing 1 up, I change Z1_Heat_Request_Temp also until Main_Target_Temp reached the real desired target value.
While Main_Outlet_Temp is > Main_Target_Temp, the HP goes to lowest possible power consumption. So I could extend the running time and lower power consumption, alltogether with a minimal better COP.
But I don´t want to destroy flash memory, so I will stop this test...

DL4JY, 2022-11-05:
Regarding flash, has somebody checked the circuits already, if the really use FLASH or RAM with battery backup?
I believe if they consistently write into flash they might end up like Tesla, where it "kills" the MCU (€3k+) because of logging to many data in /var/log.

Cheerio

michalwoz, 2022-11-05:
@Bogeyof I'm user of 9kW T-cap and HeishaMon. What you want to do was the first thing I thought about, even before I selected HP vendor. (I did equivalent with BAXI gas boiler and OpenTherm protocol.) HeishaMon was the easiest way for me to control HP. Actually, HeishaMon was the reason I've chosen Panasonic. I will also try to do some optimisations. I was really disappointed that 0-10V control don't work like we wish. I also want to take control over frequency of compressor. We cannot do it directly, so we have to make some workaround.

One thing that I've not investigated yet is the point where the COP is optimal vs frequency. First impressions after few weeks of looking on the graphs says that the lowest freq gets the best COP, but some charts say, that running on the lowest frequency don't give the best COP.

BTW, possibility of killing flash was also bad news for me. Maybe one of us will sacrifice a pump for the sake of science? Let's do a public fundraiser for a new motherboard for him 😄Anyway, if that is stored into flash, I've already lost few hundreds write cycles due to behaviour of default HA integration (described in issue recently).

MiG-41, 2022-11-06:
This "killing flash" is only speculations/assumptions , no one have it. And i also at beggining of Heishamon tests send a lot of SET commands at one time in a week , and even hear ,that there is something going inside PCB :) 
But it still alive.

z4m0, 2022-11-25:
@Bogeyof Maybe you can try to set the silent mode to cap the max frequency.

geduxas, 2022-11-25:
> @michalwoz This "killing flash" is only speculations/assumptions , no one have it. And i also at beggining of Heishamon tests send a lot of SET commands at one time in a week , and even hear ,that there is something going inside PCB :) But it still alive.
> 
> Also COP , can be individual HP , and in 9kW can be highest on lowest Hz , but on 7kW can be opposite :)

@michalwoz do you know what kind of cpu panasonic is using? Or maybe someone has detailed photo of pcb in both sides? Anyone read datasheets of cpu and/or flash memory?

MiG-41, 2022-11-26:
I had picture somwhere , generally no any descriptions on cpu and flash....

HighlyCompressedAir, 2024-09-11:
Duplicate https://github.com/Egyras/HeishaMon/issues/225

closed 2024-09-12 by Egyras