heishamon/HeishaMon #284: Request: DHW Capacity

Donpepe23, 2022-11-09:
Hi,

ist it possible to add DHW Capacity to the commands?

THX

IgorYbema, 2022-11-09:
Can you explain more what you mean and why you need it?

Donpepe23, 2022-11-09:
Hi,

here the description from the service manual:

"To select tank heating capacity to variable or standard. Variable capacity heat up tank with fast mode and keep the tank temperature with effcient mode. While standard capacity heat up tank with rated heating capacity"

In the end "variable" only reduce the compressor frequency for DHW, which leads to higher efficency and lower power consumption, but reloading DHW will take longer.  I have two ideas:

- With the "rules" funkction this  can be changed automatically: low room heating demand: variable, higher efficency and better photovoltaic consumption  / high room heating demand: standard, to reduce DHW time (more heating time)

- For DHW the target temp is always 59°C and even you set the target DHW temp to 46°C you end up with water temp. above 50°C in the end of the DHW cycle.  I would like to switch towards the end of the DHW cycle to "variable" to finish the cycle with lower water temp -> higher efficency.

I have one of the Jeishas, where the demand control is not working... 

Thanks

IgorYbema, 2022-11-09:
Hi. I can try to add that. Need to find the correct byte address. Can you post http://heishamon.local/debug outpost for when you set it to standard and again when you set it to variable? Please with a short time between both captures

michalwoz, 2022-11-09:
@Donpepe23 

> * For DHW the target temp is always 59°C and even you set the target DHW temp to 46°C you end up with water temp. above 50°C in the end of the DHW cycle.  I would like to switch towards the end of the DHW cycle to "variable" to finish the cycle with lower water temp -> higher efficency.

In TCAP H 9kW target temperature goes to 57°C, almost the same as yours. I think that a lot of above depends how high your thermometer and snake are mounted in tank. Cannot you set lower target temp? There might be an issue to set delta to start heating properly with low target temp. How long after reaching target temp your HP turns off? How many °C actual temp is higher than target temp when HP turns off? in my case it is just 1°C as you can see on chart.
[image]

Donpepe23, 2022-11-09:
@

> Hi. I can try to add that. Need to find the correct byte address. Can you post http://heishamon.local/debug outpost for when you set it to standard and again when you set it to variable? Please with a short time between both captures

Will do... THX

@michalwoz :

I have  no heat exchanger... Have a 500L tank with "fresh water station" (proper englisch term?). The delta is arround 4-6K, as the temp sensor is in the middle of the tank. So the problem is, by the time the target temp at sensor level is reached, the water temp is already 4-6K above. But I can not change the spostion of the sensor or lower the target temp, as I need to have apprx 300l 46°C hot water to run the fresh waterstation with 43°C hot water output.

closed 2024-09-12 by Egyras