heishamon/HeishaMon #286: Jeisha H74 error PCB communication

Tom54497, 2022-11-11:
After weeks of seamless operation Heishamon V3) I do get occassional H74 errors which can only be found in the error history but not on the display. Obviously the Jeisha stops working but doesn't shut down. 

I am not alone with that. See https://www.photovoltaikforum.com/thread/179235-dasmoritz-w%C3%A4rmepumpen-thread-diy-panasonic-jeisha-wh-mdc05j3e5/?pageNo=27 for reference. Post #261.

Any ideas?

IgorYbema, 2022-11-11:
H74 errors with optional pcb emulation can occur if heishamon isn't able to responde fast enough. The emulation requires constant messages from heishamon to heatpump and a short problem with heishamon timing can cause it to not respond fast enough resulting in the H74 error. This can for example be caused by looking at the webinterface to often, or if the wifi or mqtt server isn't working correctly (this causes heishamon to retry wifi/mqtt and not time for the emulation commands).
So please check if you heishamon has good enough wifi and isn't having mqtt issue etc.

IgorYbema, 2022-11-11:
Yes I did and please don't steal someone else his thread.

IgorYbema, 2022-11-11:
Wifi is a bit low. Should be 70% at least. That could cause wifi reconnects.

Tom54497, 2022-11-11:
A network socket on board would solve the problem or a connection to an external antenna. Would be a nice hardware add-on for an enhanced version of Heishamon.

Tom54497, 2022-11-11:
Do I have to pay attention to the MQTT reconnects? What is normal?

IgorYbema, 2022-11-11:
External antenna doesn't improve much. Tried that before. And for ethernet port the esp8266 is too slow and has not enough gpio's. Better wifi mesh in your house should be the solution :)
MQTT reconnects is too much but that can be caused by the bad wifi.

MiG-41, 2022-11-11:
Like @IgorYbema said , problem with H74 is lack of Optional PCB commands from Heishamon to HP. 
What i remember the problem is mostly during start or restart HP , HP needs this command quite fast , something like max 2s after power up. This was initially ensured in v2 , but during code change perhaps something changed...
In this case HP will start up with an error , but it will work further if commands arrive later ( only error will be stored in history ,but it won't block HP ).

Lack of Optional PCB commands during work can be much longer , app 40s , then HP will give an error , and stop work ( so stop compressor if runs).  Not remember ,if it is possible to work further if commands apear.

Try to log TOP44 , perhaps it tells ,what happend in that time.... Also in HP history there is time ,when it apear and you will remaind what happend then...

MiG-41, 2022-11-11:
@IgorYbema ,and about stealing a topic ;) 
Since you are very seldom on Slack , on debug page with optional pcb first line are substituted with optional pcb command ,instead to be full answer (only sometimes after refreshing i can get real full answer):
[code]
data: 71 11 01 50 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 2D 1A 16 22 95 16 9E 59 0A 09 00 00 00 
data: 00 00 00 00 00 00 80 80 80 80 AF 71 71 65 9A 00 00 00 00 00 00 00 00 00 00 00 80 80 79 8A 85 85 
data: D0 71 78 1F 7E 1F 1F 5B 65 6F 65 A6 9F 71 8F A6 A1 71 8F 8F 83 76 90 90 90 9B 90 90 8F 96 83 8F 
data: 8A 03 79 7B C1 10 00 00 00 00 00 00 00 00 55 F5 55 21 85 15 59 07 21 23 69 00 00 00 00 00 00 00 
data: 00 E2 CF 0D 86 05 12 D0 0C 95 05 9A 9B AC 8A A3 A1 9A 9B A1 A2 32 32 32 80 A1 A4 90 99 9D 89 8A 
data: 8C 8C 8D 33 01 01 01 00 00 22 00 01 01 01 01 79 79 01 01 B7 0A 00 00 18 00 03 00 00 1A 00 00 0D 
data: 02 01 01 01 01 01 01 00 00 00 2F 
[/code]

Tom54497, 2022-11-22:
Yes it was a problem of a bad wifi connection. Had 50 % signal quality and permanent mqtt reconnects. Now with over 70 % they are almost gone and no H74 occurred any more. THX

MiG-41, 2022-11-26:
@IgorYbema ,today i also had issue with H74 error - found HP switched off. 
I have wifi signal strong enough , but i played with network today , probably it was switched off for some time. 
Could you please look at the code , optionalPCB commends must flow to the HP no matter if the network is or not...

IgorYbema, 2022-11-26:
This can't be fixed 100%. The netwerk code of the esp8266 isn't full async so if there is issue's with packet delivery the code can wait and therefore blocking the optionalPCB commands. Sorry for that :(

MiG-41, 2022-11-26:
But even 40seconds ?

IgorYbema, 2022-11-26:
I am afraid that can happen indeed. TCP timeout of 60 seconds or something like that. Not sure.

IgorYbema, 2022-11-27:
Maybe it helps if we switch to this async mqtt library. I'll make a build for it to test https://github.com/marvinroger/async-mqtt-client

Bender2k, 2022-12-04:
Hi all,

I had the same issue now. Heishamon is installed for over a month now.

I have seen 2x H74 in the log of the control panel and the heatpump shutdown.
I started the Heatpump again, but the same 2x H74 error and the shutdown happened immediately again.

After that i rebooted Heishamon using the webinterface and the heatpump is now running for one hour again without issues!

Before the reboot the uptime was 13 days and i have not changed anything on my WiFi settings.

Bender2k, 2023-04-30:
Hi, update from my side.
Heishamon is now running nearly 150 days **without PCB emulation** absolutely perfect !!!
There also never was a single H74 error.

Wifi signal: 58%
Memory free: 66%
Correct received data: 99%
MQTT reconnects: 586
Uptime: 146 days 20 hours 9 minutes 23 seconds

But i want to start using PCB emulation soon.
Will this issue be fixed in a timely manner? I haven't seen a fix inside the 3.1 changelog.

Thank you!

Bender2k, 2023-09-16:
Hey, is this issues something you are investigating for the next release?

I really want to use the power limitation and SG function. If you can't solve this error, i have to think about buying the expensive optional PCB. :(

IgorYbema, 2023-09-17:
Hi,
There is no focus for me on fixing this. But without emulation you can change the temperature curve already with normal commands. So why do you want to use the SG function? You can change all set temperatures already.
This is what I do for 3 years already. Changing the set temp for the DHW when there is enough sun or tariff is low. And set room watertemp higher if the room isn't heating fast enough etc.

Bender2k, 2023-09-17:
Hey, you are right, the temp can be changed already without the PCB. As until now i am in read only mode, i havent thought of it, lol.

The power limitation would have been a nice feature as in winter my 9kW often starts with (nearly) full compressor frequency after defrost, which is a bit noisy.

Anyhow, thank you for your reply!

IgorYbema, 2023-09-17:
I believe you mean during a defrost (mode 2 defrost). That uses the compressor to heat up the internal cycle and yes that is noisy but necessary. If you slow that down it can't defrost properly.
I do remember someone tested it before and for the defrost the power limitation doesn't work. And in all, the power limitation will not be reached in most of the time. SG is implemented to fill the demand of some country rules but it doesn't have much effect.

If you defrost is too often go to a mode-2 defrost (for 9kW and higher models) that means that there isn't enough heat stored in your room water circuit causing the return temp to drop below (i believe) 27 degrees, when then mode-2 defrost kicks in.

Bender2k, 2023-09-17:
I have the 9kW LT, no TCAP.
And i really talk about AFTER defrost. The HP always starts with higher frequency (higher than in defrost) and modulates down after 5-10 minutes. I just thought to limit that initial peak a little bit.
But thats a different topic and not related to the H74 error.

IgorYbema, 2023-09-17:
Ah but that situation you can fix by matching the set temp to just 1-2 degrees above the waterout temp and compensate up until you reach the desired water out temp. It will then not burst to the high frequency.

geduxas, 2023-09-17:
I have optional pcb, and had bought 0-10V with demand control.. it's seam's that compressor is limited, but not in all cases.. there need to make more investigation.

Bender2k, 2023-09-17:
Really? I have to try...
Attached an example of last winter with -10°C outside

[image]

geduxas, 2023-09-17:
Need to wait for winter.. now i have tested only for DHW, and by compressor sound it's also seam's throttled :) after releasing control, compressor jumps to it's demand freq.. and after limiting, compressor goes down.. so not shure for now what is controlled... from manual it's seam's that current should be limited.

Bender2k, 2023-09-17:
> I have optional pcb, and had bought 0-10V with demand control.. it's seam's that compressor is limited, but not in all cases.. there need to make more investigation.

I already tested PCB emulation to limit the power and according to my screenshot it works fine. I can limit the HP to for example 80Hz in the case i talk about.

Maybe the 7kW model with lower freq would have been enough for my house, but i just didn't know before.

Bender2k, 2023-11-30:
> External antenna doesn't improve much. Tried that before. And for ethernet port the esp8266 is too slow and has not enough gpio's. Better wifi mesh in your house should be the solution :) MQTT reconnects is too much but that can be caused by the bad wifi.

Hi Igor,
i found that:
https://github.com/nopnop2002/esp8266_ethernet

Are you sure its not possible to include a library like that?
I looks like this boards only requires 5 GPIOs

I tried your suggested method to temporarily reduce the Request temp, but it is verly unreliable to control the compressor frequency that way.

The next thing i do is to buy a small wifi access point and place it directly beneath heishamon. I want to use the power limit function. :)

By the way: Is it possible to set the power limit value using Rules?
This would make it easy to use different limits according to the outside temp.