heishamon/HeishaMon #287: Heatshift doesn't work with external thermostat

dudu2082, 2022-11-15:
Hi guys. In last year I use heatshift if neccesery when HP was set by external temperature curves. 
In this year i decide to try external thermostat to control HP + those curves. So if thermostat is on then  HP control heating by heating curves. 
But now I can't use heatshift (-5 to +5 C) to correct this curves. 
I can  set curves by sending json format command to change curves in mqtt.
Can this one could be fix to use heatshift with external thermostat connected to HP?

IgorYbema, 2022-11-15:
It doesn't work due to the heatpump not supporting it. You can see the same behaviour when you try to shit using the Panasonic controller

dudu2082, 2022-11-15:
> It doesn't work due to the heatpump not supporting it. You can see the same behaviour when you try to shit using the Panasonic controller

On HP controller also I could't change it. So you 
have a right. I thought that could be done by HeishaMon in some "magic" way.

IgorYbema, 2022-11-15:
You can ofcourse change the setpoint and create some kind of shift that way.

MiG-41, 2022-11-17:
@dudu2082 ,  You can send your "shifted" curve via SetCurves command , and HP will imidietly react on it ( also during work).

HighlyCompressedAir, 2024-09-11:
Unsupported by the heat pump, can't be implemented. Issue can be closed.

closed 2024-09-12 by Egyras