heishamon/HeishaMon #288: Can't get s0 to work

traptoo, 2022-11-18:
I have two Heishamon devices installed on my system. An energy meter OR-WE-521 is connected to each device.

Unfortunately, I do not get any values displayed in Heishamon:

[image]

The configuration in the settings looks like this:

[image]

I have no experience with the s0 interface so far and maybe I have a general error in my thinking. Is it possible that a voltage source has to be connected in between?

IgorYbema, 2022-11-18:
These default settings look ok for this module. Try to swap the cables left
and right

traptoo, 2022-11-18:
That seems to be the right way. Console shows:

[image]

Do i have to wait for some time? Because there are yet no values in heishamon:

[image]

IgorYbema, 2022-11-18:
You see that there are only bad pulses. The quality is not good enough or the pulse width is longer. Try to set max pulse higher (for example 200ms)

traptoo, 2022-11-18:
Increasing the max pulse to 200ms leads to more bad pulses:

[image]

IgorYbema, 2022-11-18:
And making the minimum lower?
Clearly the quality is not good. What cables are you using for the connection?

traptoo, 2022-11-18:
Also doesn't work :(

Cable is a H03 VVH-2F with a length of ~2.5m.

[image]

IgorYbema, 2022-11-18:
what happens if you just short the gnd and s0_1 (and gnd and s0_2) for short times? That should count as pulses also

closed 2024-09-12 by Egyras