heishamon/HeishaMon #289: Unusual state changes between on and offline

Messiam, 2022-11-22:
Hello,
I'm currently checking my initial setup which is quite default. 

And I see some changes of "panasonic_heat_pump/LWT" which might be unusual.
It toggles between online and offline.
Is this normal? 

22.11.2022 21:56:05 
Offline
22.11.2022 21:55:57(-8.22 seconds) 
Online
22.11.2022 21:55:41(-15.77 seconds) 
Offline
22.11.2022 21:55:27(-14 seconds) 
Online
22.11.2022 21:55:27(-0.22 seconds) 
Online
22.11.2022 21:55:11(-15.78 seconds) 
Offline
22.11.2022 21:55:01(-9.76 seconds) 
Online
22.11.2022 21:54:57(-4.47 seconds) 
Online
22.11.2022 21:54:35(-21.78 seconds) 
Offline
22.11.2022 21:54:27(-8.22 seconds) 
Online
22.11.2022 21:54:17(-9.79 seconds) 
Offline
22.11.2022 21:54:06(-10.74 seconds) 
Online

IgorYbema, 2022-11-22:
My first guess would be your wifi is not strong enough. Should be 70% for minimal good setup.

Messiam, 2022-11-22:
Might be, Wifi signal is ~56 - 58%. The wifi hotspots are not very close. 

What is the LWT topic actually? 
The documentation just mentions "Online/Offline (automatically returns to Offline if connection with the HeishaMon lost)".

Is it the connection from Heishamon to Wifi or to the heatpump?

IgorYbema, 2022-11-22:
It is a signal in mqtt to show the state of connection from heishamon to mqtt

Messiam, 2022-11-22:
Ok. Thank you for explanation. 

I don't understand how Heishamon can set it to "offline" when there is no connection ;-) But this is probably the last state before update to mqtt is sent. Right?

IgorYbema, 2022-11-22:
No the mqtt server sets this state as it has lost connection

Messiam, 2022-11-22:
Ok, I need to learn about some mqtt basics I think. Thank you!

IgorYbema, 2022-11-22:
https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament/

closed 2024-09-12 by Egyras