heishamon/HeishaMon #290: Q: Fractional temperatures for inlet and outlet temperature?

bluberly, 2022-11-24:
When enabling monitoring on the Panasonic Service Cloud, the heatpump's inlet and outlet temperatures are displayed as fractional degrees with a resolution of 0.25°C. The downloadable CSV files also has the fractional temperatures

Has someone tried to decrypt the data stream from the heatpump and tried to find out if the fractional temperatures can be provided as well in Heishamon?

IgorYbema, 2022-11-24:
Hi,
Fractional data is calculated by the service cloud as a function of time. The data is send from the panasonic only in integers just like heishamon is receiving it.

bluberly, 2022-11-24:
Thank you! So a fake accuracy :-) Closing the issue.

closed 2022-11-24 by bluberly

geduxas, 2022-11-24:
> 

Here is for comparasion panasonic temperature vs dallas 1wire read out.. 

[image]

i think thous temperature instabilities will be rounded in panasonic cloud..

IgorYbema, 2022-11-24:
this is indeed a nice example. There where you see the integer change from 36 to 37 a few times per minute the service cloud calculates a fractional value.