heishamon/HeishaMon #291: Setcurve Commands via Heishamon Rules

mieschc, 2022-11-25:
Moin gents,

we have some gents in our german Heishamon Community which want to use soley the Heishamon Rules an no extra third-party stuff (MQTT broker etc.) to keep it simple and to have more operational safety (less equipment - less possibility for fault/failure). As far as we understand the following commands can only be used via a JSON File:
"Z1_Heat_Curve_Target_High_Temp", //TOP29
"Z1_Heat_Curve_Target_Low_Temp", //TOP30
"Z1_Heat_Curve_Outside_High_Temp", //TOP31
"Z1_Heat_Curve_Outside_Low_Temp", //TOP32

Is there a possiblity to adjust the heat curve settings directly via the Heishamon Rules or to add this feature?

Brgds

MiG-41, 2022-11-26:
To be precise SetCuve sets not only TOP 29-32 , but generally all Curves , so TOP72-75 , TOP82-89...

McMagellan, 2022-11-27:
We know that there are in summary 4 heat/cool curves setups.
What we will do is to switch off the internal Z1 heatcurve calkulation and control it via Heishamon-Rules. You can do it simply to write the target temperature in TOP27, but then your weektimer will be working corrupt.

To find a second way without TOP27 is correct the target temperatur parameter from the Z1 heatcurce setup with your own calculated parameter transfered even in TOP29 and TOP30. 
In documentation there is only a way with using JSON- string possible. We need to do this from Rules-Environment.

Why can Heishamon handle this via JSON but not from Rules? That is the question.

Does an General-Command exists in Rules to set any TOP- parameter?
For example:   @SetTopXXXto = YYY;
So we can do it this way.

Best regards

CurlyMoo, 2023-01-15:
@IgorYbema Any ideas on this one?

CurlyMoo, 2024-04-16:
Setting the curves inside the rules should now be possible with the latest code in this PR:
https://github.com/IgorYbema/HeishaMon/pull/121

It should still be tested though.

Setting the curves is then done like this:
````ruby
@SetCurves = concat('{"zone1":{"heat":{"target":"high":', @Z1_Heat_Curve_Target_High_Temp, ',"low":', @Z1_Heat_Curve_Target_Low_Temp, '}}}}');
```

HighlyCompressedAir, 2024-09-11:
This issue can be closed.

closed 2024-09-12 by Egyras