heishamon/HeishaMon #293: Implementation request for byte 118 decoding (inlet and outlet quarter water temperature)

berhan77, 2022-11-27:
Would someone please be so kind and implement the decoding of byte 118 in heishamon.

Byte 118 decrypt:
Inlet Water Temperature [°C]
1st, 2nd & 3rd bit
.00 b001
.25 b010
.50 b011
.75 b100

Outlet Water Temperature [°C]
4th, 5th & 6th bit
.00 b001
.25 b010
.50 b011
.75 b100

Best regards

michalwoz, 2022-11-28:
Which models which generations supports that resolution?

berhan77, 2022-11-28:
My model of heat pump is the splitunit WH-SDC0709J3E5 WH-UD07JE5 but I think all H and J series units have this resolution like using CZ-TAW1 module.

IgorYbema, 2022-11-28:
Hi,
If this is true, thanks for finding this.
Until now we thought CZ-TAW1 didn't have this data and that the service center of panasonic is doing time calculation for the fractionals. Read also here: https://github.com/Egyras/HeishaMon/issues/290

It still doesn't explain the fast flipping of top5 and top6 during the rise of fall of the temperature when it is in between two intergers. See the graph in the mentioned issue.  But I'll investigate

IgorYbema, 2022-11-28:
I created a branch of the code to test this. You can download the binary from here: https://github.com/IgorYbema/HeishaMon/actions/runs/3562932211

It is based on v3.0 and will only change the top5 and top6 to include this byte 118 info.

IgorYbema, 2022-11-28:
Could be right:
[image]
At around 9:15 I installed the beta code and you see the fractional information added. It matches the flow of the dallas temp sensors correctly.
@berhan77  how did you find this? Can you also find the fractionals for other temp values?

berhan77, 2022-11-28:
> It still doesn't explain the fast flipping of top5 and top6 during the rise of fall of the temperature when it is in between two intergers. See the graph in the mentioned issue. But I'll investigate

This is probably due to the resolution of the ADC, some temperatures are not displayed. For example, the values ​​24, 24.25, 24.75, 25, 25.5, 25.75, 26, 26.5, 26.75, 27.25 are displayed in the Aquarea Service Cloud, the values ​​24.50, 25.25, 26.25, 27 are missing. I installed a thermal activation of building structures with very stable water temperatures and very precise external sensors, so I can understand the behavior of byte 118 very well.

I installed the alpha firmware, it works and correlates with the external sensors, thanks.
[image]

berhan77, 2022-11-28:
> @berhan77 how did you find this? Can you also find the fractionals for other temp values?

My heat pump works with a very low dT, but the power control still works with slight temperature jumps. So, I recorded the hexdump over a longer period and compared it with temperature changes. Only byte 118 changed. Decoding was then relatively easy.

IgorYbema, 2022-11-28:
@boutermw see above, you can now use that test build

Main release will not happen soon I think. Need some more practical changes for me to have a reason for release a v3.1

geduxas, 2022-11-28:
@IgorYbema, @boutermw do you have any tools for raw data monitoring or comparison?

berhan77, 2022-11-28:
> @IgorYbema, @boutermw do you have any tools for raw data monitoring or comparison?

I used Notepad++ and a small batch script for Windows.

:loop
echo %date%_%time% >> file.txt
curl http://192.168.0.240/debug >> file.txt
timeout /t 10
goto loop

IgorYbema, 2022-11-28:
If you want to help, start reading https://github.com/IgorYbema/HeishaMon/blob/main/ProtocolByteDecrypt.md
Where the byte values aren't known yet we could have some help.

We believe that towards byte 117 there are only settings and from byte 118 and further there are the values. So you see that for example byte 119 and byte 120 are providing information which we still don't know what they do.

MiG-41, 2022-11-28:
Yes, and also these bytes ,with were considered as 0 , in new versions Panasonic can implement something. I.e. there are some new data for gen. J ,not availble for order gen.H.

IgorYbema, 2022-11-28:
Hmm.. not sure if it is correct. For outlet I don't see any weird values but for inlet I see this:
[image]
It jumps from 26.75 to 26.25 and back a few times. Should be 26.50 then. Maybe we are close but doesn't look correct to me

MiG-41, 2022-11-28:
Maybe wroga coded ,instead 0,25 you adding 0,75 ?

IgorYbema, 2022-11-28:
> Maybe wroga coded ,instead 0,25 you adding 0,75 ?

No checked that

MiG-41, 2022-11-28:
Maybe there is some mix ... 0.75 with 0.5 , or 0.25 with 0.5....
Try to cool down sensor and Look how it went thru up and dowm chang

IgorYbema, 2022-11-28:
Yes, need to check.
Anyway, typical pansonic to use a 3 bits to represent 4 cases. Could be 2 bits :)

berhan77, 2022-11-28:
No, it's the ADC, the accuracy (maybe 10bit) ist to low.

One more note, byte 175 correlates with the defrost status

berhan77, 2022-11-28:
> Defrost status we have already in byte 111, how this 175 behaves ?

It looks like a counter (from HEX 79 - 38)

IgorYbema, 2022-11-28:
> Defrost status we have already in byte 111, how this 175 behaves ?

Could be this timer:
[image]

IgorYbema, 2022-11-28:
But for this fractionals. Can someone now compare the values in this alpha release with the values in panasonic service center?

berhan77, 2022-11-28:
It's the same in the cloud, I only get the following values ​​in the panasonic service center in the past with my heat pump.

19.25, 19.75, 20, 20.25, 20.75, 21, 21.5, 21.75, 22.25, 22.5, 22.75, 23, 23.25, 23.5, 23.75, 24, 24.25, 24.75, 25, 25.5, 25.75, 26, 26.5, 26.75, 27.25, 27.5, 28, 28.25, 28.75, 29, 29.5, 29.75, 30.25, 30.5, 31, 31.5, 31.75, 32.25, 32.5, 33, 33.25, 33.75, 34.25, 34.5, 35, 35.25, 35.75, 36.25, 36.5, 37, 37.5, 37.75, 38.25, 38.75, 39.25, 39.5, 40, 40.5, 41, 41.5, 41.75, 42.25, 42.75, 43.25, 43.75, 44.25, 44.75, 45.25, 45.75, 46.25, 46.75, 47.25, 47.75, 48.25, 48.75, 49.25, 49.75, 50.25, 51, 51.5, 52, 52.5, 53.25, 53.75, 54.5, 55, 55.5, 56.25
So far, the data looks very good.
[image]

IgorYbema, 2022-11-28:
Thanks. So also in service center it sometimes jumps 0.50 instead of only steps of 0.25. Than this looks ok now and you can close this issue. It will be added in main branch later.
I'm logging byte 175 for the defrost for another change later.

closed 2022-11-28 by berhan77

MiG-41, 2022-11-29:
> It's the same in the cloud, I only get the following values ​​in the panasonic service center in the past with my heat pump.
>
> 19.25, 19.75, 20, 20.25, 20.75, 21, 21.5, 21.75, 22.25, 22.5, 22.75, 23, [...]

I also cn confirm ,that from Heishamon i also not seen values like 27,75 28,5 29,25  .....
So if it is corresponding with service cloud then it is ok. ( i don't have service cloud to confirm it).

bluberly, 2022-12-01:
Downloaded and installed the Alpha build and confirm it works and returns plausible values. Have kit WH-UD07JE5+WH-ADC0309J3E5C. Thanks a lot!