heishamon/HeishaMon #294: MQTT reconnects, connectivity issues - hardware problem?

kenzodeluxe, 2022-12-04:
I am using the `HeishaMon v3 communication PCB` and have been using it fine for about two years now. 
For the second time now (first time is about one year ago), I recently noticed that there were connectivity issues between the PCB and the Raspberry PI which acts as a wireless bridge to the network. The symptoms are lots and lots of MQTT reconnects, and when pinging/connecting from the Raspberry to Heishamon, there's huge amounts of packet loss and connectivity works only sporadically. Whenever I get a reboot through, that however doesn't solve the problem either - once rebooted, the MQTT disconnect counter will increase rapidly, and pings will - mostly - not be answered either.
Since there had not been any changes on the receiving side (Raspberry), I focused on the PCB. Ultimately, what solved the problem was to power cycle the setup. Since yesterday's power cycle, the MQTT counter has been stable at "1" and no connectivity issues have been observed.
I am using ESPs in a few places, and I'm unsure if I've seen similar issues on any of those. This issue tracker has at least two similar reports that were solved otherwise - I was wondering if folks have observed such issues on their setups before?
Might this be an issue with the ESP on v3 that could be resolved on v4?

geduxas, 2022-12-04:
I am on V3. Only one issue i have is that after reboot, heishamon stops receiving data from heatpump.. no issues with network or mqtt.. if i remember correctly, one had issues with mqtt.. but after restarting mqtt server issues gone...

jpio123, 2022-12-04:
I have the same problem with MQTT reconnecting very often. 
This is example after recent manual reboot:
Wifi signal: 78%
Memory free: 52%
Correct received data: 96%
MQTT reconnects: 45
Uptime: 0 days 1 hour 17 minutes 17 seconds

My WiFi seems to have good signal, but in general communication with heishamon is quite slow and on ping I see some lost packets, and during ping MQTT lost connection:

--- 10.220.220.138 ping statistics ---
17 packets transmitted, 12 received, 29.4118% packet loss, time 16129ms
rtt min/avg/max/mdev = 3.317/74.947/251.387/77.954 ms

Sun Dec  4 22:37:46 2022 (5061890): Checksum and header received ok!
Sun Dec  4 22:37:46 2022 (5061899): received TOP50 Discharge_Temp: 24
Sun Dec  4 22:37:46 2022 (5061902): received TOP52 Defrost_Temp: 10
Sun Dec  4 22:38:01 2022 (5076380): Lost MQTT connection!
Sun Dec  4 22:38:01 2022 (5076380): Reconnecting to mqtt server ...
Sun Dec  4 22:38:01 2022 (5076411): Heishamon stats: Uptime: 0 days 1 hour 24 minutes 36 seconds ## Free memory: 59% ## Heap fragmentation: 12% ## Max free block: 20824 bytes ## Free heap: 23576 bytes ## Wifi: 78% (RSSI: -61) ## Mqtt reconnects: 46 ## Correct data: 97.19%
Sun Dec  4 22:38:01 2022 (5076426): Requesting new panasonic data
Sun Dec  4 22:38:01 2022 (5076430): sent bytes: 111 including checksum value: 18 

Any idea what can be the problem?

geduxas, 2022-12-04:
Do you tried restart mqtt server?

jpio123, 2022-12-04:
I was restarted few times. I checked logs of MQTT (Mosquito in HA).
Heishamon PCB address is 10.220.220.138. Not sure what is "172.30.32.2" I do not use this addresses in my network, but also HA is not reachable from outside.

2022-12-05 00:16:31: New connection from 10.220.220.138:55231 on port 1883.
2022-12-05 00:16:31: New client connected from 10.220.220.138:55231 as HeishaMon (p2, c1, k5, u'mqtt').
2022-12-05 00:17:41: New connection from 172.30.32.2:58054 on port 1883.
2022-12-05 00:17:41: Client <unknown> closed its connection.
2022-12-05 00:18:48: Client HeishaMon has exceeded timeout, disconnecting.
2022-12-05 00:19:01: New connection from 10.220.220.138:57014 on port 1883.
2022-12-05 00:19:01: New client connected from 10.220.220.138:57014 as HeishaMon (p2, c1, k5, u'mqtt').
2022-12-05 00:19:41: New connection from 172.30.32.2:38404 on port 1883.
2022-12-05 00:19:41: Client <unknown> closed its connection.
2022-12-05 00:19:42: Client HeishaMon has exceeded timeout, disconnecting.
2022-12-05 00:19:46: New connection from 10.220.220.138:51501 on port 1883.
2022-12-05 00:19:46: New client connected from 10.220.220.138:51501 as HeishaMon (p2, c1, k5, u'mqtt').
2022-12-05 00:21:41: New connection from 172.30.32.2:46720 on port 1883.
2022-12-05 00:21:41: Client <unknown> closed its connection.
2022-12-05 00:21:42: Client HeishaMon has exceeded timeout, disconnecting.
2022-12-05 00:21:46: New connection from 10.220.220.138:64974 on port 1883.
2022-12-05 00:21:46: New client connected from 10.220.220.138:64974 as HeishaMon (p2, c1, k5, u'mqtt').
2022-12-05 00:22:12: Client HeishaMon has exceeded timeout, disconnecting.
2022-12-05 00:22:16: New connection from 10.220.220.138:49852 on port 1883.
2022-12-05 00:22:16: New client connected from 10.220.220.138:49852 as HeishaMon (p2, c1, k5, u'mqtt').

kenzodeluxe, 2022-12-05:
I've been observing my Heishamon instance for the last 24 hours; before the power cycle, the reconnects would reach many thousands in a few hours. Now, it has been stable at `1` reconnect within the last 24 hours. No configuration change/restart on the MQTT server. 
Also, as mentioned earlier: I wasn't even able to reliably ping the ESP from the Raspberry bridge (with >90% packet loss).

geduxas, 2022-12-05:
Interesting.. my wifi is about 76%, uptime was around 45 days with v3, mqtt reconnect's 1 count.. correct received data 98%. Also i have 8 1-wire temp sensors.. and don't have s0 connected. Also i have extension board in my heatpump so no emulation.. data request and mqtt retransmit set to 5sec.

Do you guys have different configuration? Maybe there is some common issues..

kenzodeluxe, 2022-12-05:
I do not use S0 or 1-Wire, very simple setup with just v3 connected to Heisha. New values collected every 5s and all values are sent after 60s. Unchecked `Debug log to serial1 (GPIO2)` today since I don't have any serial connection.

jpio123, 2022-12-06:
I found out that Debug log to serial1 sends logs to Console that you can view via http on main page. Just to let you know :-)
But I still have this problem with MQTT reconnection.

geduxas, 2022-12-07:
Just curious, do you tried to reboot wifi ap?

Snr001, 2023-01-22:
Similar problem here, based on "bad" wifi connection. Wifi signal 76%, but 1 out of 4 pings exceeds the time limit. Therefore at least one mqtt reconnect per minute. I have two WLAN APs, the Heishamon seems to be in the middle - one time connetcing to one, one time to the other. TAW1 had a similar problem, solution was to connect it via ethernet cable. Will try to move Heishamon to a "better" place. PS: Heishamon FW 3.1

geduxas, 2023-10-23:
Most of MQTT reconnects causes MQTT server itself, or bad network coverage..  try restart MQTT server.

holzmannk, 2023-10-23:
okay, updated Heishamon to version 3.2 (had a intermediate version installed), restarted everything. Same situation.
It looks like heishamon sends some packets twice (see heisha.zip in my recent post).  
Ping shows similar response:

```
64 bytes from 192.168.1.190: seq=66 ttl=255 time=16.342 ms
64 bytes from 192.168.1.190: seq=66 ttl=255 time=18.805 ms (DUP!)
64 bytes from 192.168.1.190: seq=67 ttl=255 time=54.819 ms
64 bytes from 192.168.1.190: seq=67 ttl=255 time=68.564 ms (DUP!)
64 bytes from 192.168.1.190: seq=67 ttl=255 time=71.071 ms (DUP!)
```

IgorYbema, 2023-10-23:
I can assure you heishamon isn't doing this. This is probably something else in your network causing this. What wifi network brand are you using? There are more people seeing this with specific wifi brands and they then need to turn of auto channel switching and other smart things like that.

geduxas, 2023-10-23:
You are talking about retransmission? It's normal behavior in tcp network. Eatch tcp packet requires ACK packages from server.. and if not recived or send NACK it will retransmit.

IgorYbema, 2023-10-23:
From your dump I see a lot of network reconnects (including mqtt reconnects). The network reconnects can be seen due to a lot of ARP announcements and MDNS announcements followed by a mqtt reconnect. I'm pretty sure it is your wifi which is actively disconnecting heishamon for some reason (band steering etc).

Snr001, 2023-10-23:
Just for info: I am using a fixed band for Wifi now, no reconnects since months.

IgorYbema, 2023-10-23:
Try to check the wifi log on openwrt or do a tcpdump there on the wifi interface on wifi level

holzmannk, 2023-10-24:
Ok, had a google crash course in wifi hacking. I think retransmissions are normal in wifi. What I could see is that Heishamon stops sending MQTT ping. I don't know why. After about 7 to 9 seconds mosquitto closes the connection, because of ping timeout. Heishamon tells mosquitto a 5 sec. timeout when opening MQTT connection.
The question is why the ping is not sent? Weak wifi? I don't think, access point says -31dBm signal, -90dBm noise.

During not sending MQTT ping  Heishamon still is servicing some ICMP pings, but not every ping, don't know why.  
Here are the packets just before closing connection, green marked ist the last packet from Heishamon to mosquitto.    

```
- 192.168.1.190 = 80:64:6f:80:2a:d1 = Heishamon
- 192.168.0.187 = b8:27:eb:f9:db:f5 = mosquitto RPI
- 192.168.1.23 = b8:27:eb:de:4d:87 = RPI for ping
```
[image]

same WPA2 encrypted:
[image]

(In the attached data you can see much more wifi traffic because there is a wifi mesh backhaul on the same channel.)
[heisha-03.zip](https://github.com/Egyras/HeishaMon/files/13114331/heisha-03.zip)

IgorYbema, 2023-10-24:
Hi,
The screenshot shows a normal ICMP ping (not a MQTT ping, that not a thing either) from 192.168.1.23 to heishamon (1.190). Not from your mqtt server (which is on another subnet 192.168.0.187). Or at least a different class-C. Is there a firewall in between? Or is it class-B subnet? (255.255.0.0)?

Why heishamon is dropping ping request is not clear to me but that isn't very important. The most valuable is the FIN,ACK from your mqtt server. The mqtt server decides it is time to disconnect. No reason why. Please check the mqtt server logs now.

IgorYbema, 2023-10-24:
If I would guess it is your mqtt server with a 10 sec timeout dropping heishamon. There is really no reason to do this.
Yes, heishamon sends 5 sec mqtt ping to try to keep the line open but the mqtt server should not disconnect that quick!

holzmannk, 2023-10-24:
- Yes it is a 255.25.240.0 subnet, there is no firewall
- ICMP ping is there to test if heishamon is processing wifi, to see if it is disturbed
- I don't know very much about MQTT, but in the connect frame heishamon tells mosquitto keep alive time is 5 seconds, [https://www.emqx.com/en/blog/mqtt-keep-alive](https://www.emqx.com/en/blog/mqtt-keep-alive), they write that after 1.5x Keep Alive time, MQTT broker disconnects the client, in my situation the MQTT ping (keep alive) is not sent for more than 7.5 seconds, but in this time ICMP pings are serviced. I don't know why heishamon stops sending MQTT ping, but mosquitto expects the ping within 7.5 seconds.
[image]

IgorYbema, 2023-10-24:
Yes your are right. That seems to be the case. Can you confirm on the mqtt server log that it is disconnecting due to timeout? Maybe heishamon is busy with something. Do you have default settings in heishamon or did you change the data read timing interval? Are you doing something other special in heishamon like long rules?

holzmannk, 2023-10-24:
yes, mqtt server log says "disconnecting"

`1698166293: Client HeishaMon has exceeded timeout, disconnecting.`  
I didn't change anyting (don't know how), updated firmware yesterday to v.3.2, before I had an intermediate version, no rules.

IgorYbema, 2023-10-24:
Did it work better on the intermediate version?

holzmannk, 2023-10-24:
no, no difference.

IgorYbema, 2023-10-24:
Ok so we need to figure out why heishamon isn't sending packets.
Two reasons:
- wifi is dropping and heishamon needs to reconnect
- heishamon is doing 'stuff' and has no time to send packets

Only way to discover is to read the log from heishamon but on the webinterface you will not see it because it will loose wifi also. So you need to connect a serial-TTL cable to the debug port of heishamon and read from there if you are able.

Your openwrt isn't logging wifi reconnects?

IgorYbema, 2023-10-24:
Btw, the reason for the 5s keepalive is to keep heishamon running in case the mqtt server isn't responding so it stops the mqtt connection from heishamon side.

holzmannk, 2023-10-24:
* heishamon reports that:
```
Tue Oct 24 12:19:54 2023 (106030): received XTOP3 Heat_Power_Production: 2082
Tue Oct 24 12:19:54 2023 (106039): testing XTOP4 Cool_Power_Production: 0
Tue Oct 24 12:19:54 2023 (106044): testing XTOP5 DHW_Power_Production: 0
Tue Oct 24 12:20:08 2023 (120009): Lost MQTT connection!
Tue Oct 24 12:20:08 2023 (120010): Reconnecting to mqtt server ...
Tue Oct 24 12:20:08 2023 (120032): Heishamon stats: Uptime: 0 days 0 hours 2 minutes 0 seconds ## Free memory: 72% ## Heap fragmentation: 29% ## Max free block: 16960 bytes ## Free heap: 28304 bytes ## Wifi: 100% (RSSI: -37) ## Mqtt reconnects: 2 ## Correct data: 100.00%
Tue Oct 24 12:20:08 2023 (120048): Requesting new panasonic data
```
* don't think wifi reconnection is done. In case of reconnection I should see WPA initialization, this is just done once in my dump, but in the dump there are three MQTT reconnections

holzmannk, 2023-10-24:
btw, in my network there is at the same channel a mesh - backhaul to another OpenWrt mesh node.  
Just an idea: They often use a Request-to-send packet to get a time slot. Maybe the ESP32 wifi waits too long in such situation (in the range of 10 to 50ms). And if there are continuously many RTS frames it can not send?
e.g. green is a MQTT ping response that never is acknowledged by heishamon, the next frame that is acknowledged by heishamon is a FIN frame. In between there are five retransmissions of the MQTT ping response.  

[image]

IgorYbema, 2023-10-24:
Haha, I now see a testing debug line still in the code :) Not important for this but I need to remove that :)

IgorYbema, 2023-10-24:
Maybe that is the case. If you remove support for 802.11b and maybe also 802.11g (leaving only 802.11n) it will lower the time wifi needs to send a rts packet. Maybe that helps for heishamon.

IgorYbema, 2023-10-24:
So disallow legacy_rates as an option in openwrt

holzmannk, 2023-10-24:
legacy_rates already was disabled all the time, so that's not the point

IgorYbema, 2023-10-24:
Ok.
Do you have more heishamon log lines about 10 seconds before?

holzmannk, 2023-10-24:
```
Tue Oct 24 12:19:38 2023 (90520): Sending command from buffer
Tue Oct 24 12:19:38 2023 (90521): sent bytes: 111 including checksum value: 1 
Tue Oct 24 12:19:38 2023 (90529): data: 71 6C 01 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:38 2023 (90540): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:38 2023 (90552): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:38 2023 (90564): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:39 2023 (90918): Received 203 bytes data
Tue Oct 24 12:19:39 2023 (90919): data: 71 C8 01 21 8A EA 01 00 00 00 00 00 00 00 AD 01 01 00 01 00 E6 07 01 00 01 00 01 00 00 00 01 00 
Tue Oct 24 12:19:39 2023 (90925): data: 00 00 01 00 00 00 01 00 00 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:39 2023 (90937): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:39 2023 (90949): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:39 2023 (90961): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:39 2023 (90973): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:39 2023 (90985): data: 00 00 00 00 00 00 00 00 00 00 8A 
Tue Oct 24 12:19:39 2023 (90991): Checksum and header received ok!
Tue Oct 24 12:19:39 2023 (90997): testing XTOP0 Heat_Power_Consumption: 428
Tue Oct 24 12:19:39 2023 (91004): testing XTOP1 Cool_Power_Consumption: 0
Tue Oct 24 12:19:39 2023 (91010): testing XTOP2 DHW_Power_Consumption: 0
Tue Oct 24 12:19:39 2023 (91017): testing XTOP3 Heat_Power_Production: 2021
Tue Oct 24 12:19:39 2023 (91023): received XTOP3 Heat_Power_Production: 2021
Tue Oct 24 12:19:39 2023 (91032): testing XTOP4 Cool_Power_Production: 0
Tue Oct 24 12:19:39 2023 (91037): testing XTOP5 DHW_Power_Production: 0
Tue Oct 24 12:19:53 2023 (105010): Heishamon stats: Uptime: 0 days 0 hours 1 minute 45 seconds ## Free memory: 76% ## Heap fragmentation: 30% ## Max free block: 16960 bytes ## Free heap: 29864 bytes ## Wifi: 100% (RSSI: -37) ## Mqtt reconnects: 1 ## Correct data: 100.00%
Tue Oct 24 12:19:53 2023 (105027): Requesting new panasonic data
Tue Oct 24 12:19:53 2023 (105029): sent bytes: 111 including checksum value: 18 
Tue Oct 24 12:19:53 2023 (105036): data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:53 2023 (105049): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:53 2023 (105061): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:53 2023 (105072): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:53 2023 (105079): Requesting new panasonic extra data
Tue Oct 24 12:19:53 2023 (105086): Already sending data. Buffering this send request
Tue Oct 24 12:19:53 2023 (105424): Received 203 bytes data
Tue Oct 24 12:19:53 2023 (105425): data: 71 C8 01 10 56 55 52 61 00 59 00 15 00 00 00 00 00 00 00 00 55 15 11 55 56 15 55 55 56 19 5A 00 
Tue Oct 24 12:19:53 2023 (105431): data: 00 00 00 00 00 00 A4 80 A3 80 B0 71 71 81 99 00 00 00 00 00 00 00 00 00 00 00 80 85 5B 8A 85 85 
Tue Oct 24 12:19:53 2023 (105444): data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D A4 A3 7B 8F A3 A3 7B 8F A3 83 6C 8F 8A 94 9E 8F 8A 94 9E 85 8F 
Tue Oct 24 12:19:53 2023 (105456): data: 8A 11 B5 7B BC 06 7E 7C 15 7C 7E 00 00 00 95 55 55 21 A9 15 59 05 12 12 65 21 51 15 01 5C 80 80 
Tue Oct 24 12:19:53 2023 (105468): data: 80 E2 D5 0B 08 95 02 D6 0E 66 95 A3 00 A8 8B A2 A3 32 32 A4 A3 A1 32 32 A4 A4 A3 A9 99 A3 87 00 
Tue Oct 24 12:19:53 2023 (105480): data: 88 00 00 6D 01 0A 15 00 00 7B 18 27 54 1F 01 42 01 01 01 8D 00 00 82 00 00 01 00 00 05 00 00 06 
Tue Oct 24 12:19:53 2023 (105491): data: 01 00 00 00 00 00 00 03 80 3C 0A 
Tue Oct 24 12:19:53 2023 (105498): Checksum and header received ok!
Tue Oct 24 12:19:53 2023 (105504): received TOP1 Pump_Flow: 24.48
Tue Oct 24 12:19:53 2023 (105519): Sending command from buffer
Tue Oct 24 12:19:53 2023 (105521): sent bytes: 111 including checksum value: 1 
Tue Oct 24 12:19:53 2023 (105523): data: 71 6C 01 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:53 2023 (105535): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:53 2023 (105547): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:53 2023 (105558): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:54 2023 (105924): Received 203 bytes data
Tue Oct 24 12:19:54 2023 (105925): data: 71 C8 01 21 8A EA 01 00 00 00 00 00 00 00 AD 01 01 00 01 00 23 08 01 00 01 00 01 00 00 00 01 00 
Tue Oct 24 12:19:54 2023 (105931): data: 00 00 01 00 00 00 01 00 00 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Tue Oct 24 12:19:54 2023 (105943): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[... cut ...]
Tue Oct 24 12:19:54 2023 (106044): testing XTOP5 DHW_Power_Production: 0
Tue Oct 24 12:20:08 2023 (120009): Lost MQTT connection!
Tue Oct 24 12:20:08 2023 (120010): Reconnecting to mqtt server ...
Tue Oct 24 12:20:08 2023 (120032): Heishamon stats: Uptime: 0 days 0 hours 2 minutes 0 seconds ## Free memory: 72% ## Heap fragmentation: 29% ## Max free block: 16960 bytes ## Free heap: 28304 bytes ## Wifi: 100% (RSSI: -37) ## Mqtt reconnects: 2 ## Correct data: 100.00%
Tue Oct 24 12:20:08 2023 (120048): Requesting new panasonic data
```

IgorYbema, 2023-10-24:
Hmm no wifi message in there also.
But the time between two runs are long

Tue Oct 24 12:19:53 2023 (105027): Requesting new panasonic data
Tue Oct 24 12:20:08 2023 (120048): Requesting new panasonic data

Did you set that to 15 sec (default 5 secs)? Does it help if you set it back to 5 sec?

Also you can disable the hexdump. It will only make heishamon slower printing that.

holzmannk, 2023-10-24:
Did not do any change. Hwo to check it? How to set it to 5 seconds?

IgorYbema, 2023-10-24:
If it is set to 5 secs it will also send more mqtt messages so keepalive will not run out :)

[image]

holzmannk, 2023-10-24:
> Did not do any change. Hwo to check it? How to set it to 5 seconds?

ok, settings page. Yes it is set to 15, don't know why.

geduxas, 2023-10-24:
Hmm whole 17sec between log lines :D something fishy

IgorYbema, 2023-10-24:
> Hmm whole 17sec between log lines :D something fishy

I see 15?

holzmannk, 2023-10-24:
Yes it was set to 15 seconds. Maybe that was the reason for disconnecting. Now set to 5, and I'm waiting for the next disconnect.

holzmannk, 2023-10-24:
ok, just now there was a disconnect ...

IgorYbema, 2023-10-24:
Ok too bad...

geduxas, 2023-10-24:
> > Hmm whole 17sec between log lines :D something fishy
> 
> I see 15?

```
Tue Oct 24 12:19:54 2023 (106044): testing XTOP5 DHW_Power_Production: 0
Tue Oct 24 12:20:08 2023 (120009): Lost MQTT connection!
```

What heishamon has done between those 2 lines? 
17sec

IgorYbema, 2023-10-24:
12 seconds indeed :)
But what is the log now with 5 sec interval and still disconnect?

holzmannk, 2023-10-24:
[image]

IgorYbema, 2023-10-24:
And heishamon log?

holzmannk, 2023-10-24:
can you give me a hint to capture the log file? copy/paste is hard because of updates.

IgorYbema, 2023-10-24:
Yes that is not easy. Best is a serial connection but then you must have that cable.

But I think it is just a too busy wifi network and heishamon not being that fast cpu :)

I need to do something else now. Maybe tomorrow more time

holzmannk, 2023-10-24:
Ok, got it:

```
Tue Oct 24 20:38:54 2023 (1432618): Requesting new panasonic extra data
Tue Oct 24 20:38:54 2023 (1432619): Already sending data. Buffering this send request
Tue Oct 24 20:38:54 2023 (1433048): Received 203 bytes data
Tue Oct 24 20:38:54 2023 (1433049): Checksum and header received ok!
Tue Oct 24 20:38:54 2023 (1433050): received TOP1 Pump_Flow: 24.48
Tue Oct 24 20:38:54 2023 (1433052): received TOP5 Main_Inlet_Temp: 33.75
Tue Oct 24 20:38:54 2023 (1433057): received TOP64 High_Pressure: 21.4
Tue Oct 24 20:38:54 2023 (1433062): Sending command from buffer
Tue Oct 24 20:38:54 2023 (1433064): sent bytes: 111 including checksum value: 1 
Tue Oct 24 20:38:55 2023 (1433447): Received 203 bytes data
Tue Oct 24 20:38:55 2023 (1433448): Checksum and header received ok!
Tue Oct 24 20:38:55 2023 (1433449): testing XTOP0 Heat_Power_Consumption: 428
Tue Oct 24 20:38:55 2023 (1433450): testing XTOP1 Cool_Power_Consumption: 0
Tue Oct 24 20:38:55 2023 (1433451): testing XTOP2 DHW_Power_Consumption: 0
Tue Oct 24 20:38:55 2023 (1433451): testing XTOP3 Heat_Power_Production: 2032
Tue Oct 24 20:38:55 2023 (1433452): testing XTOP4 Cool_Power_Production: 0
Tue Oct 24 20:38:55 2023 (1433453): testing XTOP5 DHW_Power_Production: 0
Tue Oct 24 20:38:59 2023 (1437614): Heishamon stats: Uptime: 0 days 0 hours 23 minutes 57 seconds ## Free memory: 65% ## Heap fragmentation: 17% ## Max free block: 21240 bytes ## Free heap: 25560 bytes ## Wifi: 100% (RSSI: -38) ## Mqtt reconnects: 4 ## Correct data: 98.22%
Tue Oct 24 20:38:59 2023 (1437617): Requesting new panasonic data
Tue Oct 24 20:38:59 2023 (1437617): sent bytes: 111 including checksum value: 18 
Tue Oct 24 20:38:59 2023 (1437618): Requesting new panasonic extra data
Tue Oct 24 20:38:59 2023 (1437619): Already sending data. Buffering this send request
Tue Oct 24 20:39:00 2023 (1438949): Received 203 bytes data
Tue Oct 24 20:39:00 2023 (1438949): Checksum and header received ok!
Tue Oct 24 20:39:00 2023 (1438951): received TOP1 Pump_Flow: 24.12
Tue Oct 24 20:39:00 2023 (1438955): received TOP50 Discharge_Temp: 41
Tue Oct 24 20:39:00 2023 (1438958): received TOP64 High_Pressure: 21.2
Tue Oct 24 20:39:02 2023 (1440516): Sending command from buffer
Tue Oct 24 20:39:02 2023 (1440517): sent bytes: 111 including checksum value: 1 
Tue Oct 24 20:39:02 2023 (1440949): Received 203 bytes data
Tue Oct 24 20:39:02 2023 (1440950): Checksum and header received ok!
Tue Oct 24 20:39:02 2023 (1440951): testing XTOP0 Heat_Power_Consumption: 428
Tue Oct 24 20:39:02 2023 (1440952): testing XTOP1 Cool_Power_Consumption: 0
Tue Oct 24 20:39:02 2023 (1440953): testing XTOP2 DHW_Power_Consumption: 0
Tue Oct 24 20:39:02 2023 (1440954): testing XTOP3 Heat_Power_Production: 2062
Tue Oct 24 20:39:02 2023 (1440954): received XTOP3 Heat_Power_Production: 2062
Tue Oct 24 20:39:02 2023 (1440955): testing XTOP4 Cool_Power_Production: 0
Tue Oct 24 20:39:02 2023 (1440956): testing XTOP5 DHW_Power_Production: 0
Tue Oct 24 20:39:04 2023 (1442613): Lost MQTT connection!
Tue Oct 24 20:39:04 2023 (1442613): Reconnecting to mqtt server ...
Tue Oct 24 20:39:04 2023 (1442638): Heishamon stats: Uptime: 0 days 0 hours 24 minutes 2 seconds ## Free memory: 73% ## Heap fragmentation: 19% ## Max free block: 23184 bytes ## Free heap: 28728 bytes ## Wifi: 100% (RSSI: -38) ## Mqtt reconnects: 5 ## Correct data: 98.23%
Tue Oct 24 20:39:04 2023 (1442641): Requesting new panasonic data
```

IgorYbema, 2023-10-24:
Nothing special there also. I really suspect your wifi is too busy. Maybe switch the mesh backhaul to 5ghz?
I just made a special firmware for you with a keepalive of 30 sec. Maybe that helps for you?
And really have to go now. Talk to you later.
[HeishaMon.ino.d1_mini.zip](https://github.com/Egyras/HeishaMon/files/13125771/HeishaMon.ino.d1_mini.zip)

holzmannk, 2023-10-24:
Ok, thank you! First results -> new problems. But I don't have any tcp capture and no heishamon log, yet.  
Maybe I really have to change the infrastructure. 5ghz is problematic beacuse the routers are to far away, 2.4ghz works better. Maybe I get some cable, but this will take a while.
```
1698174371: Client HeishaMon disconnected due to protocol error.
1698174376: New connection from 192.168.1.190:51797 on port 1883.
1698174376: New client connected from 192.168.1.190:51797 as HeishaMon (p2, c1, k30).
1698174392: Client HeishaMon disconnected due to protocol error.
1698174407: New connection from 192.168.1.190:49819 on port 1883.
1698174407: New client connected from 192.168.1.190:49819 as HeishaMon (p2, c1, k30).
1698174513: Client HeishaMon disconnected due to malformed packet.
1698174518: New connection from 192.168.1.190:60396 on port 1883.
1698174518: New client connected from 192.168.1.190:60396 as HeishaMon (p2, c1, k30).
```

geduxas, 2023-10-24:
Which version of mqtt you're using?

holzmannk, 2023-10-25:
Heishamon speaks v3.1.1, at least in the connection frame.  
[image]

Output from mosquitto:
```
mosquitto version 2.0.11
mosquitto is an MQTT v5.0/v3.1.1/v3.1 broker.
```

geduxas, 2023-10-25:
2.0.11.. its 2y old release. I have some flashbacks to your issue.. that something i had in past.. but it was related to old mqtt server.. and it didn't work stable. But don't remember exactly what problems did i had then.. Now i am running v 2.0.15 in docker container. And for my setup everything is working great.

is it possible for you to test it on newer mosquito server?

geduxas, 2023-10-25:
And reading through releases of mosquito server:
**2.0.12**

**Security**

- An MQTT v5 client connecting with a large number of user-property properties could cause excessive CPU usage, leading to a loss of performance and possible denial of service. This has been fixed.
- Fix max_keepalive not applying to MQTT v3.1.1 and v3.1 connections. These clients are now rejected if their keepalive value exceeds max_keepalive. This option allows [CVE-2020-13849](https://nvd.nist.gov/vuln/detail/CVE-2020-13849), which is for the MQTT v3.1.1 protocol itself rather than an implementation, to be addressed.

**Broker**

- Fix max_keepalive not applying to MQTT v3.1.1 and v3.1 connections. These clients are now rejected if their keepalive value exceeds max_keepalive. This option allows CVE-2020-13849, which is for the MQTT v3.1.1 protocol itself rather than an implementation, to be addressed.

holzmannk, 2023-10-25:
In my opinion mosquitto does everything correct. If there is no keep alive from the client, it drops the connection. That you can see in the tcp recordings. Question is why heishamon does not send further MQTT pings. I think Heishamon has troubles with huge amount of WiFi frames.

geduxas, 2023-10-25:
here is my dump for mqtt <> heishamon. v3.2.2, each read from pump 5s retransmision to mqtt server 5s. 

[image]

in which hardware do you run heishamon, could it be bad board?

holzmannk, 2023-10-25:
yes, that's fine. Looks as expected. In my wifi it works about 5 minutes in the same way, then, sporadicly heishamon stops MQTT ping or sending data.  
After mosquitto closes the connection and reopens it it runs again as expected.  

I bought my board a few weeks ago in the tindie shop ([https://www.tindie.com/products/thehognl/heishamon-communication-pcb/](https://www.tindie.com/products/thehognl/heishamon-communication-pcb/)), think it is the newest?  

I will do some experiments with an extra access point on an other channel, then I expect it will work. But question is why it stops, and is it really necessary to build up an extra infrastructure for this IoT device?

[10 more replies]