heishamon/HeishaMon #296: Failsafe for lost MQTT connection

Samureimer, 2022-12-06:
I want my HP to turn on and off at certain times during the day, and ideally want HA to be in charge of this.
I don't want to loose all heat to my home if my HA server or MQTT broker should malfunction for whatever reason.
So I was wondering: Is there a way to make sure that the HP always returns to _power on_ if the MQTT connection has been lost for a certain amount of time? Either via a rule or some other feature?

I was thinking about using a timer event to check when the HP was last turned on, and then turn on the HP if too much time had passed.
I don't know if this is possible and would like some input.

MiG-41, 2022-12-07:
Generally if you are not using OptionalPCB , there is no chance , that HP will switch off becouse of loosing connection with mqtt.

For OptionalPCB there is a posibility , that HP will not receive a commands with simulate this board , and HP will detect it as H74 error and switch off.

Samureimer, 2022-12-07:
In my example I'm talking about the case where I turn off the HP with home assistant, and for whatever reason looses connection to the ESP running HeishaMon.
In this case I want HeishaMon to be able to detect this missing connection and turn the HP back on.

geduxas, 2022-12-07:
If you have stable build, don't fink any issues could be.. otherwise you always can send commands to heishamon.. and also it's all depends to imagination and knowledge..

Samureimer, 2022-12-08:
Correct, in a perfect with a stable setup where everything is working I should be fine.
I'm talking about a scenario where some link in the communication chain breaks.
No matter how good my setup is I can never guarantee that my HA server does not loose the ability to communicate with the ESP at some point.

Samureimer, 2022-12-08:
Exactly.
I want it to turn back on automatically.
The scenario could be: if the MQTT connection has been lost for 1 hour, turn the HP back on.

geduxas, 2022-12-08:
And what if MQTT is working, but HA don't ?
and what if MQTT server hang, accepts topic refresh but not sends to listeners?

MiG-41, 2022-12-08:
> Correct, in a perfect with a stable setup where everything is working I should be fine. I'm talking about a scenario where some link in the communication chain breaks. No matter how good my setup is [...]

If one chain broke ,HP remain on it's state ,as it was before broke. So if it was ON , it remains ON. Why are you forseen ,that it will automaticly go OFF after chain broke ?

Samureimer, 2022-12-08:
> If one chain broke ,HP remain on it's state ,as it was before broke. So if it was ON , it remains ON. Why are you forseen ,that it will automaticly go OFF after chain broke ?

I'm not. I'm talking about the case where the chain breaks when the HP is in the OFF state.

closed 2024-09-12 by Egyras