heishamon/HeishaMon #297: No rules function after reboot

JockelBln, 2022-12-07:
It sometimes happens to me that HaishaMon apparently performs a reboot on its own. It works fine for a while and then suddenly the uptime in the web interface is reset again.
If that happens, no more rules will be executed afterwards.
The same thing happens when I manually reboot and then close the web interface before the reboot is complete.
The WiFi connection is very good and the data from the heat pump is also received without error.
[image]

I can not explain these independent reboots. So why aren't the rules restarted?
My HeishaMon is v4.6 and the firmware is version: 3.0.

Sorry English is not my first language, this is a google translation.

geduxas, 2022-12-08:
If somehow a new valid ruleset crashes the HeishaMon, it will be automatically disabled the next reboot allowing you to make changes. This prevents the HeishaMon getting into a boot loop.

JockelBln, 2022-12-08:
Thanks!
How can I identify which part of the rules caused the crash?
All entries are still visible under the menu item Rules and after pressing save they run again.

closed 2024-09-12 by Egyras