heishamon/HeishaMon #298: Byte #175 decrypted

rpannekoek, 2022-12-08:
I *think* I figured out what byte #175 represents: this seems to be the position for the Electronic Expansion Valve (EEV).
From what I observed, the value varies between 40 and 120 (when using getIntMinus1), where 120 represents valve max open.
Interestingly, an EEV typically only has something like 8 positions and if you look at the schematics for the WH-MDC05J3E5, only one of 4 EEV poles are actually connected, so the WH-MDC05J3E5 can actually only set 2 positions (whereas the higher power models do have all 4 EEV poles connected, so can set all 8 positions).
From what I observed on a WH-MDC05J3E5, it switches positions around a value of 50.  When it switches, you can observe significant changes in Discharge_Temp (and the difference between Main_Inlet_Temp and Inside_Pipe_Temp).
It's a rather technical, internal reading which you probably won't use to control the heat pump, but it may be an interesting value to monitor and better understand the internal working (I was initially rather surprised by above mentioned changes which did not relate to compressor frequency).

geduxas, 2022-12-08:
Hi, day 2 ago we have discussed same topic in slack chat :) yes indeed it's looks like as eev control.. also for t-cap models 176 byte looks as bypass valve control.. but it needs more investigation.. first of all what value means what? Why it is in current dataset.. and why there is no other valve records? We have 4 way valve, also 2 way valve.. meanwhile i and @IgorYbema have got value of 128 in byte 175.. so what does it means?  Also in different scenarious that value behaves little bit different or it do what it is supposed to do :)

geduxas, 2022-12-08:
Here is mine mointoring values for byte 175 (valve1) and 176 (valve2)

[image]

you could notice that sometimes bypass temp starts flickering while byte 175 is going down..  EVA temp stays preaty stable.. also byte 176 not always are triggered in defrost cycles.. 

i am preaty shure 98.999% that byte 175 is responsible for valve control.. but need to work out and understand ..

geduxas, 2022-12-08:
Here is another sample

[image]

Byte 175 is flickering between 13-31, and flicker bypass temp.. eva temperature stays in almost same position..

rpannekoek, 2022-12-08:
Nice to see that there are more people trying to figure out some of the missing sensor/control values.
Indeed, I also noticed that "Bypass_Outlet_Temp" (byte #161) does not seem to behave like a temperature at all on my WH-MDC05J3E5; it actually seems to be something like EEV control with something like 8 discrete values.  Also during a defrost it behaves similar to what's documented as EEV behavior during a mode1 deice.
But on another occasion where Discharge_Temp suddenly dropped, only byte  #175's sudden value change correlated.

geduxas, 2022-12-08:
> ... an EEV typically only has something like 8 positions and if you look at the schematics for the WH-MDC05J3E5, only one of 4 EEV poles are actually connected, so the WH-MDC05J3E5 can actually only [...]

indeed i din't noticed that.. but it maight be error in shematic's.. or maybe it just didnt shown?

Here is one in WH-MDC05J3E5

[image]

09-16 kw H series

[image]

T-CAP 9-12kw

[image]

T-CAP 16KW

[image]

as i understand it's simple stepper motor.. so it's need 2 pair of channels to work correctly..

geduxas, 2022-12-08:
> MDC05J3E5

this model don't have bypass.. only t-cap models havr that.. so ithink your's value should be same all time ?

rpannekoek, 2022-12-08:
I don;t think it's an error in the schematics; there is a separate schematic for  WH-MDC05J3E5 and the higher power variant (7 KW and 9 KW).  I think it partially explains why heat pump variants which are largely the same can perform differently.
It also corresponds with behavior I observe on the WH-MDC05J3E5: it seems to have only two positions for the EEV valve.

Indeed, the WH-MDC05J3E5 doesn't have a bypass at all, but the values of byte #161 do change (with approx. 8 distinct values) and and neither of the values look like a temperature reading (which makes sense, because it does not have a bypass temperature sensor at all).

geduxas, 2022-12-08:
> I think it partially explains why heat pump variants which are largely the same can perform differently.

I am not expert.. just learning and trying understand.. what i have found this vide:
https://youtu.be/LqUEIMl6bBM

it explains a lot for valve control.. so i think it should be controlled precisely  for heatpump work.. 
also i think it is possible to calculate same valve behaviour.. and prove 175 byte.

rpannekoek, 2022-12-08:
Nice video!  It even explains that a heat pump with a "fixed orifice" (single position EV) will work, but doesn't perform very well.
The more positions the EEV have, the better you can regulate the refrigerant flow and the more optimized the heat transfer will be.  
I think that (combined with differently sized compressors) is the main difference between the variants, but I digress.

geduxas, 2022-12-08:
> Nice video! It even explains that a heat pump with a "fixed orifice" (single position EV) will work, but doesn't perform very well. The more positions the EEV has, the better you can regulate the refrigerant flow and the more optimized the heat transfer will be. I think that (combined with differently sized compressors) is the main difference between the variants, but I digress.

For exampe t-cap series, 9kw and 12kW shares exactly same components. Even compressor power is same.. PCB is same.. but in exposed view pcb have different numbers.. so one thing that could be different is either component's on pcb... Or only firmware..

rpannekoek, 2022-12-11:
I managed to capture an occasion which illustrates byte 175 as EEV position
[image]
Note as when Tdischarge (Discharge_Temp) approaches 110 degrees, EEV (byte 175) increases quite suddenly after which Tdischarge start dropping, whereas Trefr(igerant) (Inside_Pipe_Temp) becomes significantly higher than Tinline (Main_Inlet_temp) and Phigh (High_Pressure) also drops even though Fcomp (Compressor_Freq) does not change at that time. These are all signs that the refrigerant flow suddenly increased, corresponding with byte 175 values.
As soon as byte 175 value drops below 50 again, Trefr is nearing Tinlet again; a sign of optimal heat transfer caused by a lower refrigerant flow.
For completeness, I also included the value of byte 161 (Bypass_Outlet_Temp), which also changes does not behave as a temperature at all (nor would I expect that, since the WH-MDC05J3E5 doesn't have a bypass temperature sensor) but at times (especially during defrosts) behaves a lot like EEV control. However, in this case byte 175 correlates much better to EEV position.

rpannekoek, 2022-12-11:
Byte 161 is shown as raw value (if you would decode it as temperature, you would substract 128, yielding very negative temperatures which jump by tens of degrees).
EEV (byte 175) is shown as "decoded" value, by substracting 1 (as a wild guess for how this should be decoded).
Indeed, the meaning of the values are still quite unclear, other than that higher seems to correspond to valve further open.
The highest value I have observed is 161 (160 "decoded"), which always happens when the compressor stops.
This is remarkable: from what I've read, you want the EEV to fully **close** (stop refrigerant flow) when the compressor stops. 
The step size also increases significantly for the higher values  (for lower values, the step size is 1).

rpannekoek, 2022-12-11:
As a matter of fact, during a longer run I apparently also got raw value 129 once (spike).
Some other observed raw values (from where step size gets more than 1):

Dec | Hex | Bin
-- | -- | --
58 | 3A | 111010
59 | 3B | 111011
62 | 3E | 111110
71 | 47 | 1000111
82 | 52 | 1010010
84 | 54 | 1010100
96 | 60 | 1100000
121 | 79 | 1111001
129 | 81 | 10000001

geduxas, 2022-12-11:
Heishamon, request data update every 5 sec.. so step more by 1 i think is ok.. maybe steps were made faster.. but what about value of 129 (or 128) is also mistery.. as i know stepper motor can go step by step.. but also it can skip step .. or loose it.. it could be in error while it's closing, or opening. Also this value get only as spike..

rpannekoek, 2022-12-11:
FYI: Example of an EEV stepper motor drive sequence diagram (obtained from Danfoss ETS-6 data sheet):
[image]

rpannekoek, 2022-12-11:
Ok, one more observation during a defrost (mode 1 deice):
[image]
(NOTE: different line colors used here)
I've highlighted the defrost period (Defrost_State = 1) and within that where I think (based on Panasonic documentation) the 4-way valve is reversed.
Unfortunately, there doesn't seems to be a direct correlation between 4-way valve state and byte 161 and/or 175 values.
But do not how byte 161 behavior looks like EEV position too.  As a matter of fact, byte 161 looks maybe even closer to EEV position than byte 175 where comparing to the deice mode 1 diagram as documented by Panasonic (albeit a hard to interpret diagram):
[image]
Actually, neither of them exactly match this documentation, but neither does the fan speed, so probably the documentation has to be taken with a grain of salt.

dydek, 2022-12-27:
@geduxas about the schemas you posted here 3 weeks ago - I believe that the schema for WH-MDC05J3E5 is missing something, please check below the connections for WH-UD03HE5-1  and WH-UD05HE5-1 ( which probably uses the same OU main board ) : 
[image]
It has to have 4 connections like a typical step motor has, otherwise it just simply doesn't work.

geduxas, 2022-12-27:
> @geduxas about the schemas you posted here 3 weeks ago - I believe that the schema for WH-MDC05J3E5 is missing something, please check below the connections for WH-UD03HE5-1 and WH-UD05HE5-1 ( which [...]

Yes i agree.. i told same that schemes are incorrect.. or made so for purpose...

dydek, 2022-12-27:
btw maybe you will find it useful that ( I'm almost sure ) under the number 73 of the debug menu (? I'm not sure about the proper name, arrows left + up + right ) there is a value that acts almost like a direct indicator of the position of the EEV - pls check the explanation below - I was recording everything just to get the right moments, also during the normal work, it's changing as expected ( before the defrost the EEV needs to close more and more to decrease the suction pressure/ drop the boiling point  ). 

[image]

geduxas, 2022-12-27:
> btw maybe you will find it useful that ( I'm almost sure ) under the number 73 of the debug menu (? I'm not sure about the proper name, arrows left + up + right ) there is a value that acts almost [...]

Nice find, but compering to byte 175 its little bit off.. around -4. But while monitoring thouse debug menu i have noticed values 71 and 72 are same as for 177/178 bytes.. so it could be..

rpannekoek, 2023-01-03:
Interesting! I wasn't aware that the Panasonic Remote Control also has a "debug menu".  But, indeed, the values you plotted in the deice diagram seem to match exactly what I measured for byte 175 (minus 1) during a deice.
Note that it doesn't 100% match the deice diagram itself, but neither does the fan (which does turn on shortly during the deice cycle). I also have a theory that the 4-way valve is switched back to heating mode shortly during the deice cycle (when byte 161 drops back from 97 to 55).  As a matter of fact, I started believing that byte 161 encodes the 4-way valve state (on a WH-MDC05J3E5), but that theory got disproven with the warmer weather (when byte 161 starts toggling between 97 and 55 seemingly randomly).
Regarding the schematics: I'm still in doubt; indeed, I think you need at least 2 EEV poles connected to be able to set 2 positions. The WH-MDC05J3E5 schematics also look really weird with 2 IC8 driver outputs connected together, but not connected to the EEV.  I think this may be just an error in the schematic.  The schematics of other models (including WH-UD03HE5-1) show a second driver IC for the other two EEV poles, which is missing altogether in the WH-MDC05J3E5 schematics.  So, maybe WH-MDC05J3E5 actually has two poles connected and can thus set 2 positions whereas other models can set all 8 positions. The WH-MDC05J3E5 really behaves as if the EEV only has two positions (see, for example, the first graph I posted).

closed 2024-09-12 by Egyras