heishamon/HeishaMon #300: Feature request: Set command for Top14 Outside Temperature

Tom54497, 2022-12-15:
The position of the outside temperature sensor is a pain. Icing causes a much lower value than it is in reality and forces a too high calculated outlet temperature which results in a permanent icing state of the heatpump.

Will be great to have an independant external sensor at a perfect position. The value could be written via a set command instead of the internal sensor.

Tom54497, 2022-12-15:
Sure. Deicing is not dependant of the outside temperature value. But that os not the point. Due to icing a lower outside temperature will be measured by the internal senor at the back of the heat pump. That causes a too high calculated outlet temperature on the heatcurve. That means more power and a quicker icing at higher humidities.

Right now I have three different outside temperatures:
-7 degrees by Jeisha due of permanent icing
-3 degrees by UVR16X2 sensor which controls the outlet temp of the heatcurve
-4 degrees by homematic outdoor sensor

Goalwill be to have only one value which controls all heat curves, the one at UVR16X2 and Jeisha.

 Homematic outdoor value can be send without any trouble to UVR16X2 via CMI but not to Jeisha.

geduxas, 2022-12-15:
And using external outside temperature sensor is not option? to keep it from icing?

PGnl, 2022-12-15:
In general, I do not get why your temp is 3 degrees below actual ambient and especially only when defrost is necessary. I have never seen that behaviour, nor with me or anyone else that I can remember.
Are you sure you are not “recycling “ your air. The temp of recycled air downward trend is typically “broken” during de-icing. 
this as the fan is stopped and cool air is no longer added. 
I have seen that at my place when I had that issue some years ago.

Tom54497, 2022-12-15:
For me the reason is that the internal sensor is mounted too close to the icing area of the cooling register. Furthermore it is mounted vertically on the metal housing if the pump.

Right now I have deicing cycles of 30 minutes. After deicing the pump powers that high that the next cycle is very close. It won't reach the optimum outlet temp any more. It is a devils cycle.

Tom54497, 2022-12-15:
Sure that occurs at high humidity only which right now is close to 90 percent. But under that circumstances it is impossible to heat my house consequently

PGnl, 2022-12-15:
As far as I remember, the sensor is clipped to the housing. Try to unsnap it and have it stick outward horizontally. Maybe some tape to have it stick out. This creating some distance to the ice. 
First check if that solves the issue.

Tom54497, 2022-12-15:
THX. Had that idea as well

PGnl, 2022-12-15:
Note that all pana’s have it at the same place. Would be strange if you are the only one having this issue (haven’t heard of it before)

geduxas, 2022-12-15:
Strange, in my area also high humidity.. now with -15 it's 92-98%. Could you post some photos? Maybe there is too much ice/snow and unit isn't defrosting enough?

IgorYbema, 2022-12-15:
I had that once also. My 9kW unit doing a mode-2 defrost and after 30 mins it didn't clear all ice. This was easy solved by blocking the hard and cold wind which was supercooling the radiator. I just placed some wooden plate in front of it to deflect the wind and after the next mode-2 defrost all ice was gone.

HighlyCompressedAir, 2024-09-11:
That's not something to be implemented in Heishamon, Panasonic supports external outdoor sensor that can be used if the built-in one is inaccurate for some reason.
I believe this issue can be closed.

closed 2024-09-12 by Egyras