heishamon/HeishaMon #301: Command Topic for Outlet Water Target Temperature

pekleinter, 2022-12-15:
Is there any chance including Command Topic "Outlet Water Target Temperature? This would allow a direct control of water heating temperature when heating/cooling curves are disabled.

geduxas, 2022-12-16:
Yes, in direct heating mode you control direct target temperature..

pekleinter, 2022-12-16:
Correct, but is there an mqtt publish command topic in HeishaMon dedicated to this? As a work around I can imagine using Set16 (SetCurves) command instead.

geduxas, 2022-12-16:
I think you are little wrong, or asking not what you want.
So you would like to use Compensation curve, and by able to control Direct temperature? It's wrong, and pump won't allow that.. in compensation curve, you only have ability to shift curve by -5 to +5 degrees. If you wan't to control outlet water temperature directly, so yes, you set pump for direct mode, and control temperature from heishamon or other system.

pekleinter, 2022-12-16:
I am confused. What is Set5 (SetZ1HeatRequestTemperature) for? Setting/shifting of room or water target temperature? On the other hand Set16 (SetCurves) is dealing with water target temperature.

geduxas, 2022-12-16:
It's depends on heating configuration, if it set to curve, so it works as shift value, if it's set as direct, so it depends on heat sensor source.. if it set to room tlsensor, you wil set room temperature, if it set to water, so you will control water temperature.

pekleinter, 2022-12-16:
Ok, thank you for your patience with me. So, I want to set the water outlet temperature 36 ºC. What is the command for it?

closed 2022-12-17 by pekleinter