heishamon/HeishaMon #302: Telegraf MQTT translation

4920441, 2022-12-21:
Hi,

I try to connect my influxdb with telegraf, which requests mqtt topics from my mqtt server.

If thy are encapsulated in json, that works out of the box.

If it is done like HeinshaMon does it (correctly, I wanna mention) it does not work out of the box

It has to be done somewhat like that:

`Telegraf config:

topic_measurement = "measurement/_"
topic_tags = "_/site"
data_format = "json_v2"
[[inputs.<plugin>.json_v2]]
    path = ""
    [[inputs.<plugin>.json_v2.tag]]
        path = "device"
    [[inputs.<plugin>.json_v2.field]]
        path = "rpm"
        type = "int"`

Since that is very time consuming, Is there a way to output the mqtt values in a different way on heishamon  or maybe has someone a valid telegraf mqtt Plugin Config regarding displaying it in grafana?

Thanks  a lot!

Cheers,

4920441

geduxas, 2022-12-21:
Hi, i am using heishamon with telegraph without issues :/ jus subscribe to main topic ang get all values. Indeed json could be better solution, but for that it will require to change all endpoints..

geduxas, 2022-12-21:
Here is mine telegraph config 

[[inputs.mqtt_consumer]]

   servers = ["tcp://{IP-MQTT}:1883"]
   ## Topics that will be subscribed to.
   topics = [
     "panasonic_heat_pump/main/#",
     "panasonic_heat_pump/s0/#",
     "panasonic_heat_pump/1wire/#",
   ]

4920441, 2022-12-21:
Hmm... nearly the same as I tried it in the first place.

Thanks  a lot, So I just debug a little further...

4920441, 2022-12-22:
Hi,
would you mind to share your  

[[outputs.influxdb]]

config with me?  Maybe thats the difference in config.

Thanks !

Cheers

4920441

geduxas, 2022-12-22:
It's almost blank.. i see that i didn't post whole config in that section. Forgot in the end i have of mqtt topics i have this:

   data_format = "value"
   data_type = "float"

closed 2024-09-12 by Egyras