heishamon/HeishaMon #303: Demand control with optional PCB installed

burt2902, 2022-12-22:
First of all, thank you for your great work that many of us have fun and pleasure using.
I'm looking for a way to play with demand control and I was wondering if it was possible to send demand control messages from HeishaMon to the HP with optional PCB installed. As far as I understand from the [optional PCB emulation documentation](https://github.com/Egyras/HeishaMon/blob/master/OptionalPCB.md), in such setup, HP is connected with optional PCB and then with HeishaMon.
Q: Will it be possible to send PCB emulation messages to HP or will those be discarded by the physical optional PCB? Or maybe there is a different way to get access to setting demand control?

MiG-41, 2022-12-22:
This will not work.  Optional PCB is sending very frequently ( more then one per second) it's infornation packet to HP (also with information about demand control voltge) , so even you will send something with Heishamon  , Optional PCB imidietly overwrite it for his own information (about it's true voltage).
If you have Optional PCB the only way (and maybe even better) is to give proportional voltage 0 -10V to terminals. 

But first try this function manually ( i.e.) via some potentiometer ) , since using this function is not so color as it's sounds.... (i.e. you will not see a difference , when there is around 5 and more degrees outside , you will see limitations when there is -5 and below...) Shortly this function limits the compressor current...

IgorYbema, 2022-12-22:
And to be clear, emulation of optional PCB from heishamon doesn't work with optional PCB already installed. You can only use normal heishamon functions then, not the emulation anymore.

burt2902, 2022-12-22:
Thank you Marcin and Igor for clarifying.

closed 2022-12-22 by burt2902