heishamon/HeishaMon #307: Add watchdog-reset to code

DUD-74, 2023-01-12:
Hi

I have been using HEISHAMON for quite some time and it is working perfectly. 

However, from time to time, HEISHAMON stops working and a can make it work again by plugging out and in again. It might be related to bad wifi-reception, but this can occur  sometimes (e.g. if you fiddle around with the wifi-router).

Suggestion: Could you please enable the inbuilt ESP8266-watchdog to automatically restart the device if it hangs?

IgorYbema, 2023-01-12:
Hi,
A microcontroller as the ESP8266 doesn't hang like a computer. There is already a watchdog in it which reboots it on a real bug (but nog 'hanging').

Also it should not hang. Older code used to have a reset on wifi disconnect but it newer we removed it as it should not happen at all. Heishamon reconnects to a new wifi if available and if not present the hotspot again. I have fiddeled around with my own wifi last few weeks but always heishamon got back without rebooting.
So what you see is probably due to other issue's you have. Please try to describe more what you are doing which needs a reboot of heishamon. Does it present a hotspot? what is the logs of heishamon just before the disconnect (check mqtt log from heishamon topic).

DUD-74, 2023-01-12:
Hi Igor

Thanks for your fast response! In fact, I will try to gather more data next time heishamon becomes unresponsive. In case I have useable infos, I will open a new issue. Thanks alot!