heishamon/HeishaMon #309: Can't build with esp8266 v3.1.0

ddokupil, 2023-01-13:
I know this isn't really an issue with code but it's something I can't overcome. At first I tried to build the project with standard Arduino IDE but quickly realized I have additional buld flags unset. Couldn't find a way to set them using IDE so moved to arduino-cli.
The thing is it keeps downloading latest esp8622 version which is 3.1.0 at the moment and I can't force 3.0.2. even when using `arduino-cli core install esp8266:esp8266@3.0.2 --additional-urls https://github.com/esp8266/Arduino/releases/download/3.0.2/package_esp8266com_index.json -v` it keeps using `INFO[0004] Loaded platform                               platform="esp8266:esp8266@3.1.0"`

And I woul'd even care if 3.1.0 worked but it doesn't. Can't build the thing cause I'm getting
`C:\Users\TwojaStara\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.0\cores\esp8266/core_esp8266_features.h:132:38: error: expected ',' or ';' before 'esp_is_8285'
  132 | bool __attribute__((const, nothrow)) esp_is_8285();`
Which makes no sense but I'm out of ideas why is this happening.

So I'm asking you guys to help with either of these issues, to build it with IDE or to set v3.0.2. Or maybe problem is elsewhere.

IgorYbema, 2023-01-14:
Yes i noticed this yesterday also. This release is in one week old and also there is a change to fix this after 3.1.0. I think they will release a new version soon. You can also stick with 3.0.2 for now

IgorYbema, 2023-01-14:
Owh in Arduino IDE make sure you switch MMU to the 16kb- not shared option. And non 32-bit access to the 'very slow' option.

ddokupil, 2023-01-18:
Managed to clean some caches manually and forced 3.0.2 and built perfectly in IDE using above options.

IgorYbema, 2023-01-18:
FYI, 3.1.1 is now available also and you can build it with that

closed 2024-09-12 by Egyras