heishamon/HeishaMon #311: Rules questions

kr0815, 2023-01-15:
Hi Igor,

long time no see - Heishamon is running like a charme here :-)

There are few very engaged folks here in germany that try to improve the internal Heisha algorithms using your software, especially the rules function
Goal is a independently running system, no external control by Openhab and such things

I try to help them, translate their questions, and publish them here in Github

So maybe you could help?

Their questions:

- if there is a running timer, could it be somehow stopped by another rule? 
- is it possible, when system boots - we have a rule "on System#Boot..."  to wait with rules interpretation till all values are collected from Heisha?
- is it possible to alter heating curves from / within rules?
- could the value "Watt" of an S0-Counter  be fetched in a rule?
Hope you / we could help them

Best regards

IgorYbema, 2023-01-15:
@CurlyMoo would you help in this discussion?

CurlyMoo, 2023-01-15:
> if there is a running timer, could it be somehow stopped by another rule?

Yes, but making the seconds zero:
```
setTimer(1, 0);
```

> is it possible, when system boots - we have a rule "on System#Boot..." to wait with rules interpretation till all values are collected from Heisha?

You can check for values using `isset` or by comparing the to `NULL` whatever you prefer:
```
if isset(@Heatpump_State) or @Heatpump_State != NULL then
....
end
```

> is it possible to alter heating curves from / within rules?

If it can be controlled from the REST API / MQTT, you should be able to control it by the rules. Syntax is the same.

> could the value "Watt" of an S0-Counter be fetched in a rule?

Not at this moment.

McMagellan, 2023-01-16:
Thanks for your information.
"If it can be controlled from the REST API / MQTT, you should be able to control it by the rules. Syntax is the same. "

Can you give me an example what the Syntax is for this commands in Rules?
http://192.168.178.36/command?SetCurves={"zone1":{"heat":{"target":{"low":28}}}}
http://192.168.178.36/command?SetCurves={"zone1":{"heat":{"target":{"high":37}}}}

In my Rules i will find a second way to transfer my own calculated target temperatur cause TOP27 is only for use with weekly timer.

JockelBln, 2023-01-16:
> The only option is for @IgorYbema to split the set values instead of having one single SetCurves command. This would be the most easy implementation.

If that were really easy, I would be very grateful for an implementation.
I would also like to regulate with rules and keep the functionality of the weekly timer.

CurlyMoo, 2023-01-16:
>I would also like to regulate with rules and keep the functionality of the weekly timer.

I don't understand this sentence.

McMagellan, 2023-01-16:
Here is the sketch of what we want to do. 
Excuse my bad English
[image]

CurlyMoo, 2023-01-16:
I don't get it. Do you @IgorYbema ?

1N4148, 2023-01-16:
Basically they want to slow down the reaction to outdoor temperature changes. So they're calculating the target temperature with a damped outdoor temperature. If they write it directly into the target temp, they cannot use the heat shift by the week timers anymore. So, they are looking for a way to write the new target temperature as a modified heat curve.

CurlyMoo, 2023-01-16:
Aha, now i get it. It doesn't change what i said before. For it to work, we need seperate set topic for the individual curve values.

JockelBln, 2023-01-16:
> Basically they want to slow down the reaction to outdoor temperature changes. So they're calculating the target temperature with a damped outdoor temperature. If they write it directly into the [...]

You explained it correctly, that's what we intend to do.

JockelBln, 2023-02-24:
> The only option is for @IgorYbema to split the set values instead of having one single SetCurves command. This would be the most easy implementation.

@IgorYbema Is there any hope that this will be implemented?

CurlyMoo, 2024-04-16:
Setting the curves inside the rules should now be possible with the latest code in this PR:
https://github.com/IgorYbema/HeishaMon/pull/121

It should still be tested though.

Setting the curves is then done like this:
````ruby
@SetCurves = concat('{"zone1":{"heat":{"target":"high":', @Z1_Heat_Curve_Target_High_Temp, ',"low":', @Z1_Heat_Curve_Target_Low_Temp, '}}}}');
```

closed 2024-09-12 by Egyras