heishamon/HeishaMon #312: Demand control at Jeisha

Tom54497, 2023-01-18:
Although it is in experimental state I would like to know if somebody is there where the whole thing works?

I can set any value but no limitation of the compressor current will take place.

Besides that heishamon is working fine!

MiG-41, 2023-01-18:
First ,you have to enable in HP Optional PCB ,and Demand control (booth to yes)
Also in Heishamon you have to enable Optional PCB...

Then you have to wait for a cold period ( in 0 deg ambient you maybe will notiice only with DHW )

Tom54497, 2023-01-18:
@MiG-41 Of course that was done before. Heishamon is fully implemented in IOBroker aswell! Zero temperature is also given.

ddokupil, 2023-01-18:
That is an interesting topic. I have an optional PCB but still am interested in how does this thing work. Manual is pretty laconic:
```
12.27 Demand Control (Optional PCB)
   Remote control setting:
      When Optional PCB connection select 'YES", Demand Control function can select "YES" or "NO".

   Purpose:
      After the demand control select YES, below control will activated
      0-10V Demand control

   0-10V Demand control
      Demand control is use to reduce the current usage of heat pump unit by third party device.

   Control start condition:
      Select "YES" at Demand control at installer menu.
      0-10V input for this electrical current control is detected.

   Control content:
      If start condition is fulfilled, indoor will receive the voltage signal from optional PCB. Indoor will send the rateovalue to outdoor unit.
      Outdoor will change the current limit according to the percentage receive from indoor unit.
```
So, if I apply 5V to the demand input I will force my HP to work on 50% regardless and overriding any other conditions? Or will it only set the limit never to work above 50%?

fruehwi, 2023-01-18:
Is it possible to get the current demond  value and smart grid value of the Optional PCB emulation, per mqtt?
Or is that a new requirement? It make it easier for handle any updates in e.g. iobroker scripts.

MiG-41, 2023-01-18:
I'll describe on my example T-CAP 12kW gen.H.
When i set Demand control to min value ,i reduce compressor current to 3.2A....
And at the same time at 0 deg outside at full demand current compressor value is 3A. So there have to be -5deg ,when i find just a Little bit limitation ( at  full demand 3.4A ,at min demand 3.2A)
When it is -10deg , full demand is 4A ,and at min demand 3.2A.... And so on . 
During DHW And 0 deg outside ,at full demand (at the end of dhw heating) j have compressor curent 4A while at min demand it will redutę to 3.2A
So as you can see it is completly no dependent from frequency ,power produced ect.

geduxas, 2023-01-18:
@Tom54497 by manual, demand control is same in J and H series. Difers only SGReady (if i remember it correctly) terminals, which you can set as push button's and reduce pump usage by given scenario

MiG-41, 2023-01-19:
@Tom54497 if you know how J behaves,why you asking ...
Optional PCB is the same for J and H, and it sends only the commands to main PCB (values from 43 to 231 translated linear from 0-10V aplied to its terminals). 
And how HP reacts on this values is in main PCB...

Tom54497, 2023-01-19:
Obviously you did not understand my question. Original PCB is working with demand control but Heishamon not. I do own both in exclusive use. J series can not be limited that low as H series. That is the finding of several people on the net.

Again.... did anybody get demand control with Heishamon to work with J series?

geduxas, 2023-01-19:
I have optional pcb, now need to find out controller 0-10V and just test it.. i am doubt that Heishamon emulation is doing something wrong... Indeed never tested that..

MiG-41, 2023-01-19:
I tested Demand Control on Heishamon.
Yes Set the values in mqtt from range 43 to 255.
It was also confirmed by other user ,that it behavies exacly the same as a real Optional PCB.
And Yes, on H series.
 @Tom54497 if your HP works with "orginal PCB" (In meaning  real Optional PCB) and not with Heishamon ,what is the difference ?

naseweiss01, 2023-01-21:
Just a word of warning, my J-T-Cap 12 interprets the low value of "43" still as "0". Means no reduction. It took some time to realise this. Therefore I am starting with "50" as lowest value, to be on the safe side.

Safs, 2023-01-27:
How can I send data for Demand Control in mqtt?

MiG-41, 2023-01-27:
Just pubish topic panasonic_heat_pump/commands/SetDemandControl with particular value .... like 50 ( also first enable OptionalPCB and Demand Control in HP , and enable emulation in Heishamon).

Safs, 2023-01-27:
If I enable OptionalPCB and Demand Control in HP, enable emulation in Heishamon, but not pubish topic panasonic_heat_pump/commands/SetDemandControl with any value, then HP will turn off?

MiG-41, 2023-01-27:
Default value from Heishamon is 234 (EB in hex) with correspondes to 10V

mlapp, 2023-01-27:
Weiß jemand, ob die Leistungsbeschränkung auch beim Enteisen funktioniert? Das wäre klasse, weil bis 55Hz ist der 7kW Monoblock relativ leise. Wenn er nach dem Enteisen aufheizt, läuft der Kompressor mit 69Hz und dann wird sie relativ laut. Das würde ich gerne unterbinden.

Does anyone know if the power limit works with de-icing as well? That would be great, because the 7kW monoblock is relatively quiet up to 55Hz. When heating up after defrosting, the compressor runs at 69Hz and then it gets relatively noisy. I would like to stop that.

Safs, 2023-02-02:
If I will install real Optional PCB, then is it will be possible still sending demand commands via MQTT?

MiG-41, 2023-02-02:
Not possible. 
Comands in this case would be real status or sensor resistances...

geduxas, 2023-02-02:
> If I will install real Optional PCB, then is it will be possible still sending demand commands via MQTT?

Why you need set it with optional board? Just add external 0-10V device

Safs, 2023-02-02:
Now I have all already configured via HomeAssistants without PCB, but I want real link with room thermostat, not virtual (for security of heat supply reason), therefor think about real PCB, but I don't  want change my existing system for demand response.

geduxas, 2023-02-02:
> Now I have all already configured via HomeAssistants without PCB, but I want real link with room thermostat, not virtual (for security of heat supply reason), therefor think about real PCB, but I don't want change my existing system for demand response.

i have optional pcb to, and also want to test demand control.. so have ordered some 0-10V Wifi/Zigbee controllable devices.. i don't think there will be problems to change from one control device to another in your automation rules

Safs, 2023-02-02:
I agree, it certainly won't be difficult, but something has to be done anyway, but I'm lazy...
I have a spare ESP32 module, will probably have to use it...

geduxas, 2023-02-04:
Today i got ZigBee dimmer for 0-10V and attached to optional PCB to test it out.. and.. truely i am not happy with that at all.. yes, it's works, it's reduces consumption but in unknown and low values.. as @MiG-41 told..

Also it will effect and DHW heating. While heating DHW i lowered dimmer to 10% so it should be around 1V and compressor frequency dropped by 10-12Hz... Also while heating house it wasn't effected anything, just a little bit lowered start frequency.

piposant, 2023-02-09:
I just acquired a heishamon, in read mode everything is ok. When you enable in HP Optional PCB ,and Demand control to Yes.
also active on Heishamon Optional PCB.
When I activate the options in HP, it does not turn on when it receives the order from the thermostat to start. Could you help me?
I want to be able to manage demand control from Heishamon.

geduxas, 2023-02-09:
In read only mode it will not work

piposant, 2023-02-09:
So now the only way to activate HP is via MQTT? The physical thermostats can no longer activate it?

piposant, 2023-02-09:
Okay. 
All these "SetExternalThermostat1State" and "SetDemandControl" are not implemented in the HACS integration for homeassitant?

IgorYbema, 2023-02-09:
I don't use it so I don't know but if they aren't it should be easy to add. Use other Set topics as example switches.

Safs, 2023-02-09:
> Okay. All these "SetExternalThermostat1State" and "SetDemandControl" are not implemented in the HACS integration for homeassitant?

You can use it with MQTT commads.

Safs, 2023-02-09:
For example my automation command:
`alias: Siltumsūknis PCB On
description: ""
trigger:
  - platform: state
    entity_id:
      - climate.netatmo_smart_thermostat
    attribute: hvac_action
    to: heating
condition: []
action:
  - service: mqtt.publish
    data:
      topic: panasonic_heat_pump/commands/SetExternalThermostat1State
      payload: "2"
      qos: 0
      retain: false
mode: single
`

geduxas, 2023-02-09:
> Okay. All these "SetExternalThermostat1State" and "SetDemandControl" are not implemented in the HACS integration for homeassitant?

No HACS is only starting.. and yes it's in basic support for now.

piposant, 2023-02-09:
ok thanks.
@Safs: sorry for my ignorance, but where do you create mushroom automations? nodered? code? where?

Safs, 2023-02-10:
Sorry, this is Homeassistant autattion.

he-so, 2023-11-28:
> Just a word of warning, my J-T-Cap 12 interprets the low value of "43" still as "0". Means no reduction. It took some time to realise this. Therefore I am starting with "50" as lowest value, to be on the safe side.

My T-CAP J 9KW machine accepts Value 43 very well. Attached, there is a graph showing the regulation characteristics when using the minimal setting (43, also referred to as 5%) This will lead to compressor RPM of less than 30Hz at 0°C ambient temperature, regardless of the requested target temperature, it will not arrive there for hours. So PID regulation is somehow disabled. Later, at 20:35h I set the value to zero (/command?SetDemandControl=0) and the pump runs without limits to meet the target temperature. So, it works fine for me. But is there a difference between value 0 and 231?

[image]

MiG-41, 2023-11-28:
It would be good to add on your chart Compressor Curent , then you would see , that this is the real value ,with SG drives.

An value out of range 43 - 231 are treated as 100% ( or SG off ).

HighlyCompressedAir, 2024-09-11:
[image]

I think it was explained well enough, working as intended, issue can be closed.

geduxas, 2024-09-11:
Truly it will be always asking about it.. because demand control works bit different when it's explained in service manual.. 

Better is to write some FAQ manuals and post in front page..

HighlyCompressedAir, 2024-09-11:
I don't think it works differently. It's just this part:
`*A minimum operating current is applied on each model for protection purpose.`
Minimum operating current is absurdly high and that makes people confused.

closed 2024-09-12 by Egyras