heishamon/HeishaMon #313: Calculate power production using external temperature sensors

JohannesKlug, 2023-01-23:
Using external DS18B20 temperature sensors for inlet and outlet, and using the pump flow rate reported by the heatpump, I think it would be possible to calculate a more realistic value of heat production than the one reported by the heatpump itself.

Has anyone done something like this yet, or is there even code for this already in HeishaMon?

IgorYbema, 2023-01-23:
Yes sure. That is the reason why Heishamon has the temp sensors include. And with the S0 kWh meter you can calculate a better COP also.
Heat production = delta T x flow (in L/H) x 4096 (which is a constant for the specific heat of water times the density)
So for a heishamon report TOP1, flow rate of 20 L/Min (which is 1200L/H) and a temp difference of 5 you get:
Heat production = 5 * 1200 * 4096 =  24.576.000 Joules. To calculate to kiloWatt divide that with 3.600.000. So you get about 6.8 kW production.

MiG-41, 2023-01-23:
But you have to do calculation in your automation system , Heishamon is not doing it (yet :) )
You have to also do some trigger for on/off ( i.e. Compressor freq>0) ,since flow have not 0 , so in long period power is increasing without running compressor.

JohannesKlug, 2023-01-23:
Great, many thanks for the responses. I'm using Home Assistant currently -- I'll try to do the calculations there. If I have something working, I'll share it via the integrations.

geduxas, 2023-01-24:
If someone curious, i have added energy calculations with calibrated dallas sensors for my t-cap heat pump.. and it's doesn't seams that heatpump lies much :D yes i need to monitor it more, but here is first results.. I attached dallas sensors in inlet/outlet pipes of heatpump, and also for interest attached to Zone2 pipes.

[image]

Candide1976, 2024-01-27:
> If someone curious, i have added energy calculations with calibrated dallas sensors for my t-cap heat pump.. and it's doesn't seams that heatpump lies much :D yes i need to monitor it more, but here [...]

Could you share the rule generating this graph?

geduxas, 2024-01-27:
It's home assistant history card. Nothing special, also you could use grafana for that