heishamon/HeishaMon #314: Wrong S0 metering, only bad pulses

Alkazar-DH, 2023-01-24:
Today I discovered a false power value in my smart home system, I checked Heishamon and found a lot of bad values in the console.
[image]
I using Heishamon V3.0 and it worked perfectly for around a half year.
Any suggestions how to solve this problem?

IgorYbema, 2023-01-24:
This is because the line is picking up noise from around. Please use better cabling (shielded signal cable) and maybe add some ferrite beads to the signal line.

jacco-hass, 2023-12-09:
I've also got this issue, even when using STP cabling and ferrite beads. I checked the pulses from my kWh-meter with an oscilloscope, which looks good (exactly 80ms pulse):
[image]

Any ideas what might be the issue? At first I suspected a dirty ground of the Heishamon board, but I tested different types of power supplies and the issue remains.

closed 2024-09-12 by Egyras