heishamon/HeishaMon #315: So and 1Wire not working in 3.1?

kr0815, 2023-01-30:
I upgraded to 3.1 - didn´t see any 1Wire-sensors, also pulse counter counted nothing

After switching back to 3.0, it works again?

MiG-41, 2023-01-30:
1wire worked at 3.0 , and now after update to 3.1 working as usual... 
Dont use S0...
Can you wrote more about your proble?

geduxas, 2023-01-30:
I am on custom built but based on 3.1 no issues with 1wire.

IgorYbema, 2023-01-30:
Should work and works for me

kr0815, 2023-01-30:
I had an old Version of Heishamon, about 1 year old, don´t know which version

After the upgrade to 3.1 -
- no 1Wire Sensors are present
- the S0 sensor is strange, reports mucg to high values

After downgradng to 3.0

- 1Wire sensors are back as expected
- so shows also strange readings

Messiam, 2023-01-31:
Is there a change list from 3.0 to 3.1? 
I'm still on 3.0

geduxas, 2023-01-31:
> Is there a change list from 3.0 to 3.1? I'm still on 3.0
Only viewing PR commits.. 

Some new topic's exposed.. truely nothing special for regular use.

IgorYbema, 2023-01-31:
You can find the changes in the PR: https://github.com/Egyras/HeishaMon/pull/310

kr0815, 2023-02-02:
result after 1 day:
Back to 2.0 - working like expected

[image]

thenebu, 2023-08-24:
I got the same problem. I bought the Heishamon 4.7 PCB with Firmware 3.1 on it. 
Everything worked except the s0 meters. I use Eltako wsz15d-32a meters with 2000 imp and 30ms. 
When i found this Issue, I installed the 3.0 firmware. Without any changes the s0 meters worked fine. So it must be the 3.1 firmware causing the problems.

Thu Aug 24 10:22:21 2023 (3865133): Pulses seen on S0 port 2: Good: 614 Bad: 0 Average good pulse width: 32
Thu Aug 24 10:22:21 2023 (3865134): Measured Watthour on S0 port 2: 1.00
Thu Aug 24 10:22:21 2023 (3865141): Measured total Watthour on S0 port 2: 307.00
Thu Aug 24 10:22:21 2023 (3865147): Calculated Watt on S0 port 2: 896

thenebu, 2023-10-24:
With Firmware 3.2 the S0 is working fine!

closed 2024-09-12 by Egyras