heishamon/HeishaMon #316: TOP 99 buffer control

Grufti68, 2023-02-08:
Hello,

is there a possibility to control the TOP99 buffer?

Grufti68, 2023-02-08:
I have a buffer memory with my Panasonic. if I start hot water production with the buffer deactivated, my heating pumps switch off. but if I activate the buffer then the pumps continue to run. I want to control it via IO-Broker

MiG-41, 2023-02-09:
Buffer memory is simply buffer for extand water capacity ? (since memory is the word i don't understand in this context...)
Hot water production - DHW ,has nothing to deal with buffer , 3-way valve "connect" Heat Pump directly with DHW tank and therefore it is heating without buffer. Is your system coniguration is really ok ? Maybe you can pass some scheme of it ...

geduxas, 2023-02-09:
@MiG-41  i think he talks about buffer tank and pumps after buffer tank with extension board.. or maybe 2 zones. Yes if buffer tank is enabled in DHW heating, zone pumps will run.. but without buffer tank i think they are stoped. 
@Grufti68  why you need to disable buffer tank?

MiG-41, 2023-02-09:
No , they are  not stopped.  I have such system with 2 zones , without buffer (phisicaly ,and disabled in HP ) , and two zones works.

geduxas, 2023-02-09:
> No , they are not stopped. I have such system with 2 zones , without buffer (phisicaly ,and disabled in HP ) , and two zones works (and zone pumps run).

oh, i have also 2 zones with extension board and buffer tank, didn't test that setup disabling buffer tank, just thought that should be normal.. becouse if you close 3 valve, and you lose water circulation source for thouse pumps.. so it would be logicaly to turn off.. it could be still running but you will need to close 3 way valve for zones.. otherwise from pumps will left without circulating water?

MiG-41, 2023-02-09:
@geduxas , can we go to slack , just to not talk in this thread ?
First , i need to understand , if for you 3valve is the  mixing valve on zones , or 3way valve for DHW ?:)

Grufti68, 2023-02-09:
i have send a massage in slack

Grufti68, 2023-02-09:
@geduxas
 i have a buffer for 2 zones and i have a water tank.
I use the SG-Ready for the PV surplus.

Normally I work without a buffer but when the SG-Ready is active I want to activate the buffer to save more energy

sorry for my english i use a translator

IgorYbema, 2023-02-09:
Ok I understand. Let me see if I can add that

IgorYbema, 2023-02-09:
I have created a test build for you:
https://github.com/IgorYbema/HeishaMon/actions/runs/4137527245
Check the build binary and install it.

In https://github.com/IgorYbema/HeishaMon/blob/setbuffer/MQTT-Topics.md at SET28 you can see how to use it.

Please let me know if this works for you

Grufti68, 2023-02-09:
WOW that was fast, yes it works. Thanks a lot for your help

Grufti68, 2023-02-18:
Hello, I have the following problem.
Buffer memory programming is built in, but I always get an error message about Node Red when I start up. What could be the problem?
[https://panasonic-wemos.slack.com/files/U04NNEYPW2Z/F04Q95RFHMY/image.png](url)

IgorYbema, 2023-02-18:
I'm not sure, but why are you using port 8003? Is it forwarded in an upstream router or something like that? Can be that that doesn't work properly

Grufti68, 2023-02-18:
It is connected to a range extender, I can reach heishamon via the port, it also works for all commands but not for buffer

Grufti68, 2023-02-18:
but it makes no difference whether I am directly connected and run the command or not. the console doesn't work for me either

IgorYbema, 2023-02-19:
Ok so, the new command doesn't seem to work. Also if you are sending the command manually (send the url)?

IgorYbema, 2023-02-19:
I see that it also say 'external pad heater'. I think I forgot to change something :)

Grufti68, 2023-02-19:
if I send the command as a url via the browser, it works. Do I need a break between recommendations? I send 5-7 commands at a time

IgorYbema, 2023-02-19:
Can you try? Not sure but indeed it works here also manually (even it says wrong text)

Grufti68, 2023-02-19:
it works via url but via node red I get an error message (see above)

IgorYbema, 2023-02-19:
I'll try with node-red also.

Grufti68, 2023-02-19:
I suspect that it just has to do with the wrong feedback

geduxas, 2023-02-19:
> I suspect that it just has to do with the wrong feedback

In slack you told that you send multiple set commands one after another.. is it related to this wrong feedback? Or you have same error after setting only this one?

Grufti68, 2023-02-19:
I also have this error when I only send the command one at a time

IgorYbema, 2023-02-19:
I am builing a changed version here (with the correct response): https://github.com/IgorYbema/HeishaMon/actions/runs/4215449652

Grufti68, 2023-02-19:
It works with Node Red, Blockly and url so far. The status message via url is also correct. I will test it further and will report if it doesn't work. I also have Correcr receives data 100% for the first time. Thank you

closed 2023-02-19 by Grufti68