heishamon/HeishaMon #318: Activating holiday mode

Safs, 2023-02-15:
Is there solution for activating holiday mode from HomeAssitant side? With that predefined switch not working, because HP need to be set start date/time and end date/time. Is there some MQTT commands for that?

geduxas, 2023-02-15:
> Is there solution for activating holiday mode from HomeAssitant side? With that predefined switch not working, because HP need to be set start date/time and end date/time. Is there some MQTT commands for that?

I think it's controlled from display only. So i think displays only controls pump as scheduled task.. i think you can implement it in your system.

curzon01, 2023-02-15:
you can enable/disable holiday mode using [SET2 command](https://github.com/Egyras/HeishaMon/blob/master/MQTT-Topics.md#command-topics) e. g. topic  anasonic_heat_pump/commands/SetHolidayMode 0|1

there is currently no possibility to set the holiday mode parameter start/end time

Safs, 2023-02-15:
Without setting date/time, HP disabling holiday mode by itself, after sended topic panasonic_heat_pump/commands/SetHolidayMode 1

curzon01, 2023-02-15:
It works in any case with FW 3.0 and 3.1 - but the change to enable holiday mode takes a few seconds (here approx. 3-6) - be patient

be sure to use the topic and payload as seperate values, you can't set anything using a topic "panasonic_heat_pump/commands/SetHolidayMode 1" - the "1" is the payload and not part of a topic.

Safs, 2023-02-15:
I understood, but after approx 10-15 sec, holiday mode is disabled by HP.

curzon01, 2023-02-15:
can't confirm - that's only the case if you start/end date are invalid (when they in the past).

Try start/end times in the future, then the state also remains active

Safs, 2023-02-15:
Yes, then it work.

closed 2023-02-15 by Safs