heishamon/HeishaMon #323: Implement command for byte 23

Celphor, 2023-02-19:
Hi there,

I'm optimizing my setup to avoid fast clocking during moderate outside temperatures. I found out that setting byte 23 to 0x95 forces the HP to disable the compressor (according to the protocol description), even without an external switch connected. Zone pumps, mixer and everything else remain working. A setting of 0x56 stops the zone pumps too and drives the mixer to cold, which is very annoying when it comes to restart.
So you can easily stretch out the idle interval and control it via external software. Together with some overdrive during the working-cycle you can get a very efficient and compressor-life saving behavior. Tested with H-generation 12kW split model

IgorYbema, 2023-02-19:
Hi, Thank you.
I just read the information in the manual. This 0x95 to this byte set the 'external compressor control' to on. Which itself is normally a connector on the optional PCB. On this pcb the description is 'dry contact, open=compressor ON, closed=compressor OFF'. So changing only this control setting to on should not disable the compressor yet. It can be done with the emulated PCB however.
But you managed to see it working. Do you have the optional pcb and maybe a wire-bridge connected to that connector?

Celphor, 2023-02-19:
Hi Igor, I have an optional PCB but nothing connected to the connector

Celphor, 2023-02-19:
And: 0x56 shows a message on the display, saying device is locked externally. 0x95 does not show a message.

IgorYbema, 2023-02-19:
Ok that is weird. The manual states that when this compressor switch connector is open the result should be that compressor is on, and closed is compressor off. The byte 23 is only for enabling this setting.

For the 0x56 (probably the external control then) it is indeed open=turn off, closed=turn on (so that seems to work indeed without any cables connected).

IgorYbema, 2023-02-19:
Also manual says that for the compressor=off there is no message on the controller. So it is hard to check. I'll try today with a normal run with that setting on to see what happens.

Celphor, 2023-02-19:
[image]

I set a lock time of 2 hours and it is working fine

Celphor, 2023-02-19:
> The byte 23 is only for enabling this setting.

If I remember right, setting the "external switch enable" using the display did not change the byte. I guess byte 23 only reflects the hardware switch and can also be set via command.

IgorYbema, 2023-02-19:
No byte 23 should be changed. It reflects the setting, not the actual state of the switch.
I just checked at my heatpump and indeed it changes:
If you just switch external control on the byte 23 changes from 55 to 56
If you just switch remote error signal on, it changes from 55 to 65
If you just switch external compressor control on, it changes from 55 to 95.

If you do a mix, it changes to the corresponding bits turned on.

But this does not explain why you are seeing that the compressor is turned off. Maybe the manual is wrong and it turns off when the connector is open. Can you try to install a wire between the 2 connectors of the port (closing the connector). And then see what happens to the compressor (if the byte 23 is on 0x95)

Celphor, 2023-02-19:
I will install a switch to the port later during the day and keep you updated

IgorYbema, 2023-02-19:
Great. I'll try something also but don't have the optional pcb so can only emulate

Celphor, 2023-02-19:
First the simple stuff, the results after setting the options from the display menu:

E = External switch (main board)
EC = External controller switch (optional PCB)

EC disabled in Menu, E disabled in Menu : 0x55  0101 0101
EC disabled in Menu, E enabled in Menu  : 0x56  0101 0110
EC enabled in Menu, E disabled in Menu  : 0x95  1001 0101 
EC enabled in Menu, E enabled in Menu,  : 0x96  1001 0110

Conclusion for byte 23:

Bit1&2: 1 = EC disabled
Bit1&2: 2 = EC enabled

Bit3&4: always 1
Bit5&6: always 1

Bit7&8: 1 = E disabled
Bit7&8: 2 = E enabled

It does not matter if the hardware switches are opened or closed, the result in byte 23 is like described above

Celphor, 2023-02-19:
But now the fancy stuff: Something seamed to have impact on bytes 110 and 116 while i was fiddling around. As these two are in the read only area of the byte chunk, they may reflect the switch status.
I have a listener on all undocumented bytes and these two never changed before.

Seen:

116: 55, A9, A5 <- compressor related ?
110: 55, 56 <- Ext. switch related ?

IgorYbema, 2023-02-19:
You will notice bit3&4 change if you also enable external error signal. Bit5&6 i haven't found yet.

Celphor, 2023-02-19:
Thanks for the hint pointing me in the right direction. Bit5&6 is the Heat/Cool-Switch setting. So byte 23 is fully disclosed :-)

geduxas, 2023-02-19:
> But now the fancy stuff: Something seamed to have impact on bytes 110 and 116 while i was fiddling around. As these two are in the read only area of the byte chunk, they may reflect the switch [...]

Byte 116 is related to zone pump's and 3-way valve. They chnage relative to start/stop zone 1/2 pumps. And switching to DHW

IgorYbema, 2023-02-19:
Byte 111 is already used for 3-way valve status but maybe 116 has something related to it

geduxas, 2023-02-19:
> Byte 111 is already used for 3-way valve status but maybe 116 has something related to it

Yes 116 one of bytes also chanes in same time as 3-way valve.. also it's changes with same delay after you Switch zone off.. after delay it changes in same time as zone pump goes off.. also it was confirmed by @MiG-41 

Byte #116
1st & 2nd bit Zone 2 Pump ( b01 - OFF, b10 - ON ) , 3rd & 4th bit = Zone 1 Pump ( b01 - OFF, b10 - ON ), 5th & 6th bit = 2way Valve ( b01 - Cooling, b10 - Heating) , 7th & 8th bit = 3way Valve ( b01 - Room, b10 - Tank)

IgorYbema, 2023-02-19:
But let's focus back on topic 23. The question was related to turning off and on the compressor. Does it indeed work without anything connected to the fysical port?
As, again, the manual states that an OPEN port will result in compressor=ON and only closing the port will turn off the compressor  (if the setting is enabled). Is this confirmed yet?

Celphor, 2023-02-19:
The manual says:

When the External Compressor Switch is ON:
-a) Heat pump system operate normally
When the External Compressor Switch is OFF:
-a) Heat pump, Indoor water pump & Heater ( Booster heater & Backup Heater ) need to turn OFF
b) Solar, Boiler and zone control can be operate follow normal control condition.* pump delay OFF also included in this control

(There is NO pop up screen like External Switch when this External Compressor Switch is OFF.

Switch Off means "contacts open/no switch connected". So when you enable the setting in the menu (or via serial command), the compressor turns off when nothing is connected or the switch is open. I see no contradiction here.

Celphor, 2023-02-19:
The manual I refer to is https://www.manualslib.com/manual/1273892/Panasonic-Wh-Sdc09h3e8.html?page=134#manual Page 134

MiG-41, 2023-02-19:
Also default optionalPCB command gives compressor off as default :)
And byte #116, can someone check how it behaves when there is only one zone ? Is it only zone 1 pump set to on , or booth ,or none...

Celphor, 2023-02-19:
116:
  2 Zones:
     A9 = Both pumps on / LED on
     55 = Both pumps off / LED off
   Zone 1 only:
     69 = Pump 1 on / LED on
     65 = Pump 1 off / LED off
  Zone 2 only:
     99 = Pump 2 on / LED on
     99 = Pump 2 off / LED off

LED means the led on the on/off-button of the controller

IgorYbema, 2023-02-19:
Check page 74:https://www.manualslib.com/manual/1273892/Panasonic-Wh-Sdc09h3e8.html?page=74#manual
Dry contact Open=Comp.ON,Short=Comp.OFF(System setup necessary)

IgorYbema, 2023-02-19:
> The manual says:
>
> When the External Compressor Switch is ON: -a) Heat pump system operate normally When the External Compressor Switch is OFF: -a) Heat pump, Indoor water pump & Heater ( Booster [...]

Check page 74:https://www.manualslib.com/manual/1273892/Panasonic-Wh-Sdc09h3e8.html?page=74#manual
Dry contact Open=Comp.ON,Short=Comp.OFF(System setup necessary)

Celphor, 2023-02-19:
Well, seems the manual is not consistant. Simply give it a try. For me it works both for the compressor and the whole machine (with optional PCB connected and no switches attached)

IgorYbema, 2023-02-19:
It is consistent. The page which  you are pointing at just talks about the switch on or off, not the electrical state of the switch.

I can not test it because I don't have the optional PCB.

But I can add the set command for byte 23 setting, no problem. Doesn't matter for me what it eventually does to the compressor as it is just a settings ;-)

Celphor, 2023-02-19:
Maybe it works even without the optinal PCB ;-) And it does nothing to the compressor. It will be simply shutdown or not be startet at all.

IgorYbema, 2023-02-19:
I implemented a beta run on: https://github.com/IgorYbema/HeishaMon/actions/runs/4217667071
Topics and Commands in: https://github.com/IgorYbema/HeishaMon/blob/byte23test/MQTT-Topics.md

IgorYbema, 2023-02-20:
> Also default optionalPCB command gives compressor off as default :) And byte #116, can someone check how it behaves when there is only one zone ? Is it only zone 1 pump set to on , or booth ,or none...

The two pumps in byte 116 are not changing in my 1-zone only setup. So these only change in a 2 zone config I think

geduxas, 2023-02-20:
> The two pumps in byte 116 are not changing in my 1-zone only setup. So these only change in a 2 zone config I think

So it could be directry related to external pumps.. don't remember if you have access tu external pump without expansion board? becouse only expansion board is controlling Z1 and Z2 pump.

IgorYbema, 2023-02-20:
There is also one external pump connection on the main pcb. I use that for my floor pump. Run always together with main pump. But does not seem to be this value.

closed 2024-09-12 by Egyras