heishamon/HeishaMon #325: Byte 118 decoded

Celphor, 2023-02-20:
Hi,

Byte 118 is encoded as follows:

Bit 3,4 and 5 are the fraction of the outlet temperature in quarter Kelvin + 1
Bit 6,7 and 8 are the fraction of the inlet(return) temperature in quarter Kelvin + 1
Bit 1 and 2 are always zero

Bit              12   345   678
Pattern     | 00 | 000 | 000 |

Outlet Temperature:

  float outlet = ((uint8_t)data[144])-128;
  outlet += ((((data[118] & 0b00111000) >> 3)-1)*0.25);

Inlet Temperature:

  float inlet = ((uint8_t)data[143])-128;
  inlet += (((data[118] & 0b00000111)-1)*0.25);

Cheers and happy hacking

As seeing is believing;

[image]

IgorYbema, 2023-02-20:
Yes, that was already known :)

Celphor, 2023-02-20:
But not implemented :-) Double check is never bad

IgorYbema, 2023-02-20:
Yes is implemented, since 3.1 , see https://github.com/IgorYbema/HeishaMon/releases/tag/v3.1

Celphor, 2023-02-20:
My bad ;-)

closed 2023-02-20 by Celphor