heishamon/HeishaMon #326: H43 error at boot with Optional PCB emulation activated

jllussa, 2023-02-24:
Hi, I  have an Aquarea J + Buffer tank with Heishamon 3.0 + One Wire sensor attached since september 2022. I request values from HP/OneWire every 45 seconds.

Since day one, at boot, Aquarea throws error H43 "Z1WaterSensor problem".
I overcomed this issue with Heishamon rule engine:

`on System#Boot then
  @SetZ1WaterTemp = 36;
  @SetBufferTemp  = 36;
end`

This works like a charm and Aquarea boots ok, but from time to time I think about solving this ¿"maybe bug"? directly in Heishamon source code.

Checking source code I see the function setupConditionals()

`if (heishamonSettings.optionalPCB) {
    if (loadOptionalPCB(optionalPCBQuery, OPTIONALPCBQUERYSIZE)) {
      log_message(F("Succesfully loaded optional PCB data from saved flash!"));
    }
    else {
      log_message(F("Failed to load optional PCB data from flash!"));
    }
    delay(1500); //need 1.5 sec delay before sending first datagram
    send_optionalpcb_query(); //send one datagram already at start
    lastOptionalPCBRunTime = millis();
  }  `

My guess is that it's failing to load the file optionalpcb.raw from saved flash and instead it sends the "default data"

`byte optionalPCBQuery[] = {0xF1, 0x11, 0x01, 0x50, 0x00, 0x00, 0x40, 0xFF, 0xFF, 0xE5, 0xFF, 0xFF, 0x00, 0xFF, 0xEB, 0xFF, 0xFF, 0x00, 0x00}; `

As byte 8 is BufferTemp [FF] and byte 16 is Z1WaterTemp [FF]  with BufferTank 
connected the error H43 triggers inmediately.

As function setupConditionals has some log message I'm trying to get those logs but I can't see them in webpage "Console output" nor toggling "MQTT LOG".

Can someone point me so I can get those "extra" logs?

Of course the final goal is to get Aquarea running from cold boot without using the rule engine tweak .

MiG-41, 2023-02-24:
Like discussed on Slack , "Hardware log" (parallel connection to Heishamon with usb serial converter and laptop ) would help to see , what values are sent as first.

jllussa, 2023-02-25:
Well I have managed to get those logs as appointed by Mig-41.
Unfortunately as I thought the file isn't loaded.

Full log --> https://file.io/gLV3RGwr22sZ

`
Starting debugging
Thu Jan  1 00:00:02 1970 (2264): mounting FS...
Thu Jan  1 00:00:02 1970 (2310): mounted file system
Thu Jan  1 00:00:02 1970 (2348): reading config file
Thu Jan  1 00:00:02 1970 (2379): opened config file
Thu Jan  1 00:00:02 1970 (2382): {"wifi_hostname":"HeishaMon","wifi_password":"superaaabbb","wifi_ssid":"casa","ota_password":"aaabbb","mqtt_topic_base":"panasonic_heat_pump","mqtt_server":"192.168.1.112","mqtt_port":"1883","mqtt_username":"admin","mqtt_password":"aaabbb","use_1wire":"enabled","use_s0":"","listenonly":"","logMqtt":"","logHexdump":"","logSerial1":"enabled","optionalPCB":"enabled","waitTime":"45","waitDallasTime":"45","dallasResolution":"12","updateAllTime":"300","updataAllDallasTime":"300","timezone":"345","ntp_servers":"po0üP
Thu Jan  1 00:00:02 1970 (2421): parsed json
Thu Jan  1 01:00:02 1970 (2427): Wifi reconnecting with new configuration...
Thu Jan  1 01:00:02 1970 (2440): Wifi client mode...
Thu Jan  1 01:00:02 1970 (2444): Sync webserver started at port: 80
Mode: STA
PHY mode: N
Channel: 1
AP id: 0
Status: 1
Auto connect: 1
SSID (4): casa
Passphrase (11): superaaabbb
BSSID set: 0
Thu Jan  1 01:00:02 1970 (2486): Failed to load optional PCB data from flash!
Thu Jan  1 01:00:03 1970 (3987): Sending optional PCB data
Thu Jan  1 01:00:03 1970 (3988): sent bytes: 20 including checksum value: 164 
Thu Jan  1 01:00:03 1970 (4023): Number of 1wire sensors on bus: 1
Thu Jan  1 01:00:04 1970 (4085): Found 1wire sensor: 28ff641e8537c1eb
Reset reason: 4, exception cause: 0
4086: rules mempool size: 16384
4086: reading rules
4156: rule #1 was prepared in 4109 microseconds
4165: rule #1 was parsed in 8773 microseconds
4165: bytecode is 112 bytes
4166: rule_function_set_timer_callback set timer #1 to 1 seconds
4167: rule #1 was executed in 1571 microseconds
4170: bytecode is 112 bytes
4176: rule #2 was prepared in 3021 microseconds
4190: rule #2 was parsed in 12820 microseconds
4191: bytecode is 140 bytes
4192: rule_function_set_timer_callback set timer #2 to 1 seconds
4193: rule_function_set_timer_callback set timer #1 to 1 seconds
4198: rule #2 was executed in 6882 microseconds
4202: bytecode is 140 bytes
4212: rule #3 was prepared in 7822 microseconds
4287: rule #3 was parsed in 71656 microseconds
4288: bytecode is 495 bytes
4294: rule_function_set_timer_callback set timer #2 to 45 seconds
4294: rule #3 was executed in 6497 microseconds
4295: bytecode is 495 bytes
4296: rules memory used: 1516 / 16384
4300: ==== SYSTEM#BOOT ====
4302: >>> rule 0 nrbytes: 112
4304: >>> global stack nrbytes: 4
Thu Jan  1 01:00:04 1970 (4310): set optional pcb 'set_z1_water_temp' to temp 36.00 (62)
Thu Jan  1 01:00:04 1970 (4317): set optional pcb 'set_buffer_temp' to temp 36.00 (62)
4323: rule_function_set_timer_callback set timer #1 to 1 seconds
4329: rule #1 was executed in 21325 microseconds
4333: 
>>> local variables
4335: 
4336: >>> global variables
4338: 
Thu Jan  1 01:00:04 1970 (4340): Received 20 bytes data
Thu Jan  1 01:00:04 1970 (4344): Checksum and header received ok!
Thu Jan  1 01:00:04 1970 (4350): Received optional PCB ack answer. Decoding this in OPT topics.`

jllussa, 2023-02-25:
Checking the code: 

 // Callback function that is called when a message has been pushed to one of your topics.
void mqtt_callback(char* topic, byte* payload, unsigned int length){
.
.
.
else if (strncmp(topic_command, mqtt_topic_commands, 8) == 0)  // check for optional pcb commands
    {
      char* topic_sendcommand = topic_command + 9; //strip the first 9 "commands/" from the topic to get what we need
      **send_heatpump_command(topic_sendcommand, msg, send_command, log_message, heishamonSettings.optionalPCB);**
    }

Optional_pcb.raw file is created only when heishamon receives an update of optional pcb MQTT topic!

So for the file to be created at least you have to send one time an update through MQTT topic.
I have posted a message to commands/SetBufferTankTemp via MQTT client and booom, file created.
If a user only updates optional pcb topics through CurlyMo internal rules engine, this file will never be created.

Ok mistery solved. If devs want to improve this in next version is up to then.
I will keep enjoying Heishamon PCB. 
Thanx !

jllussa, 2023-02-27:
To devs:
I think using this piece of code on commands.cpp the "issue" would be gone. 
I don't have Arduino IDE to test so this is only a suggestion.

`unsigned int set_xxx_temp(char *msg, char *log_msg, int byte, const char *func) {
  String set_pcb_string(msg);

  float temp = set_pcb_string.toFloat();

  {
    char tmp[256] = { 0 };
    snprintf_P(tmp, 255, PSTR("set optional pcb '%s' to temp %.2f (%02x)"), func, temp, temp2hex(temp));
    memcpy(log_msg, tmp, sizeof(tmp));
  }

  {
    optionalPCBQuery[byte] = temp2hex(temp);
  }
  
  if ((unsigned long)(millis() - lastOptionalPCBSave) > (1000 * OPTIONALPCBSAVETIME)) {  // only save each 5 minutes
    lastOptionalPCBSave = millis();
    if (saveOptionalPCB(optionalPCBQuery, OPTIONALPCBQUERYSIZE)) {
      log_message((char*)"Succesfully saved optional PCB data to flash!");
    } else {
      log_message((char*)"Failed to save optional PCB data to flash!");
    }
  }
  
  return sizeof(optionalPCBQuery);
}
`

IgorYbema, 2023-02-27:
Hi,
I moved the save routine to the main loop so it saves every 5 minutes. Not depending on a new command (mqtt or rules).
If you want you can try it after it finished building a test firmware:
https://github.com/IgorYbema/HeishaMon/actions/runs/4285495143

jllussa, 2023-06-17:
Tested on build HeishaMon-alpha-44fe6c52988bf49845ee6653e10d54b7.bin
All ok. Thanx!

closed 2023-06-17 by jllussa