heishamon/HeishaMon #328: Integration of esp-knx-ip possible?

MAButz, 2023-03-02:
Hi,
in tasmotoa is esp-knx-ip integrated -> https://github.com/envy/esp-knx-ip

I would like to try this, unfortunately my c++ knowledge is not very good.

Is there any general interest? And could someone support with this?

Greetings
Marc

MiG-41, 2023-03-02:
Ok , but in HP there is not any knx .... 

There is some module to have KNX with Aquarea , bu whatfor , if you can connect esp8266 diretly,

geduxas, 2023-03-02:
https://www.intesis.com/products/ac-interfaces/knx-gateways/panasonic-knx-air-to-water-pa-aw2-knx-1?ordercode=INKNXPAN001A000

MAButz, 2023-03-03:
Thank you!
i know this module, i want to use mqtt and knx, that was my idea.

MiG-41, 2023-03-03:
So if ou use mqtt ,Heishamon is good for that , you don't need knx in beetween.

MAButz, 2023-03-08:
ok, now i use for the moment node-red and send the required values from heishaMon to knx.
Tanks.

closed 2024-09-12 by Egyras