heishamon/HeishaMon #330: Problems with global variables in Rules.

McMagellan, 2023-03-04:
Sometimes, after several hours of correct operation, I see two variables with the same name in the console. One of them has the value NULL. If this variable is necessary for further operation, I have a problem. The rule can then no longer fulfill its task. Otherwise the rules continue.
 
[image]

I have the impression that the problem occurs when a certain number of variables is exceeded. Since I've expanded the program by 2 more variables, I've had this twice within 3 days.

What I have done wrong?

CurlyMoo, 2023-04-03:
Can you post the ruleset you arer running?

McMagellan, 2023-04-04:
I think there is no problem with the program.

It seems to me that the time to find the right variable is running out, so it is simply recreated with a value of NULL. Since I reduced the number of variables, this program has been trouble-free.

My rules cannot be understood without description probability because I control several functions with this ruleset.
If there is a need, I will deliver it.

[Rules DS 4.4.23.txt](https://github.com/Egyras/HeishaMon/files/11150794/Rules.DS.4.4.23.txt)

McMagellan, 2023-04-04:
I forgot to say that I have 13 sensors running in parallel on the 1Wire port and 1 interface for current measurement without any problems. MQTT to iobroker. Attached is a console recording for information.

[Logfile Console v1.txt](https://github.com/Egyras/HeishaMon/files/11151006/Logfile.Console.v1.txt)

CurlyMoo, 2023-04-04:
Would it be possible to create a minimal ruleset that triggers the issue?

CurlyMoo, 2023-04-12:
I found the issue which i will fix as soon as https://github.com/IgorYbema/HeishaMon/pulls has been merged.

@IgorYbema

IgorYbema, 2023-04-12:
I will make time to test again tomorrow

McMagellan, 2023-04-12:
Sounds good.
Unfortunately, I was not able to provide a program that shows this sporadically occurring error.

But I had another problem that is probably related to a runtime problem. Others have also reported about it in the forum while there is no error in the rules. I got around the problem by relocating the write function at the end of the main loop timer to the HP with a extra timer. I had this error about 5x in the last 4 months.

What happens:
After processing the Rules command, Heishamon hangs himself. After a short time, Heishamon can be spoken to again. However, the uptime was restarted and the rules engine is no longer running. No messages from Rules in the console. The time in the time stamp is no longer updated. If I remember correctly, only Heishamon helped to power down, no reboot.

This just for info. I could always find a way to make Heishamon run stably by rewriting the program.
Best regards

CurlyMoo, 2023-10-22:
@McMagellan This should be fixed in the latest build. Also, it would be great if you could help testing:
https://github.com/IgorYbema/HeishaMon/pull/121

Can you confirm if the issue is fixed?

McMagellan, 2023-10-28:
I had previously avoided the problems with the double global variables by not using more than 12 in my rules. After they asked, I increased the number to 16 in my current rule. Unfortunately the error still occurs. I'm already using the new version.3.2.3. Attached are screenshots from the console recording. Upon request, I can make the entire recording (approx. 600KB) available, although I can't really see a trigger.

[image]

I also noticed another error in the recording that I hadn't seen before. It is possible that it is a consequence of the first error.

[image]

I made the ruleset available to you yesterday in #392

closed 2024-09-12 by Egyras