heishamon/HeishaMon #334: IOBroker MQTT datapoints totally weired

Tom54497, 2023-03-09:
[image]

Running the latest IOBroker version with latest MQTT adapter the transmitted datapoints from Heishamon show up like in the attached picture. IOBroker ist still working with it but how to fix this?

Tom54497, 2023-07-26:
Anybody an idea?

geduxas, 2023-07-26:
Once i hade similar issues with very old mqtt server... But you telling that have latest. Could you clarify which version do you have? And which vendor?

Tom54497, 2023-07-26:
It is the MQTT broker/client adapter for IOBroker V 4.1.1

Tom54497, 2023-10-26:
Still having that trouble. Cleared all wrong datapoints and after a while they appear again. Might be an Mqtt Adapter issue. No one an idea?

geduxas, 2023-10-26:
Did you test your mqtt server? Did you connect to it with mqtt explorer? Same results?

Tom54497, 2023-10-26:
If I setup a new instance all datapoints are quiet normal. After a while ... might be after a restart of mqtt adapter but that is only an asumption .... new datapoints will be created additionally to the old and correct structure.

Will try mqtt Explorer

Tom54497, 2023-10-26:
It is the latest stable version of mqtt adapter for iobroker

geduxas, 2023-10-26:
I am not familiar with iobroker, could me point out to resources? Would like to check it.

Tom54497, 2023-10-26:
https://github.com/ioBroker/ioBroker.mqtt

closed 2024-09-12 by Egyras