heishamon/HeishaMon #335: Setting WatthourTotal fails

maust, 2023-03-12:
I am trying to sync my two s0 power meters. As documented I am sending the values with retain flag:

```
mosquitto_pub -t panasonic_heat_pump/s0/WatthourTotal/1 -r -m "1326400.00"
mosquitto_pub -t panasonic_heat_pump/s0/WatthourTotal/2 -r -m "19500.00"
```

Unfortunately, even after several tries it never imports these values during reboot. Is there a way to debug this?

IgorYbema, 2023-03-13:
You need to send this while Heishamon is not running (so during the restart of heishamon, or just turn heishamon off, send the command and then turn it on again).
Heishamon only picks up this during boot.

maust, 2023-03-13:
That's actually what I am doing, to be really sure I disconnected from the heat pump so that heishamon is definitely off.

I turn it off, publish the retain messages, and then also check with a different client, which shows the retained messages:

[image]

Reconnecting heishamon the values start from 0. If I reconnect with the same client the values (in the retained messages) are overridden:

[image]

Thanks for your support.

maust, 2023-03-13:
What I also don't understand is: if heishamon itself sends these values with retained flag (second screenshot), why does it then not pick them up during reboot? On every reboot the values for WatthourTotal start from 0.

IgorYbema, 2023-03-13:
That is the whole reason why it should work as it should. It should restart with the previous retained value. So not sure now why it isn't working for you.

Can you enable 'log to mqtt from start' and then follow the panasonic_heat_pump/log topic while rebooting?
The only hint that it receives the retained values is that it then unsubscribes from this topic after that:
"Unsubscribed from S0 watthour restore topic". This should be in the first few log lines.

maust, 2023-03-13:
I enabled the logging, but I don't see the unsubscribe:

```

Mon Mar 13 12:07:30 2023 (665487): Requesting new 1wire temperatures
Mon Mar 13 12:07:31 2023 (666339): Heishamon stats: Uptime: 0 days 0 hours 11 minutes 6 seconds ## Free memory: 73% ## Heap fragmentation: 13% ## Max free block: 25184 bytes ## Free heap: 28856 bytes ## Wifi: 100% (RSSI: -34) ## Mqtt reconnects: 1 ## Correct data: 100.00%
Mon Mar 13 12:07:31 2023 (666356): Requesting new panasonic data
Mon Mar 13 12:07:31 2023 (666359): sent bytes: 111 including checksum value: 18 
Mon Mar 13 12:07:31 2023 (666796): Received 203 bytes data
Mon Mar 13 12:07:31 2023 (666799): Checksum and header received ok!
Mon Mar 13 12:07:31 2023 (666802): received TOP8 Compressor_Freq: 20
Mon Mar 13 12:07:31 2023 (666807): received TOP33 Room_Thermostat_Temp: 25
Mon Mar 13 12:07:32 2023 (667549): Pulses seen on S0 port 1: Good: 32 Bad: 0 Average good pulse width: 31
Mon Mar 13 12:07:32 2023 (667552): Measured Watthour on S0 port 1: 1.00
Mon Mar 13 12:07:32 2023 (667556): Measured total Watthour on S0 port 1: 32.00
Mon Mar 13 12:07:32 2023 (667562): Calculated Watt on S0 port 1: 652
Mon Mar 13 12:07:33 2023 (668656): Pulses seen on S0 port 2: Good: 0 Bad: 0 Average good pulse width: 0
Mon Mar 13 12:07:33 2023 (668658): Measured Watthour on S0 port 2: 0.00
Mon Mar 13 12:07:33 2023 (668662): Measured total Watthour on S0 port 2: 0.00
Mon Mar 13 12:07:33 2023 (668668): Calculated Watt on S0 port 2: 0
Mon Mar 13 12:07:34 2023 (670141): New webserver client: 192.168.178.20:55798
Mon Mar 13 12:07:35 2023 (670233): Closing webserver client: 192.168.178.20:55798
Mon Mar 13 12:07:35 2023 (670491): Requesting new 1wire temperatures
Mon Mar 13 12:07:36 2023 (671340): Heishamon stats: Uptime: 0 days 0 hours 11 minutes 11 seconds ## Free memory: 73% ## Heap fragmentation: 12% ## Max free block: 25184 bytes ## Free heap: 28664 bytes ## Wifi: 100% (RSSI: -35) ## Mqtt reconnects: 1 ## Correct data: 100.00%
Mon Mar 13 12:07:36 2023 (671357): Requesting new panasonic data
Mon Mar 13 12:07:36 2023 (671360): sent bytes: 111 including checksum value: 18 
Mon Mar 13 12:07:36 2023 (671796): Received 203 bytes data
Mon Mar 13 12:07:36 2023 (671799): Checksum and header received ok!
Mon Mar 13 12:07:36 2023 (671801): received TOP5 Main_Inlet_Temp: 31
Mon Mar 13 12:07:36 2023 (671804): received TOP6 Main_Outlet_Temp: 34.25
Mon Mar 13 12:07:36 2023 (671811): received TOP8 Compressor_Freq: 21
Mon Mar 13 12:07:36 2023 (671817): received TOP15 Heat_Energy_Production: 3600
Mon Mar 13 12:07:36 2023 (671826): received TOP33 Room_Thermostat_Temp: 24
Mon Mar 13 12:07:36 2023 (671834): received TOP65 Pump_Speed: 3950
Mon Mar 13 12:07:36 2023 (671842): received TOP93 Pump_Duty: 146
Mon Mar 13 12:07:37 2023 (672550): Pulses seen on S0 port 1: Good: 33 Bad: 0 Average good pulse width: 31
Mon Mar 13 12:07:37 2023 (672553): Measured Watthour on S0 port 1: 1.00
Mon Mar 13 12:07:37 2023 (672557): Measured total Watthour on S0 port 1: 33.00
Mon Mar 13 12:07:37 2023 (672563): Calculated Watt on S0 port 1: 656
Mon Mar 13 12:07:55 2023 (11180): Heishamon stats: Uptime: 0 days 0 hours 0 minutes 11 seconds ## Free memory: 75% ## Heap fragmentation: 2% ## Max free block: 29216 bytes ## Free heap: 29568 bytes ## Wifi: 100% (RSSI: -33) ## Mqtt reconnects: 1 ## Correct data: 100.00%
Mon Mar 13 12:07:55 2023 (11196): Requesting new panasonic data
Mon Mar 13 12:07:55 2023 (11200): sent bytes: 111 including checksum value: 18 
Mon Mar 13 12:07:55 2023 (11625): Received 203 bytes data
Mon Mar 13 12:07:55 2023 (11627): Checksum and header received ok!
Mon Mar 13 12:07:55 2023 (11629): received TOP0 Heatpump_State: 1
Mon Mar 13 12:07:55 2023 (11632): received TOP1 Pump_Flow: 17.32
Mon Mar 13 12:07:55 2023 (11638): received TOP2 Force_DHW_State: 0
Mon Mar 13 12:07:55 2023 (11644): received TOP3 Quiet_Mode_Schedule: 0
Mon Mar 13 12:07:55 2023 (11651): received TOP4 Operating_Mode_State: 0
Mon Mar 13 12:07:55 2023 (11657): received TOP5 Main_Inlet_Temp: 31
Mon Mar 13 12:07:55 2023 (11662): received TOP6 Main_Outlet_Temp: 34.25
Mon Mar 13 12:07:55 2023 (11670): received TOP7 Main_Target_Temp: 36
Mon Mar 13 12:07:55 2023 (11675): received TOP8 Compressor_Freq: 20
Mon Mar 13 12:07:55 2023 (11681): received TOP9 DHW_Target_Temp: 46
Mon Mar 13 12:07:55 2023 (11688): received TOP10 DHW_Temp: 47
Mon Mar 13 12:07:55 2023 (11695): received TOP11 Operations_Hours: 160
Mon Mar 13 12:07:55 2023 (11698): received TOP12 Operations_Counter: 100
Mon Mar 13 12:07:55 2023 (11705): received TOP13 Main_Schedule_State: 1
Mon Mar 13 12:07:55 2023 (11711): received TOP14 Outside_Temp: 12
Mon Mar 13 12:07:55 2023 (11718): received TOP15 Heat_Energy_Production: 3800
Mon Mar 13 12:07:55 2023 (11724): received TOP16 Heat_Energy_Consumption: 600
Mon Mar 13 12:07:55 2023 (11731): received TOP17 Powerful_Mode_Time: 0
Mon Mar 13 12:07:55 2023 (11738): received TOP18 Quiet_Mode_Level: 3
Mon Mar 13 12:07:55 2023 (11743): received TOP19 Holiday_Mode_State: 0
Mon Mar 13 12:07:55 2023 (11749): received TOP20 ThreeWay_Valve_State: 0
Mon Mar 13 12:07:55 2023 (11757): received TOP21 Outside_Pipe_Temp: 4
Mon Mar 13 12:07:55 2023 (11765): received TOP22 DHW_Heat_Delta: -3
Mon Mar 13 12:07:55 2023 (11768): received TOP23 Heat_Delta: 4
Mon Mar 13 12:07:55 2023 (11773): received TOP24 Cool_Delta: 5
Mon Mar 13 12:07:55 2023 (11779): received TOP25 DHW_Holiday_Shift_Temp: -15
[... 122 lines cut ...]
[... cut ...]
Mon Mar 13 12:08:11 2023 (27202): New webserver client: 192.168.178.20:55804
Mon Mar 13 12:08:13 2023 (29283): Pulses seen on S0 port 1: Good: 5 Bad: 0 Average good pulse width: 31
Mon Mar 13 12:08:13 2023 (29286): Measured Watthour on S0 port 1: 1.00
Mon Mar 13 12:08:13 2023 (29290): Measured total Watthour on S0 port 1: 5.00
Mon Mar 13 12:08:13 2023 (29296): Calculated Watt on S0 port 1: 668
Mon Mar 13 12:08:14 2023 (30083): Requesting new 1wire temperatures
Mon Mar 13 12:08:14 2023 (30576): Received 1wire sensor temperature (2877f05704e13d97): 44.12
```

IgorYbema, 2023-03-13:
Is it possible that your mqtt server isn't supporting retained messages? Did you see existing (older) values when you connected another mqtt client while heishamon was offline?

maust, 2023-03-13:
Like always, the cause of the problem sits in front of the screen. It took me some time but I found the reason:

I am using mosquitto 2.0.11 with user authentication, ACL is configured was allow write, but not read on the topics.

Thanks for your help.

closed 2023-03-13 by maust