heishamon/HeishaMon #336: Question: Explanation for some Values

NeoCortex97, 2023-03-15:
Hi, do you have enough information about some of the parameters?

I was wondering about the **DUTY**  parameter?
I thought it might be the duty cycle of either a heater, or a pump, but that seems to be incorrect.

So I would be very grateful if you could share that information with me if you had it.

geduxas, 2023-03-15:
Do you clarify about what dutty you're talking about?

In Heishamon there is only one pump_dutty, it's work of circulation pump..

MiG-41, 2023-03-15:
Yes , it should be called Max. Dury ( acc manual) , and be in HEX values ,then would be more intuitive...

geduxas, 2023-03-15:
> Yes , it should be called Max. Dury ( acc manual) , and be in HEX values ,then would be more intuitive...

@MiG-41 we have max duty.. just with pump in front :) and also hex value will not make any sense.. just will be same as in screen :) but that's all..

NeoCortex97, 2023-03-15:
I will take a look in the docs for the bytes I meant. But something just was strange. On my pump it is above 100 decimal. That is why I was asking about. since if it is a duty cycle, it would not make sense if it was above 100

NeoCortex97, 2023-03-15:
It is top93 aka byte 172. Pump duty.

Also there is a max pump duty, which I did not look up for now, but that is fixed  on 130 Dec.

NeoCortex97, 2023-03-15:
To me it seems to be the duty cycle of the water pump. But why does it go over 100

geduxas, 2023-03-15:
> I will take a look in the docs for the bytes I meant. But something just was strange. On my pump it is above 100 decimal. That is why I was asking about. since if it is a duty cycle, it would not make sense if it was above 100

No, duty cycle in electronics is not same duty for pump work :) for pump it's more about pumps efficient point between flow ratio and pressure..

geduxas, 2023-03-15:
[image]

MiG-41, 2023-03-16:
> 

Becouse you are thinking in DEC :)
In HEX max value is xFF , with corresponce to DEC 256 , and that is max ( in HP this values are -1 , so 255 in reality)....
You do SetPump and this will set max. duty ( same as in menu and TOP95) , and then TOP73 (so real vaue of the pump ) will not exceed that value ( during normal work , in defrost is not valid).

NeoCortex97, 2023-03-16:
That makes sense to me. Since you guys are way more into that topic, do you know of a good introduction to the operating parameters of a heatpump?
I want to learn, but don't know how to start - as many others newly getting a HP would likely do.

Even if it is not useful to you pros, could you either compile explanations of the parameters, or link a good explanation for these who are techsavy and want to learn. If you would point me to literature, I would be willing to write explanations and issue a PR for you.

I am an IT-guy and act as a smart home integrator for most of my friends, so I need to turn this partly freestanding numbers into graphs with units and alerting rules. 

But thank you all for your help!

geduxas, 2023-03-16:
Try to start reading service manual.. i think there is all you need to know and understand :)

MiG-41, 2023-03-16:
Any PR are welcome , @Egyras is an owner ,and accepting them.

mieschc, 2023-06-05:
@NeoCortex97 If you are from Germany check out: https://www.haustechnikdialog.de/Forum/30/Waermepumpen

In this Forum are many people with self installed Panasonic Heat Pump and many questions/ issues are discussed. Use "Jeisha" or "Heisha" as Key word for a first search.

HighlyCompressedAir, 2024-09-11:
Maybe adding links to service manuals somewhere on the front page is a good idea. This issue can be closed.

closed 2024-09-12 by Egyras