heishamon/HeishaMon #338: Add C2 D3 0C 35 65 B2 D3 0B 96 65 as 	WH-MDC09J3E5

timovd, 2023-03-20:
https://github.com/Egyras/HeishaMon/blob/09e9ae8c0c347881b6de55ffd096fc68f025c78d/HeishaMon/decode.h#L92

C2 D3 **0B** 35 65 B2 D3 0B 96 65 is defined as WH-MDC09J3E5
C2 D3 **0C** 35 65 B2 D3 0B 96 65 is also a WH-MDC09J3E5 (without prefix or extension)

TOP92 is currently sending 255/unknown

geduxas, 2023-03-20:
> https://github.com/Egyras/HeishaMon/blob/09e9ae8c0c347881b6de55ffd096fc68f025c78d/HeishaMon/decode.h#L92
> 
> C2 D3 **0B** 35 65 B2 D3 0B 96 65 is defined as WH-MDC09J3E5 C2 D3 **0C** 35 65 B2 D3 0B 96 65 is also a WH-MDC09J3E5 (without prefix or extension)
> 
> TOP92 is currently sending 255/unknown

In panasonic debug menu those bytes are split in seperate groups.. it maigh be that it could have different purposes.. or maybe different versions.. 

in panasonic menu it would be in this group:

 C2 D3 | **0C** 35 65 |  B2 D3 | 0B 96 65 |

And read from revers side with minus 1 byte
C2 D3 = D3C1
0C 35 65 = 0x65350B
B2 D3 = 0xD3B1
0B 96 65 = 0x65960C

It's requires more investigations, and more data do get what's it means.. it could be internal/external PCB versions, or firmware versions..

timovd, 2023-03-20:
Other (Dutch) user also mentioned the same issue: https://github.com/Egyras/HeishaMon/issues/333

IgorYbema, 2023-03-20:
already in  https://github.com/IgorYbema/HeishaMon/blob/main/HeatPumpType.md
need to add in code yet

MiG-41, 2023-03-20:
BTW @IgorYbema  , are you sure WH-SXC12H6E5 what you just now added id 3phase ? E5 points for singlaphase....

IgorYbema, 2023-03-20:
@MiG-41 that is indeed weird.. panasonic says 1-phase/3-phase

IgorYbema, 2023-03-20:
service manual clearly states 1-phase.. so i'll change that

MiG-41, 2023-03-21:
> service manual clearly states 1-phase.. so i'll change that

Do you have service manual for it ? 
Maybe 6kW heaters are divided in two sections ( 3kW +3KW) so you could conect it to 2 phases , and 3rd phase is outher /compressor unit....

ClassicGOD, 2023-08-31:
Having similar issue with WH-SXC09H3E8 + WH-UX09HE8 kit. 

Returns:
E2 CF 0D 85 05 12 D0 **0E** 94 05
instead of expected:
E2 CF 0D 85 05 12 D0 0C 94 05

Probably a new revision as it was purchased recently.

IgorYbema, 2023-08-31:
I added it to my documentation and it will be added to the code on next release

HighlyCompressedAir, 2024-09-11:
It was added, this issue can be closed.

closed 2024-09-12 by Egyras