heishamon/HeishaMon #343: The Heishamon PCB is unstable.

majorp80, 2023-04-02:
Hello,

I have a problem with the HeishaMon PCB. It has repeatedly lost connection and stopped sending data. During these times, the web interface is also not accessible. The MQTT reconnection sensor value is increasing in the Home Assistant application. The HeishaMon PCB is not stable and unfortunately needs to be restarted weekly. What can be done to solve this problem?

Thank you for your response.

IgorYbema, 2023-04-02:
This is mostly due to bad wifi, network or a not stable mqtt. 
A good heishamon wifi should show at least 65% wifi on the heishamon. If the mqtt reconnects are rising, then this due to the bad wifi or a network issue.

majorp80, 2023-04-02:
[image]

majorp80, 2023-04-02:
[image]

majorp80, 2023-04-02:
Yes, the wifi router is close to the HeishaMon. There is an outdoor ESP-S device further away from it that does not disconnect from the network.
Thank you for the advice.
Is a phone charger an adequate 5V power supply?
Should I connect the GND with the ground of the heat pump?
How should I connect the external power supply?
Should I cut the red and black wires of the CN-Mode cable and connect the external 5V power supply to them?

Thanks for the reply!

IgorYbema, 2023-04-02:
Best is just to remove the 5v pin from the connector on the heishamon side of the cable. You can lift it out easily. And yes leave the gnd connected.
Then just connect the 5v (phone charger) to the 1wire/dallas port where gnd/5v is visible.

Polosaz, 2023-10-05:
Helllo. A doubt, please. The procedure would be to disconnect the red cable from the connector on the heishamon side of the cable and connect the 5v cable from the mobile charger there?. Also leave the black GND cable connected from the connector on the heishamon side of the cable and also connect there the black cable from the mobile charger. Is this correct?

geduxas, 2023-10-05:
Yes for serial communication you need 3 wires, rx/tx/gnd. You can power it from external source, +5v and negative to gnd.

closed 2024-09-12 by Egyras