heishamon/HeishaMon #344: Lost MQTT connection

mkan4B, 2023-04-03:
Hi, I think that I was trying every settings but still have this message:
Lost MQTT connection!

 Heishamon stats: Uptime: 1 day 7 hours 32 minutes 32 seconds ## Free memory: 76% ## Heap fragmentation: 15% ## Max free block: 25408 bytes ## Free heap: 30072 bytes ## Wifi: 100% (RSSI: -42) ## Mqtt reconnects: 0 ## Correct data: 100.00%
Mon Apr  3 07:19:39 2023 (113552733): Received 203 bytes data
Mon Apr  3 07:19:39 2023 (113552733): Checksum and header received ok!
.....

AccessPoint is on the next wall: 

Wifi signal: 100%
Memory free: 65%
Correct received data: 100%
MQTT reconnects: 0
Uptime: 0 days 0 hours 5 minutes 30 seconds
Listen only mode active

but other than that everything seems to be working, I just couldn't receive any MQTT packet

geduxas, 2023-04-03:
Which version of mqtt server you are running? Someone already had problems like this, with old mqtt server..

mkan4B, 2023-04-03:
I was trying 3 difreent:  HA: mosquitto 6.2.0, HiveMQ Cloud, from notebook.

mkan4B, 2023-04-03:
I think that problem is solved but not sure exactly how.

thenebu, 2023-09-16:
I had Heishamon running for 4 weeks. Everything was fine. Nothing changed, but suddenly I got no MQTT Packages. Heishamon log says "Lost MQTT Connection" 
Restartet Heishamon, Home Assistant. Still not working. Cant figure it out, why?

thenebu, 2023-09-16:
I removed the "http://" in front of the ip and now it works! 
Dont know, why it worked for 4 weeks WITH http:// in front...... strange

Benicz, 2023-11-06:
Sidenote:
Heishamon seems to chop long passwords.
So, if it doesn't connect, make sure the password isn't a very long one...

closed 2024-09-12 by Egyras