heishamon/HeishaMon #345: Very high Bad Header Reads, still ok (95%) Correct recieved Data

Happyboy0980, 2023-04-16:
Today, playing with heisamon in Home assistant i noticed that the over the last 3 months i have 1,6 M bad header reads. 
I never noticed this, since Heisamon is indication, at the heisamon frontpage, that the correct recieved data is 95%. 

Don't fully understand this, since more then 50% of the data (1.6 M bad header reads, and 1.4M correct recieved) sounds to me like a bad situation / not correct recieved.
That the are not recieved wel make sense, since the cable between the WP and heisamon is 12 Meter (but shielded). 

Should we indicate this / calculate this in a different way, or at least some how indicate that connection between heisamon and the WP is  not good ? 

My WP : 

Panasonic WH-MDC05H3E5 with additional PCB
Heisamon 3.1

Heisamon:

[image]

HomeAssistant (Mqqt data):

[image]

IgorYbema, 2023-04-16:
Bad header reads are not for each read correctly measured.
As a complete good read should be 233 bytes one good read will count as 233-bytes being read correctly. 
But if the header of that 233-bytes is wrong, it will count already 1-bad header. It will then look at the next byte (for a new header) but will find that that is wrong also. So in 233-bytes there can be 233 bad headers.
This means that one complete bad read will destroy the bad header counter.

The only good sense of the quality is the total divided by the good received. And that is shown as the 95% value. Not bad.

Happyboy0980, 2023-04-16:
Thank you very much for your clear answer ! 
I am little surprissed since i have 12 meter (at least) cable in between so was expecting this would give some isseus.
I will close this one !

IgorYbema, 2023-04-16:
I once tested with 15 meter and still had a good connection. Didn't try for long so don't have exact figures.
Do you use external power or also powering over that 12 meter cable?

closed 2024-09-12 by Egyras