heishamon/HeishaMon #348: SG Ready Function

cvastag, 2023-05-01:
I use HA and the sg_ready function of the PCB.

my configuration.yaml
rest_command:
  sg_ready:
    url: "http://192.168.1.72/command?SetSmartGridMode={{text}}"

my automation automation

service: rest_command.sg_ready
data:
  text: 3

can you explain me the difference between 0,1,2,3? 
I think:
0= off
1= auto
2= higher Power
3= full power

MiG-41, 2023-05-01:
For mqtt we have: 
00 = normal,
10 = HP/DHW off
01 = Capacity 1
11 = Capacity 2
Capacity 1 and Capacity 2 you can edit/modify in controler SG menu . 
At the end it will multiply set values in SG menu by target temperature and "implemen that temperature" as new target.

MiG-41, 2023-05-02:
Acc this: https://github.com/Egyras/HeishaMon/blob/master/OptionalPCB.md
So the topic is panasonic_heat_pump/commands/SetSmartGridMode with values 0,1,2,3...  It will work if OptionalPCB Emulation is enabled in Heishamon , and in HP also Optional PCB and SG Ready enabled.

Grufti68, 2023-05-05:
Is it also possible to use the SG-Ready option if the optional circuit board is installed?

HighlyCompressedAir, 2024-09-11:
> Is it also possible to use the SG-Ready option if the optional circuit board is installed?
It's not.

This one probably should be explained better along with SetExternalThermostat1State/SetExternalThermostat2State in https://github.com/Egyras/HeishaMon/blob/master/OptionalPCB.md because topic values are decimal and decrypted values are binary.
00 = 0
01 = 1
10 = 2
11 = 3

Other than that SG Ready is explained in the service manual quite well, this issue can be closed.

closed 2024-09-12 by Egyras